mongoose-os-apps/shelly-homekit #90: Shelly 1 shown as "unavailable" in Home App

Whiterabbit64, 2020-06-18:
Hello.

I've tried everything on my WiFi network to solve this problem: my 3 SHELLY 1 switches go offline several times a day under Home app.
I have other HomeKit devices (Hue, Netatmo, Xiaomi) on the same WiFi and the SHELLY are the only ones to have this behavior.
When they go offline in Home ap they are still "pingable" with no packet loss. I also see them with the app Discovery on my iPhone. The WiFi coverage is excellent in my house.
I've setup a dedicated SSID for SHELLY after reading posts giving some specific parameters (no WMM, DTIM 2), but no success.
I suspect an issue in the wifi driver of the SHELLY. Any way to check that ?

Thanks.

andyblac, 2020-06-18:
another thread here https://github.com/mongoose-os-apps/shelly-homekit/issues/46 same issue.

andyblac, 2020-08-09:
did the update to 2.0.12 fix your issue ?

Whiterabbit64, 2020-08-09:
As I’m not at home, I can’t update them. Should be able to that by the end of this month.

Whiterabbit64, 2020-08-24:
Upgrade done yesterday. 

What I observe is a great improvement : Shelly's web page load faster. In Home app, Shelly's status is now instantaneous. Siri on HomePod is now able to interact with Shelly.
So far so good.

I'll post again here in few days to confirm if everything is stable.

Whiterabbit64, 2020-08-26:
After several days, I can say it's very stable and works like a charm 👍
Issue is now solved.

Well done 👌

rojer, 2020-08-26:
thanks for getting back, closing this now!

closed 2020-08-26 by rojer