mongoose-os-apps/shelly-homekit #91: Shelly 1 not recognized in Apple Home App?

Sindbad5, 2020-06-20:
Hi, I am using two Shelly 1 v3.
Before changing the firmware I tried both with the Shelly app. Worked fine.
After rebooting with the newest HomeKit firmware I opened the browser and configured both with their individual HomeKit setup codes. Then I tried to add the devices with Apple Home App on iPhone and iPad (my Apple TV is the central device for HomeKit). No Shelly is present?
Rebooting, removing fuse, waiting and praying - nothing helps. All over the time they are connected to the router and reachable over Webbrowser.
Any idea?

Whiterabbit64, 2020-06-22:
Hello.

Do you have 1 single SSID for both 5 and 2.4 GHz ?
I had the same issue. I solved by separating my WiFi network into 2 different networks : 1 SSID for 2.4GHz devices and 1 for 5GHz devices. My Homekit hub is the HomePod.

andyblac, 2020-06-22:
and make sure your not using 40Mhz 2.4.Ghz band, as Shelly's only support the 20Mhz bands

Sindbad5, 2020-06-24:
Hello, thanks for replying.
I do have different SSIDs for 2.4 and 5 GHz and I am using 20 MHz bands.
After experimenting a lot I could **once** see both  shelly's in Homekit. One of them I was able to connect to Homekit. The second is never seen in Homekit again but both are every time reachable over Browser. The one connected to Homekit doesn't work reliable because the connection is missing.
My network is Fritz!Box as router with some Fritz Repeaters (Mesh) in one line.
The last thing I will try is to put an extra repeater directly connected to the router (WLAN) half a meter beneath the shelly's and have a look if this works.

Sindbad5, 2020-06-25:
I‘ve tried some more things ...
Now I can say the WLAN connection is stable!

Using the Shelly-HomeKit UI with browser Safari or Firefox I do have some crazy experience: every time I „TurnON“ the shelly „Uptime“ starts new, so every time I want to change the state of the switch I loose the network connection and a reboot happens. That‘s not what I expected.

I tried with both shelly‘s - they act same.

Now I contacted Shelly support.

Sindbad5, 2020-06-30:
After a lot of structured testing ...

I do have a Fritz!Box with some repeaters. After stopping the whole Mesh-Service for all Fritz-Devices it seems to be working.
🤔🙂

rojer, 2020-06-30:
in https://github.com/mongoose-os-apps/shelly-homekit/issues/30#issuecomment-650724954 i posted firmware images with updated sdk. can you 5est them and let know if it's any better than before?

Sindbad5, 2020-06-30:
On first examination it seems to work more stable (without Mesh).
I will also take a closer look using mesh.

Whiterabbit64, 2020-06-30:
> in [#30 (comment)](https://github.com/mongoose-os-apps/shelly-homekit/issues/30#issuecomment-650724954) i posted firmware images with updated sdk. can you test them and let know if it's any better than before?

Where is the new firmware ? Same version number (2.0.1) ?

rojer, 2020-06-30:
> > in [#30 (comment)](https://github.com/mongoose-os-apps/shelly-homekit/issues/30#issuecomment-650724954) i posted firmware images with updated sdk. can you test them and let know if it's any better than before?
> 
> Where is the new firmware ? Same version number (2.0.1) ?

files are attached to that comment.

Whiterabbit64, 2020-06-30:
Thanks.

I've just upgraded my 3 Shelly1 to 2.0.2 firmware.
On my network (Cisco Wireless) they still go off from HomeApp several times an hour.
What I observed is that they connect in G mode on my 2.4Ghz SSID which support both G and N modes.
Is that a normal behavior (connecting in G mode only) ?

andyblac, 2020-06-30:
I only have N enabled in 2.4Ghz on my AP's.

rojer, 2020-06-30:
> Thanks.
> 
> I've just upgraded my 3 Shelly1 to 2.0.2 firmware.
> On my network (Cisco Wireless) they still go off from HomeApp several times an hour.
> What I observed is that they connect in G mode on my 2.4Ghz SSID which support both G and N modes.
> Is that a normal behavior (connecting in G mode only) ?

honestly, i don't know. what exact model of router is it? if only i could reproduce it reliably...

Whiterabbit64, 2020-06-30:
I use a WIFI controller from Cisco (WLC-2504) and 3702i Access Points (from Cisco). My network is a layer 2 network, several SSID but fully bridged to the same router. No VLAN. The router is my ISP Box (Orange in France) and act as DHCP server.
The point is : I always see my 3 Shelly1 under the _hap._tcp section in ther DISCOVERY app, but they show NOT AVAILABLE in the HomeApp several times. But this also depends on the device I check the HomeApp. It can be available on my iPhone but unavailable on my iPad. I've tried all possible configurations on my WIFI network withtout success.

rojer, 2020-06-30:
ok, looks like it's a different case of unavailability.
i actually just had this recently too: working just fine on ipad, "not responding" on an iphone.
but i had one bridge in the way too: ipad directly connected to the same ap as devices, iphone to a different one, bridged over ethernet. hm.

Whiterabbit64, 2020-07-02:
What I can observe (always with the DISCOBERY app that show all the homekit devices on a network) is that SHELLY devices disappear several times a minute from the _hap._tcp. section while my other homekit devices remain always on (Netatmo weather, cameras, Xiaomi gateway, Philips Hue). I beleive there's an issue in the protocol layer (TCP ? mDNS ? Other ?) of the firmware that is more or less visible depending on the local network setup.

Sindbad5, 2020-07-05:
After a closer look at the system events WLAN of the Fritz!Box I recognized that due to ‚Auto-channeling‘  the repeaters as well as the shellys disconnect and connect again. Sometimes it works but most often it doesn‘t connect correct. It seems that it connects again when the original channel from the first connection is selected?!

andyblac, 2020-09-20:
is this still an issue ?

Whiterabbit64, 2020-09-20:
Everything worked fine until yesterday. 2 of my 3 Shelly were unreachable via Home App. I had to reboot them through the web interface. The last reboot of the first one happened 26 days ago, and 27 days ago for the second one. The connections showed 3/0/9 if I remember well.

andyblac, 2020-09-20:
ok this in an issue being discussed here https://github.com/mongoose-os-apps/shelly-homekit/issues/129 I'll close this ticket please drop over to the other ticket to add your comments thanks.

closed 2020-09-20 by andyblac