mongoose-os-apps/shelly-homekit #93: Switch status in Homekit

Marfre888, 2020-07-09:
Hi,

Firstly thank you for this firmware. I first tried using Shelly homebridge to retains Shelly firmware usability, but unfortunately, I was faced with a lot of disconnects and some relays wouldn't even connect. Happy to report that this is working perfectly though.

I'm after a small enhancement to this firmware, if at all possible. I've run into a small problem with automation and motion detectors. So far I have in setup to turn on/off my lights when motion is detected. However, I sometimes want to keep the lights on even when no motion is detected.

Would it be possible to either: Create an additional field which states if wall switch status is on/off. The idea with this is that I can tell automation to not run if wall switch is on.

or

Create an additional field in the webUI which allows you to set priority to the wall switches. IE: If wall switch is on, homekit cannot turn it off.

Marfre888, 2020-07-09:
Another idea could be to expose the switch to homekit. That would effectively give you an on/off state and would hit two birds with one stone (as I'm seeing a lot of people asking to run automations off the switch.)

timoschilling, 2020-07-09:
Your first idea is not possible.

> Another idea could be to expose the switch to homekit.

That works and would be an nice enhancement to have an feature flipper for that. But it would mean that you need to link the switch and the light with an automatisation.

Marfre888, 2020-07-09:
That's fine. Only an additional rule would be required to ignore the automation if wall switch state is set to on.

andyblac, 2020-07-12:
why not just add a Condition to your HomeKit automation, I have same setup for my hall light. if the light has manually been turned on do not run the automation, but, if you walk in the room, and the outside LUX is less than 50, and light is off, turn on the light, and turn of the light after 3 mins of no motion.

[image]

Marfre888, 2020-07-14:
I understand your reasoning, however, in my case it would never work correctly.

Firstly, I have two motion sensors in that location, and both trigger independently, depending on which side I walk in from. Homekit doesn't cater for this approach, so I had to come up with a creative approach to get it to work.

Secondly, the motion sensor will trigger before I walk to the switch, and if I do turn it on from the wall switch, if won't make any difference as the automation will kick in and turn off the light anyway.

pivale, 2020-07-14:
Have you considered having two automations?
1. turns on if switch if off and lux < 50 (no turnoff)
2. turns off if switch is off when motion is not detected for over 3 minutes

Marfre888, 2020-07-14:
@pivale I'm not sure how that would help.

The shelly devices current expose the state of the light, not the state of the switch to Homekit. ie: It makes no distinction between turning the light on from a switch or from the Home app. They both show up as being 'on' in Homekit irrelevant of the switch state.

There is currently no way to know the state of the switch, which is what I'm after.

Reason being is that I run timers from homebridge, which reset if either of the two motion sensors connected to the Shelly detected motion. There is no easy reliable way to do this in homekit.

Without the state of the switch exposed, it is impossible for me to stop the lights from turning off, because the motion sensor will always trigger before I can turn the switch on, and in turn, the timer will always turn the light back off once it runs out

andyblac, 2020-07-14:
in this case your after this imohttps://github.com/mongoose-os-apps/shelly-homekit/issues/9, once this is added, the switch would have not physical link the light, you can then control the automation within HomeKit. ie single press " IF light is off, turn in it on, IF light is ON, cancel your Homebridge countdown"

Marfre888, 2020-07-14:
@andyblac Yes, I think that would work nicely

I have no reason to need a double press/long press cause I only have an on/off switch

and it would be nice to have the switch control the light by default, and avoid this instruction being passed through Homekit, so in case there ever is a failure to Homekit, the light still retain the manual way of turning them off. 

But #9 would a great start for what I'm after. Any timeframe on this?

andyblac, 2020-07-14:
@Marfre888 I don't, but @rojer has said in the last message in that thread, it is next on the todo list, so lets hope 😄

Marfre888, 2020-07-14:
Looking forward to it....

To summarise: what I'd like to see is the physical switch retaining it's current functionality. IE: It is locally able to turn the light On/Off without needing Homekit. But Shelly is able to report the state of the switch, effectively giving me the option of disabling the automation if the switch is in the 'On' position.

'Wall Switch' Off and 'Light' Off = If motion detected triggers automation
'Wall Switch' Off and 'Light' On = if motion detected automation time extended
'Wall Switch' On and 'Light' On = light remains on
'Wall Switch' On and 'Light' Off = should not be possible (but resettable by turning 'Wall Switch' Off, or On and back Off

A nice bonus would be if they could be grouped together in one tile.

andyblac, 2020-07-28:
@Marfre888 been thinking about this, I have something similar in the Kitchen, I do it like this;

replace wall switch to a Momentary Switch, 

Automation: IF kitchen light is OFF, turn ON kitchen light (auto off in HomeKit set 3 mins)

if I keep motion going in the Kitchen the Light stays on, the way it works is by the Motion Sensor resets the auto off each time it is activated, and IF the light is already on it will NOT turn ON the light with the auto OFF.

andyblac, 2020-07-28:
also what could be done is get the shelly AutoOff have it shown in HomeKit as a tile.

@rojer could you do this ?, would it be possible to show the AutoOff as dummy switch tile in HomeKit, so it turns ON when the light is on (as it starts count down in code) and when count down is reached as it turns off Switch relay, turn OFF the dummy switch tile in HomeKit.

handy-manny-swe, 2020-08-02:
Sorry @andyblac , I won’t be able to look at this for several months.

Marfre888, 2020-08-11:
@andyblac I understand. That is how it is setup at the moment. with the delay plugin in homebridge.

This however doesn't address my initial issue. Which is that, I want to be able to stop the motion sensor from turning off the lights is the switch is in the ON position. Reason being that occasionally I have guests over, or I'm cleaning and want to keep the lights on.

I know I could just stop the automation or just off the motion sensor, but I don't like having to use my phone all the time, plus the rest of my family will never be able to figure it out.

andyblac, 2020-09-19:
Detached Mode is now supported in https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.1.7, please try and report back if it fixes this issue for you, it separates the physical switch from HomeKit button, both can now work independent of each other, and give you a new input button in HomeKIt with Single, Double, Long Press options for the physical switch.

[image]

rojer, 2020-09-19:
thanks Andy!

closed 2020-09-19 by rojer