mongoose-os-apps/shelly-homekit #95: After flashing to 2.5 and setup can’t connect

macguru341, 2020-07-10:
Hello
So I’ve been able to flash the HomeKit version to my Shelly 2.5.
I connected to the Shelly API and went to the address of the Shelly to setup HomeKit.
I changed the names and also insert my WiFi name and password.
Than I plug out the ECP USB and and plug it back in to check the HomeKit and the Shelly is not loading up.
I see a red fast blinking led and every 1 Min it start flashing slow for 1 Sec and than back to flashing fast.
What can be the problem? It looks like it can’t mange to connecting to my WiFi? 
How can I fix it?
I also want to mention that I did the process from the beginning for 3 times , also flash back the original Shelly 2.5 and again the HomeKit version but I get the same problem.
As I said I’m not using the OTA option instead I’m using the ECP-USB option with MOS.
Thanks

andyblac, 2020-07-12:
if you can get it back on original firmware, and access the WebUI, then just refresh via OTA, no reason to use serial flash. Also try power up the shelly via 240v and not USB power, as that is not always reliable to boot the the Shelly 2.5.

andyblac, 2020-09-20:
is this still an issue ?

andyblac, 2020-09-22:
I'll close this ticket, please reopen if it's still an issue.

closed 2020-09-22 by andyblac