mongoose-os-apps/shelly-homekit #96: Intehgrate Prometheus Exporter

Subnet-Zero, 2020-07-10:
Hi,
i think it's useful to integrate a prometheus exporter in the software, e.g. to check the switch state.
An exporter is no more than a website, that shows the metrics.
https://prometheus.io/docs/instrumenting/writing_exporters/

Do you think its possible? That would be great.

Thanks.

rojer, 2020-07-10:
we even have a library already - https://github.com/mongoose-os-libs/prometheus-metrics
however, i don't think we have the space on flash for this... unfortunately

Subnet-Zero, 2020-07-13:
Thanks for reply.

I've seen there is a JSON-Endpoint on it.
I use a JSON-Exporter to expose the metrics to Prometheus (.../rpc/Shelly.GetInfo).

That works well. :)

closed 2020-07-13 by Subnet-Zero

rojer, 2020-07-13:
clever :)
note that it's not an official API though, it only exists to be used by the Web UI and as such is subject to change.