mongoose-os-apps/shelly-homekit #98: shelly 2.5 disconnected and can't recover it

eldad007, 2020-07-15:
Hi
my shelly 2.5 disconnected from wifi after I used it on HomeKit and now it can't log in back to the wifi network
I tried to press the button on the shelly for 10 seconds and the lite is stuck on:
"On, short off pulses - HAP server provisioned, WiFi not provisioned (AP active)."

I can see the AP of the shelly in my wifi list but can't do anything

how can I make the shelly log in to my wifi again?
how can I log in the web api of this shelly?

Eldad

andyblac, 2020-09-20:
did you fix your issue ?

eldad007, 2020-09-20:
> ‫ב-20 בספט׳ 2020, בשעה 16:18, ‏‏Andy Blackburn ‏<notifications@github.com> כתב/ה:‬
>
>
> did you fix your issue ?

Yes
Thanks

closed 2020-09-20 by andyblac