mongoose-os-apps/shelly-homekit #99: Can't Tell if 1PM is working or not

400HPMustang, 2020-07-19:
I have just flashed a 1PM with Mongoose. I have it added to HomeKit, I can connect via browser, I can see it in the Home app. 

I have the device connected with usb power just for testing at the moment- 3.3v and ground. 

The light on the 1PM does not light up and when I flip the switch off/on either in HomeKit or the browser I do not hear the relay clicking. When I would flash these with HAA, with power applied the LED would light up and when cycling off/on I could hear the relay.

How can I tell if the device is working before installing it? I don't want to get the thing installed, find out it does not work, and then have to take everything apart.

andyblac, 2020-09-22:
still need help on this?

400HPMustang, 2020-09-22:
No, sorry I forgot to close it. Apparently there's just no light on the
puck when using Mongoose and the relay doesn't click when it's just being
powered via USB for flashing/testing purposes. I managed to configure 3 of
them since I made my post.

andyblac, 2020-09-22:
thx I'll close ticket,

closed 2020-09-22 by andyblac