mongoose-os-apps/shelly-homekit #102: Support for Shelly i3

gijsstegehuis, 2020-07-25:
Would it be possible to add support for the Shelly i3?
It would make an awesome Homekit stateless button.

Would be much appreciated.

rojer, 2020-10-02:
i've actually take a stab at it recently and ran into a problem with inputs in i3 being extremely noisy for some reason.
not a problem on 1 or 2.5, but for some reason on i3 it is, and a big one.
need to investigate.

rojer, 2020-10-29:
will be in the next release

closed 2020-10-29 by rojer

timoschilling, 2020-11-09:
@rojer can you ship a new release with i3 support please

rojer, 2020-11-09:
2.5.0 is coming soon, just waiting for confirmation for a couple fixes and thinking of maybe getting a couple more in. soon.

rojer, 2020-11-09:
2.5.0 released.

gijsstegehuis, 2020-11-10:
Awesome work @rojer. Will take this for a spin in the coming days.

gijsstegehuis, 2020-11-11:
Works like a charm @rojer

jerzytraczynski, 2020-11-15:
Does this support double or long press?

rojer, 2020-11-15:
yes