mongoose-os-apps/shelly-homekit #103: Shelly 1 Switch status via SW-Connection

Saronovic, 2020-07-25:
Hi,

i use a Shelly 1 for my garagedoor. "L" and "N" is connected to 230V and "0" and "1" I use as potential-free contact. Is it possible to use the "SW" for an end switch to see if the door is closed or open? For an example when the end switch is closed you have voltage at the "SW" contact and the door is closed. When you use the original FW you can see then that the small line is getting blue.
 
[image]

andyblac, 2020-07-26:
@rojer would be better to confirm, but I would think the sautés should be on "O" if it's relay is open or closed. but what you are asking should be possible to for, as we have 4 different button types for SW, so it dose know when the SW has a closed / open circuit.

Saronovic, 2020-07-26:
I think you have to use the "SW" for that. "I" and "O" is used as a momentary contact switch and they are directly connected to the motor drive from the garagedoor. The auto off delay is 0.5sec. A short press and the door open/stop/close and so on.

fritte87, 2020-07-30:
Hi, here the same...at amazon there is a shelly 1 with HOMEKIT Firmware which operated with a reed contact on SW...
so how I have to configure the Shelly 1 ? Anybody Knows?

Thanks

pfandrade, 2020-08-15:
I'm actually doing this with the stock firmware and a fork of homebridge-shelly. I'm keen on flashing this firmware to no longer need homebridge so here's my +1 for this feature.

andyblac, 2020-10-11:
@rojer as this kind of replicates what garage door state is, could this now be easily added ?

rojer, 2020-10-11:
thanks @andyblac , this is exactly that. @Saronovic @pfandrade, please join the discussion on #142 and help test pre-release versions (links in the comments on that issue).

closed 2020-10-11 by rojer