mongoose-os-apps/shelly-homekit #104: Support for Button1

andyblac, 2020-07-26:
could support for this be added easily ?, it would be nice have it acting like this https://github.com/mongoose-os-apps/shelly-homekit/issues/9

pivale, 2020-08-04:
It would be really cool, but Button1 doesn't seem to be build on ESP8266.
Or at least they don't advertise it like that...

DavRin, 2020-11-18:
Hi, news for Button1 HomeKit firmware?

rojer, 2020-11-19:
no news yet. i actually have one for testing, i've only had a brief look at it and it's one of those that have slave micro in them, so it won't be trivial.

RedAnon, 2020-12-01:
In the meantime is there an alternative for physical switch/button to use alongside the Shellys? Preferably also with FOSS firmware?

andyblac, 2020-12-01:
> In the meantime is there an alternative for physical switch/button to use alongside the Shellys? Preferably also with FOSS firmware?

not that i have come across, i tried using Homebridge Shelly plugin, but it does not work very well either, i could only get Long Press action to work.

andyblac, 2020-12-01:
only homekit button is Fabaro but dam those are expensive at £50 ea https://www.amazon.co.uk/Fibaro-Homekit-Ausführung-verschiedener-drahtlos/dp/B07CZQGV42/ref=sr_1_5?crid=5CSVTVLST78E&dchild=1&keywords=homekit+button&qid=1606833828&sprefix=homekit+butt%2Caps%2C147&sr=8-5

RedAnon, 2020-12-01:
Or for "only" 70 quid you can get something like the AQARA hub and one mini switch but I kinda dislike anything that needs a hub as it feels like an even bigger vendor lock in plus the main reason why I choose these Shelly devices is that if I one day get bored with the Apple ecosystem (which as a long time Linux user I inevitably will) I can simply re-flash them.

All in all there probably aren't any nice buttons on the market, the Shelly Button1 isn't ESP8266 so it might take a while (if ever) to get supported. So I'll probably have to look into some DIY 3D printed solutions like (https://www.thingiverse.com/thing:2803769)

sebastianheise, 2020-12-05:
I have looked at this, and while I don't think the button can be easily directly supported in HomeKit, I do see an opportunity to still use it to control any switches directly. 
But we would need to change (or add to) the rcp calls:
- The Shelly button can be configured to call a URL when pressed.
- The current web interface has a "turn on/turn off" button, that does a `POST /rpc/Shelly.SetSwitch` with a json encoded body of `{"id":1,"state":true}` or `{"id":1,"state":false}`. 
- There currently is no real security implemented (not even CSRF), so my suggestion would be to allow a simple `GET /rpc/Shelly.SetSwitch?id=1&state=true|false` request?

Any thoughts on that?

andyblac, 2020-12-05:
ye, i plan on doing same with shelly motion sensors, but i need it to control other (none shelly) homekit devices.

rojer, 2020-12-05:
> my suggestion would be to allow a simple GET /rpc/Shelly.SetSwitch?id=1&state=true|false request

this is already possible

andyblac, 2020-12-05:
documented here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API i use it already.

andyblac, 2020-12-05:
and once this https://github.com/mongoose-os-apps/shelly-homekit/issues/309 gets implemented, we could use shelly to controll other HomeKit devices by the API.

sebastianheise, 2020-12-06:
Oh interested. Sorry, I missed that. Cool. So that's the workaround to use Button1 for the time being. Thanks! :)

andyblac, 2020-12-06:
you can use deteched inputs to `http://192.168.3.49/rpc/Shelly.InjectInputEvent?id=1&event=2`

Loggss, 2021-11-05:
in case where I need to activate two shelly 2.5 with a single press on Shelly button, how should we write the url?

andyblac, 2021-11-05:
ATM the API does not have a 'Toggle' action, so only way to achieve this, is to use homebridge with the Shelly Button1, and use a HomeKit action like here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

Loggss, 2021-11-05:
> ATM the API does not have a 'Toggle' action, so only way to achieve this, is to use homebridge with the Shelly Button1, and use a HomeKit action like here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

thanks, sounds pretty clear for a clean integration. ATM I do not have a homebridge server, may I launch 2 urls with the shelly cloud interface ? (and write two different urls commands like you mentionned before, e.g. http://LOCAL_IP_ADRESS/rpc/Shelly.InjectInputEvent?id=1&event=2)

andyblac, 2021-11-05:
you could do via 1 press ON, 2 presses OFF in the shelly firmware web interface

timoschilling, 2022-02-05:
@pivale @RedAnon I have Button1 at hand, the mac address says it is a Espressif, so I expect it is a esp8266. But since I wasn't able to open it I can't validate it.

timoschilling, 2023-02-21:
replaced by https://github.com/mongoose-os-apps/shelly-homekit/issues/1233

closed 2023-02-21 by timoschilling