mongoose-os-apps/shelly-homekit #105: Shelly1 Garagedoor Status

fritte87, 2020-07-31:
Hi @ all,
@ Amazon there is a reseller (HomekitBG) who sells Shelly1 with Homekit Firmware, which can give out the status from a reed contact.

So @rojer is it possible to get the same of your HomekitFirmware?

Here is the link from the Reseller:

https://www.amazon.de/HomeKit-compatible-Shelly-Garage-Door/dp/B07WN8YXGP/ref=sr_1_1_sspa?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=shelly1+homekit&qid=1596187613&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFXTDlCSTZORFZEVFAmZW5jcnlwdGVkSWQ9QTA2MDQwNjMzOFgwTTVCUVc0SlozJmVuY3J5cHRlZEFkSWQ9QTAzOTM5MjhUWFVGOTFESjZaNFQmd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

Maybe it is possible....so this feature will be very very nice!

fritte87, 2020-08-01:
Thanks Andy,
I read the Whole #5 but Here they talk about roller shutter.
I Need the Shelly1 because there a dry contacts in it...
So Shelly 2.5 is Not usable for me die garagedoor

Ludelag, 2020-08-07:
Hello, I have the same question about it, see this video: https://www.youtube.com/watch?v=P5mQADnZXOI
and also this video with a multy accessory:
https://www.youtube.com/watch?v=_hFVtwb0wug&t=508s
There is any possibility to have this firmware?

SniFManS, 2020-08-14:
Shelly1 for Garage + sensor setup, please. 

Passionate domotica is using his firmware for Homekit and uses Eve to fine-tune it so it is working in Homekit.

Martinx64, 2020-08-16:
Yes, it would be very nice to have firmware for garagedoor opener..

Ludelag, 2020-08-17:
> Shelly1 for Garage + sensor setup, please.
> 
> Passionate domotica is using his firmware for Homekit and uses Eve to fine-tune it so it is working in Homekit.

Good evening, what sensor will you recommend.? And how to make them work together?

SniFManS, 2020-08-19:
> > Shelly1 for Garage + sensor setup, please.
> > Passionate domotica is using his firmware for Homekit and uses Eve to fine-tune it so it is working in Homekit.
> 
> Good evening, what sensor will you recommend.? And how to make them work together?

Look at this setup : [https://www.youtube.com/watch?v=WEZUxXNiERQ&t=520s(url)

fritte87, 2020-08-27:
any news for the firmware ? Thanks

andyblac, 2020-10-11:
closing this one one as it being worked on.

closed 2020-10-11 by andyblac

rojer, 2020-10-11:
@fritte87 @SniFManS yes, work is underway. if possible please join testing of early versions. you'll find links in comments on #142.

SniFManS, 2020-10-11:
Thanks I will