mongoose-os-apps/shelly-homekit #106: Local file Firmware update procedure

Andi161, 2020-08-02:
No success in updating Shelly-1 Firmware from Version 2.0.1 to 2.0.11:
dragged the downloaded zip-file into the box "file selection" on the bottom of the web-GUI, and hit the blue Update button. The GUI is replaced by a message "-1 Archived file size is unknown" Shelly-HomeKit webpage is gone for 2 minutes, and can be pulled back again with no firmware change, still at 2.0.1. Same with update using 2.0.10.
What am I doing wrong?

Andi161, 2020-08-02:
Ok, found the problem: when clicking the download link MacOS Catalina did not download the zip file, but the contents of it. Only by right click - load file, the zip is downloaded. 
So I actually never tried to install the zip.

Andi161, 2020-08-02:
Please close the issue

rojer, 2020-08-02:
yes, mac os is trying to be helpful and ruining everything :)

closed 2020-08-02 by rojer