mongoose-os-apps/shelly-homekit #108: Devices appear and disappear from Bonjour

rojer, 2020-08-05:
Originally reported here:

https://github.com/mongoose-os-apps/shelly-homekit/issues/30#issuecomment-665771893

Devices seem to be constantly appear and disappear from Bonjour.
This may or may not impact connectivity, this is not clear, but should be looked into anyway.

rojer, 2020-10-23:
can you try this firmware: https://rojer.me/files/shelly/2.4.0-beta2/
it has more frequent advertisements (every 20 seconds instead of every 60), hopefully enough will get through so records don't expire. @NS-K, can you try and let me know if it's any better?

Supereg, 2020-10-23:
@rojer quick question (as I'm also developing some mdns stuff currently). Does the mongoose mdns library send periodic announcements? Quickly checked the code and it seems like it does.
If I'm not wrong a mdns responder must not send periodic announcements.

[From RFC 6762 8.3. Announcing](https://tools.ietf.org/html/rfc6762#section-8) 
```
   A Multicast DNS responder MUST NOT send announcements in the absence
   of information that its network connectivity may have changed in some
   relevant way. In particular, a Multicast DNS responder MUST NOT send
   regular periodic announcements as a matter of course.
```

Am I missing any information? I'm currently also struggling with mdns related in my own mdns implementation (seemingly only in vlan setups), so I'm currently searching for anything I might have missed in my implementation.

rojer, 2020-10-23:
um... ok, that's news to me. yes, we do advertise at regular intervals. not sure i'[m comfortable with removing this... the standard is pretty clear though.

Supereg, 2020-10-24:
@rojer saw the comment this issue originates from. This is actually normal behavior with the dns-sd tool on macOS. At least that is what I also observer on more or less crowded networks with **certified** HomeKit accessories.
Every hour (when mDNSResponder makes its query) sometimes all services get displayed as being removed and then added again a few milliseconds later. Don’t ask why, but doesn’t impact operation, unsure if this is related to this exact case.

What TTL does the mongoose mdns implementation use for the main PTR record (and TXT and SRV records). The required 75 minutes (4500 seconds) and 120 seconds for SRV?

rojer, 2020-10-24:
we use 120 for all records

Supereg, 2020-10-24:
120 seconds should only be used for records where the hostname is included (SRV, A, AAAA). All records other than that (in the context of dns-as) has 4500.
https://tools.ietf.org/html/rfc6762#section-10

Otherwise it should be 4500 seconds, especially the main _hap._tcp.local PTR record. This will result in the querier only sending one query all 60 minutes. Otherwise a querier would need to reduce its intervals to 1-2 minutes. Not every querier might do that and might loose the advertisement (I experienced that with mDNSResponder). Could be the problem here.

rojer, 2020-10-24:
updated the dns-sd library, PTR records now have 75 minute TTL

Supereg, 2020-11-09:
Additionally TXT records should use 75 minutes as well.

@rojer are you responsible for the dns-sd implementation? Scanned the code a bit and you might want to check out the [RFC 6762]()  in general as I think there is a bunch of things missing. Especially the section about [RFC 6762 7. Traffic Reduction](https://tools.ietf.org/html/rfc6762#section-7) (with focus on the **known-answer suppression**) is something you should checkout as its a must according to the spec.

gregDEweb, 2020-11-13:
I was out if order for a little while and now try to catch up with everything. I spend some time last weekend to assign all network devices an ip address and all the sudden there is no issue with shellys not responding in Bonjour. 
Maybe this will help others as well.

pivale, 2020-11-14:
> I was out of order for a little while and now try to catch up with everything. I spend some time last weekend to assign all network devices an ip address and all the sudden there is no issue with shellys not responding in Bonjour.
> Maybe this will help others as well.

You mean you've configured static ip addresses on your router?
That solved your issues?

andyblac, 2020-11-14:
Or try assigning an IP in your DHCP Server to each device, I do that here

gregDEweb, 2020-11-14:
@pivale 
yes i assigned a static ip for all devices in my network. not just shellys.

pivale, 2020-11-16:
I've added the suggestion to the trouble shooting guide. Thank you!

andyblac, 2021-01-04:
is this still an issue ?

gregDEweb, 2021-01-04:
Yes and no. In the last few days i explored issues with 2 devices which showed that they did not respond. So i checked the devices and they lost connection. As well all devices are connecting to the whorst possible access point. 
So yes it is still an issue but not a bonjour issue. Issue seems to be connection related. 
Even the shellys are right next to an access point they connect  to one far away and do not reconnect later on. All running 2.7

andyblac, 2021-03-03:
@gregDEweb this issue https://github.com/mongoose-os-apps/shelly-homekit/issues/129, i'll close this one but please reopen if you feel you need too.

closed 2021-03-03 by andyblac