mongoose-os-apps/shelly-homekit #109: Bulk Flash / Auto update

andyblac, 2020-08-05:
is there away to flash all Shelly's at once ?, As I get more and more Shelly's keeping them unto date is starting to take a some time.

So... I was wondering could you write a MacOS app to flash all shelly's at once ?, It could go and probe the network via bonjour to find shelly's, device type, current FW version, then do a bulk update.

@rojer is this something you could do?

rojer, 2020-08-05:
you can script it with `curl`.

```
curl -F file=@XXX.zip http://shelly-XXX.local/update
```

andyblac, 2020-08-05:
is there a curl to probe FW version ?, I m not bad linux scripting.

rojer, 2020-08-05:
```
$ curl -qsS http://shellyswitch25-B955B6.local/rpc/Sys.GetInfo | jq -r .fw_version
2.0.11
```

andyblac, 2020-08-05:
> $ curl -qsS http://shellyswitch25-B955B6.local/rpc/Sys.GetInfo | jq -r .fw_version
> 2.0.11

```
curl -qsS http://shellyswitch1-055FFF.local/rpc/Sys.GetInfo | jq -r .fw_version
-bash: jq: command not found
^C
```

rojer, 2020-08-05:
`brew install jq`, perhaps?

well, you don't have to use jq, it's just a convenient way to parse JSON in shell (https://stedolan.github.io/jq/).

andyblac, 2020-08-05:
@rojer is there anyway I can probe the device to confirm it is running your firmware, it just that when I probe my lan for shelly it is picking all of them up, including those on official firmware.

```
arp -a | grep shelly
shelly1-f3bee6.local (192.168.3.40) at xxxxxxxxxxxx
shelly1-f2a067.local (192.168.3.41) at xxxxxxxxxxxx
shelly1-f28318.local (192.168.3.42) at xxxxxxxxxxxx
shelly1-f28095.local (192.168.3.43) at xxxxxxxxxxxx
shellyswitch25-05ad13.local (192.168.3.44) at xxxxxxxxxxxx
shellydimmer-f3cd89.local (192.168.3.46) at xxxxxxxxxxxx
shelly1-055fff.local (192.168.3.47) at xxxxxxxxxxxx
shellyswitch25-05b323.local (192.168.3.48) at xxxxxxxxxxxx
shellyrgbw2-b075d3.local (192.168.3.49) at xxxxxxxxxxxx
shellyrgbw2-b0c8bb.local (192.168.3.50) at xxxxxxxxxxxx
shellyrgbw2-b09a29.local (192.168.3.51) at xxxxxxxxxxxx
shellyrgbw2-b0a3ef.local (192.168.3.52) at xxxxxxxxxxxx
```

rojer, 2020-08-05:
yeah, Sys.GetInfo is a generic mos rpc.
ok, we can use one specific to shelly-homekit:

```
$ curl -qsS http://shellyswitch25-B955B6.local/rpc/Shelly.GetInfo | jq -r .app,.version
switch25
2.0.11
```

ones not running shelly-homekit will return 404.

andyblac, 2020-08-05:
ok thanks. I have got most of required values now, current FW, vastest FW on GitHub, download urls from githhub, now I have this I'll start work on a script.

andyblac, 2020-08-05:
@rojer can you add "Model: xxxx" to firmware section in webUI, so I can use to pull the correct download URL from GitHub, I tried using "App: switch1" but the is no link between the 2.

rojer, 2020-08-05:
i can add it, for now you can have an app -> model map in the script

andyblac, 2020-08-05:
> i can add it, for now you can have an app -> model map in the script

any idea how, ever used dictionary's in BASH.

rojer, 2020-08-05:
bash has a `case` statement, google it.

andyblac, 2020-08-05:
> bash has a `case` statement, google it.

cheers,  sorted...

```
Host: shelly1-f3bee6.local
Type: switch1
Model: Shelly1
Current: 2.0.12
Latest: 2.0.12
DURL: https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.0.12/shelly-homekit-Shelly1.zip
```

andyblac, 2020-08-05:
@rojer something just happened `curl --silent "https://api.github.com/repos/mongoose-os-apps/shelly-homekit/releases/latest" | jq -r .tag_name` no longer working just stopped you changed something?

```
curl --silent "https://api.github.com/repos/mongoose-os-apps/shelly-homekit/releases/latest" | jq -r .tag_name
null
```

andyblac, 2020-08-05:
oh bugger

```
{"message":"API rate limit exceeded for xx.xxx.xxx.xxx. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)","documentation_url":"https://developer.github.com/v3/#rate-limiting"}
```

rojer, 2020-08-05:
right, you need to either take a break or add a token. meanwhile, i added model to GetInfo output - https://github.com/mongoose-os-apps/shelly-homekit/commit/8b0beaf4c910eef3f2ae4e2de1d49b17dab41e20

andyblac, 2020-08-05:
thx, I'm not stuck of how to compare the version number in bash, is it only seems to handle decimal or integer (xx or xx.xx) not xx.xx.xx

any ideas ?

rojer, 2020-08-05:
ugly... but possible. google yields https://github.com/cloudflare/semver_bash, which seems decent.

vanrri, 2020-08-05:
any chance you can add firmware to shelly Dimmer and Shelly Pro?

andyblac, 2020-08-05:
> ugly... but possible. google yields https://github.com/cloudflare/semver_bash, which seems decent.

thx but managed to code my own way.

andyblac, 2020-08-05:
@rojer ok here is the 1st beta of script, it dose not flash as line 31 has 

`				[Yy]* )  echo "Now Flashing.."; echo "curl -F file=shelly-flash.zip http://$device/update"; break;;`

to actually get it flash replace line 31 with

`				[Yy]* )  echo "Now Flashing.."; curl -F file=shelly-flash.zip http://$device/update; break;;`

let me know what you think, or if I have done something wrong.

[flash_shellys.sh.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5030316/flash_shellys.sh.zip)

rojer, 2020-08-05:
> [flash_shellys.sh.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5030316/flash_shellys.sh.zip)

hey, this is github! make a PR out of it :)

andyblac, 2020-08-05:
> > [flash_shellys.sh.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5030316/flash_shellys.sh.zip)
> 
> hey, this is github! make a PR out of it :)

yeah, I will, once you give it the once over, don't want to brick people devices incase I fucked something up 😄

rojer, 2020-08-05:
it's a pull _request_, i don't have to merge it straight away. PR only starts the conversation, i can add line comments, etc.

andyblac, 2020-08-05:
ok

andyblac, 2020-08-05:
@rojer done https://github.com/mongoose-os-apps/shelly-homekit/pull/110

I have enabled the actual flash process on line 31.

andyblac, 2020-08-27:
since this has now been added, so I will close it.

closed 2020-08-27 by andyblac

bglnelissen, 2024-07-04:
nice script, thanks. I had a device that was in my arp but not responding. To delete the device from my arp list I did the following:

    arp -a | grep -i shellyflood
    > shellyflood-244cab43079b.lan (10.0.0.97) at (incomplete) on en0 ifscope [ethernet]

mark the ip address (10.0.0.97) and delete that ip from the arp list

    sudo arp -d 10.0.0.97