mongoose-os-apps/shelly-homekit #111: cat flash shelly1 who has tasmota firmware

testpaul999, 2020-08-07:
Hi,
i have a new shelly with tasmota FW 8.4.0, have try with minimal and normal FW and get 
Shelly 1 Modul
Shelly 1
Upload fehlgeschlagen

Magic Byte ist nicht 0xE9

need help please

rojer, 2020-08-07:
you need first to go back to stock firmware, we don't support upgrades from tasmota.
@yaourdt (author of https://github.com/yaourdt/mgos-to-tasmota) mentioned thayt he had plans to build a bridge back (from tasmota to stock / mongoose os), i don't know if that is still planned.

closed 2020-08-07 by rojer

yaourdt, 2020-09-24:
It took some time and I don't know if it is still relevant for you @testpaul999 , but [tasmota-to-mgos](https://github.com/yaourdt/tasmota-to-mgos) should now offer a way back. However, be aware that the method is still quite untested and could put your device in a state that can only be recovered by attaching a wired programmer.