mongoose-os-apps/shelly-homekit #113: Server/File http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip not reachable

Olly68, 2020-08-08:
Hi,

First at all thank you for your great work.

I have updated 5 Shelly 1 the last weeks and they are working fine with the HomeKit Firmware:
App: switch1
Version: 2.0.7
Build ID: 20200723-232904/2.0.6-1-gd2f0795-master

Today I have implemented a new one with the following Shelly firmware:
The current Firmware version of your Shelly device is 20200601-122823/v1.7.0@d7961837 No newer firmware available.

I have tried everything to upgrade this Shelly with the HomeKit firmware:
http://X.X.X.X/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip
Instead of X.X.X.X the IP from the Shelly.

After several try’s, always the Shelly firmware appears.

Do you have any idea, what is wrong?

Thank you in advance for your help!

andyblac, 2020-08-08:
make sure that the shelly device has internet access, maybe check your DNS / firewall.

Olly68, 2020-08-08:
With the Shelly app is everything fine also over IP.

andyblac, 2020-08-08:
try manual update via webUI, download file from here https://github.com/mongoose-os-apps/shelly-homekit/releases

andyblac, 2020-08-08:
or could try url from GitHub https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.0.12/shelly-homekit-Shelly1.zip

Olly68, 2020-08-08:
Sorry, how can I make a manual update over webUI?
Is this complicated?

andyblac, 2020-08-08:
> Sorry, how can I make a manual update over webUI?
> Is this complicated?

are you still on official shelly firmware or are you on this firmware, nd just want to update to latest ?

Olly68, 2020-08-08:
Iàm still on the Shelly Firmware: 20200601-122823/v1.7.0@d7961837

Olly68, 2020-08-08:
I want to Update to the HomeKit firmware?

andyblac, 2020-08-08:
> I want to Update to the HomeKit firmware?

ok, it dose seem @rojer firmware link is down, so maybe try link from GitHub like

http://X.X.X.X/ota?url=https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.0.12/shelly-homekit-Shelly1.zip

Olly68, 2020-08-08:
Status in web browser: {"status":"updating","has_update":false,"new_version":"20200601-122823/v1.7.0@d7961837","old_version":"20200601-122823/v1.7.0@d7961837"}

Nothing is happen on the Shelly.
The Shelly is working with the Shelly Firmware.
I can’t understand why, it’s the first time that the firmware upgrade is not working.

andyblac, 2020-08-08:
> Status in web browser: {"status":"updating","has_update":false,"new_version":"20200601-122823/v1.7.0@d7961837","old_version":"20200601-122823/v1.7.0@d7961837"}
>
> Nothing is happen on the Shelly.
> The [...]

ok,  guess you'll have to wait for @rojer to fix his web server then.

Olly68, 2020-08-08:
Okay, thank you for your help.

TheRock89, 2020-08-08:
Hey Guys, 

i have the same problem as Olly68. I have also tried the different URLS, but I come to the same result as Olly68.

andyblac, 2020-08-08:
yeah, rojer host is down, and GitHub I guess uses redirects to AWS to host files, the why dose not work, it requires a direct link to file.

rojer, 2020-08-08:
oops, sorry, the server was down. it's back up now.

closed 2020-08-08 by rojer

Olly68, 2020-08-08:
Thanks ab lot for the fast help

MarcAroni81, 2025-01-06:
Hi @rojer I was finding that post in google as I also have the exact same issue. But firstly: have happy new year :) 

Shelly1 Switch is at firmware:
`The current Firmware version of your Shelly device is 20200601-122823/v1.7.0@d7961837 No newer firmware available.`

1. I tried several times to update to newest version via:

`http://x.x.x.x/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip`
With response:
`{"status":"updating","has_update":false,"new_version":"","old_version":"20200601-122823/v1.7.0@d7961837"}`

2. Even I tried to revert to stock Firmware via:

`http://192.168.0.90/ota?url=http://rojer.me/files/shelly/stock/latest/SHSW-1.zip`

Still same response:
`{"status":"updating","has_update":false,"new_version":"","old_version":"20200601-122823/v1.7.0@d7961837"}`

Any ideas left that I could follow along?

BR
Marcus

MarcAroni81, 2025-01-06:
Solved it myself. Problem was: Shelly switch was not able to connect to the internet. The cause for that was that the Shelly had "fixed IP address" enabled in the device menu. I deactivated at (still pinned the IP by my router) and now it worked!