mongoose-os-apps/shelly-homekit #114: Wifi signal

yuvalbeck, 2020-08-14:
Hi, 
Wifi signal strength status would be a nice feature in the web UI.
Thank you

andyblac, 2020-09-30:
@rojer while your doing update to webUI, I this some that could be added ?

rojer, 2020-10-02:
ok, this one didn't make the cut for 2.2, but it's easy to add - an extra field to the Shelly.GetInfo RPC, and a UI change.
if anyone feels like it, feel free to take this one up.

rojer, 2020-10-08:
added, will be in the next release

closed 2020-10-08 by rojer

andyblac, 2020-10-08:
one for @schemhad 😄

rojer, 2020-10-08:
yeah, for now i'm showing RSSI. i don't necessarily agree with signal strength calculation in the PR mentioned above, i'd like to know more about it.
i mean, i know > -50 is great, < -90 is really bad, but what is -70?

andyblac, 2020-10-08:
from Unifi UI

[image]

schemhad, 2020-10-08:
-30 -> -66 Very Good
-67 -> -70 Good
-71 -> -79 Okay
-80 -> -90 Bad

rojer, 2020-10-08:
@schemhad feel free to send a pull request!