mongoose-os-apps/shelly-homekit #115: Reverting to stock firmware

andyblac, 2020-08-14:
Shelly has updated the API in v1.8.0, as a result all the zip's are now giving 404 in the readme.

rojer, 2020-08-14:
@kzyapkov what are the new locations?

kzyapkov, 2020-08-15:
URLs for all devices are listed at

https://api.shelly.cloud/files/firmware

kikofhm, 2020-08-18:
how revert to stock firmware? 
the link <http://api.shelly.cloud/firmware/SHSW-1_build.zip>lead me for "404 Not Found" 
I can't revert firmware for original via interface (attach file)

andyblac, 2020-08-18:
> how revert SHELLY 1 to stock firmware?
> the link http://api.shelly.cloud/firmware/SHSW-1_build.ziplead me for "404 Not Found"
> I can't revert firmware for original via interface (attach file)

Links are in the above post but
Here is one for Shelly 1 http://api.shelly.cloud/firmware/SHSW-1.zip

pivale, 2020-08-19:
> URLs for all devices are listed at
> 
> https://api.shelly.cloud/files/firmware

Is it possible to add a field for the comercial name?
It will be easier for people not familiar with the product_id

rojer, 2020-08-19:
updated the links

closed 2020-08-19 by rojer

andyblac, 2020-08-27:
I have wrote a tool to do this automatically for you, you can find the script here https://github.com/mongoose-os-apps/shelly-homekit/blob/master/tools/flash_shelly.sh simply run `./flashscript.sh -m revert _hostname.local_`