mongoose-os-apps/shelly-homekit #121: Shelly 2.5 Switches in Separate Rooms

joe210599, 2020-08-31:
I have the Problem in putting the two switches in different Room. When I would like to put one Switch in Room A, the second one also goes to Room A. How can I fix that Issue?

Thanks in advance!

andyblac, 2020-08-31:
this is a duplicate ticket, add your support here https://github.com/mongoose-os-apps/shelly-homekit/issues/18

rojer, 2020-08-31:
yes, dup of #18

closed 2020-08-31 by rojer

yuvalbeck, 2020-09-19:
Thank you for that.
In case that you only have 1 switch connected you won’t be able to delete the other switch.
The way I over come it I groups the 2 switches together but a better way might be having 2 different switches totally

andyblac, 2020-09-19:
> Thank you for that.
> In case that you only have 1 switch connected you won’t be able to delete the other switch.
> The way I over come it I groups the 2 switches together but a better way might be having 2 different switches totally

You can now mimics switches between room on the Shelly 2.5 I. Ver 2.1.7

yuvalbeck, 2020-09-19:
> > Thank you for that.
> > In case that you only have 1 switch connected you won’t be able to delete the other switch.
> > The way I over come it I groups the 2 switches together but a better way might be having 2 different switches totally
> 
> You can now move switches between room on the Shelly 2.5 in Ver 2.1.7

Yes but not to delete unused switch completely (1 of the 2)

rojer, 2020-09-19:
> In case that you only have 1 switch connected you won’t be able to delete the other switch.

yes you can. just set HAP type of the unused switch to "disabled" and it will disappear from the home app.

andyblac, 2020-09-19:
> > In case that you only have 1 switch connected you won’t be able to delete the other switch.
> 
> yes you can. just set HAP type of the unused switch to "disabled" and it will disappear from the home app.

Beat me too it lol