mongoose-os-apps/shelly-homekit #124: Shelly Sensors support

Alex-Bricklayer, 2020-08-31:
First of all many thanks for the wonderful firmware. Is it possible to extend the software to the Shelly sensors? Especially the Shelly Door/Window 2 would be very useful.

Greetings 

Alex

rojer, 2020-08-31:
it's possible, but no ETA, as usual.

andyblac, 2020-09-29:
I'll close this as new ticket has been issued to track this.

rojer, 2020-09-29:
no, not quite. this is about support for hardware shelly door/window sensors.

andyblac, 2020-09-29:
ok, my apologises.

rojer, 2020-09-29:
https://shelly.cloud/products/shelly-door-window-2-smart-home-automation-sensor/

andyblac, 2020-09-29:
add `SHELLY MOTION` to the list once it's released then.

MilanBacaCz, 2020-10-02:
Hi all, if never mind I will try make this one [(https://shelly.cloud/products/shelly-door-window-2-smart-home-automation-sensor/).](https://shelly.cloud/products/shelly-door-window-2-smart-home-automation-sensor/) Next week I hope the sensor will arrive and I can start work on it.

Ludelag, 2020-10-16:
So, I didn't understood at all,will it be done? 

How could I help

andyblac, 2020-10-16:
it on the to do list.

MilanBacaCz, 2020-10-16:
Is here anyone who can prepare web UI for Door/Window 2 sensor?

andyblac, 2020-10-16:
> Is here anyone who can prepare web UI for Door/Window 2 sensor?

it needs more than just webUI, the device needs coding, as i said, it on the todo list, you will will just have to be patient.

MilanBacaCz, 2020-10-16:
i had the device at home and want to help with coding. I just now study the Rojer code, HomeKit and mongoose api.

Ludelag, 2020-10-22:
Is there any news redarding the implementation of the firmware for the D/W 2 ?

MilanBacaCz, 2020-10-22:
Sorry, it does not. I am sick :-( high fever, headaches. With pills from the doctor i am not able thinking. :-( I will be continue when i will be Ok.

S pozdravem,
Milan Bača

andyblac, 2020-10-25:
@MilanBacaCz get well soon.

Ludelag, 2020-10-28:
@MilanBacaCz hope you are doing better and recovered. Take care.

Ludelag, 2020-11-05:
Hello every body, any news here? @MilanBacaCz How are you doing? hope you feel better.

Ludelag, 2020-11-18:
Hello everyone, there is any news about this issue? thank you

BlancoDeMallloca, 2020-12-13:
Hi all,
Hope everyone is fine. 

Any update about the sensor support? 
Shelly door&window Lux measurement is a great feature for automation. 
It would be highly appreciable  if this could be implemented.

MilanBacaCz, 2020-12-13:
I am sorry, i have very few time. :-( There is no progres on my side.

S pozdravem,
Milan Bača

jsiegenthaler, 2020-12-27:
Hello All. I now have 2 x DW2 myself, and am happy to help in any way that I can.... but that might be limited to Web GUI and Readme updates... however I can code and can test and am willing to help...

louis49, 2021-03-24:
1073 / 5000
Résultats de traduction
I am completely junior on this type of development, I have a Shelly door window 2, but I started to:
 * Implement the simplest firmware part for the contact sensor part for which a library had already been coded (I have not yet looked into the luxmeter)
* Successfully compiled but not yet tested a flash.
=> I will see in the next few days to test the result of my work when I will have succeeded in connecting it to the serial port (not very standard connection)

I face two questions:
1: How did you go about determining the PIN IDs? (I plan to determine them empirically)
2: From what I could understand the original firmware has a particular battery management (the sensor cuts off when not in use): if this new firmware reduces the battery life from 2 years to 2 weeks I'm not sure users will like it. (There is this type of solution to work around the problem: https://stefan-rehnberg.com/almost-always-on-shelly-dw2/ (not tried))

ma-jkl, 2021-04-06:
Hi, ano progress guys? Thank you, Michal

Ludelag, 2021-04-07:
I use it thru Shelly Homebridge plugin and works pretty good, But will be nice to have it shelly homekit firmware

ma-jkl, 2021-04-12:
> I use it thru Shelly Homebridge plugin and works pretty good, But will be nice to have it shelly homekit firmware

Thank you Ludelag. Will try to do that.

ma-jkl, 2021-12-11:
Hi,
time flies. Is there any Progress? Regards

timoschilling, 2023-02-21:
Replaced by https://github.com/mongoose-os-apps/shelly-homekit/issues/1233

closed 2023-02-21 by timoschilling