mongoose-os-apps/shelly-homekit #127: HomeKit QR setup code

yuvalbeck, 2020-09-03:
It would be great to have auto generated HomeKit codes and QR codes in particular- not having to setup it manually.

rojer, 2020-09-03:
good idea

schemhad, 2020-09-03:
That shouldn't be complicated... I'll take a look at that when I got some time to...

schemhad, 2020-09-04:
@rojer
Is there already some kind of Setup-PIN like e.g. "ABCD"? If not, we have to define it...

The rest looks like Devicetype (like e.g. "5" for a lightbulb), Setup-Code ("111-22-333"), Version "0", Revised "0", Flag "8" (Wireless IP)

So: "511122333008" - to Base36: "6IT11G7K" + Setup-PIN ("ABCD")

URL: "X-HM://6IT11G7KABCD"

This, generated to an QR-Code should/could work ...

rojer, 2020-09-04:
thing, is we don't store plain-text pin on the device, this is explicitly prohibited by the standard.
does the qr code contain pin in plain text?

rojer, 2020-09-05:
the only point when we have code in plaintext is when users enters it into the text field. we can generate and display qr code at that point.

schemhad, 2020-09-07:
I didn't got the dev-docs atm, but ist looks like plain text Setup-PIN to me... maybe we could open an popup window or overlay while setting the Setup-Code with a warning, that the generated QR code will only displayed this one time... otherwise it would be a serious security problem.

rojer, 2020-09-07:
yeah, we can do that. user can save or print the code for later use if they like.

sebastianheise, 2020-09-07:
It could just be displayed on the setup website. (Unless there isn't enough memory on the device...?)

rojer, 2020-09-07:
yes, that's the idea. basically, when you press "save" in the homekit section of the configuration and we have plaintext code value, we should pop up a window with QR code. code can be generated in javascript, so it won't even use much space (only some JS code).

schemhad, 2020-09-09:
But is there already a default (Setup-)PIN / Password like "ABCD"? 
To secure it, we should generate a random four character PIN at this point, but we have to give it to the OS...

rojer, 2020-09-10:
> But is there already a default (Setup-)PIN / Password like "ABCD"?

no, there isn't. user has to assign setup pin after flashing. they enter it as plaintext, it is sent to the device as plaintext but is not stored directly, and is instead converted into cryptographic material used for HAP pairing.

cduvenhorst, 2020-10-11:
This needs be implemented in homekit-adk PAL "HAPPlatformAccessorySetupDisplay.c".

The setup payload will then dynamically change every 5 minutes or so if unpaired. Precondition is a persistent setupID, which may be set in the config or could be generated if the config setting is missing.

rojer, 2020-10-12:
hm, interesting. @cduvenhorst any chance you could put that in a form of a PR? :)

cduvenhorst, 2020-10-12:
Sure. But need some more time ...  ;-) The public HAP documentation does not describe R14 devices. Currently homekit-adk acts like a pre r10 device. (HAPAccessorySetupInfo.c)

Had coded most QR stuff in the app, but I feel it belongs in the lib.

Need to implement setupDisplay and setupNFC and understand HAPAccessorySetupInfo-functions ... will devide  the whole thing in some PRs.

jsiegenthaler, 2020-10-16:
Why not calculate a pin or setup code or QR code from the device MAC address? The MAC address is currently displayed as the Homekit accessory serial number.  That way you have a unique device-specific code.

Pixel-Chris, 2020-10-22:
My idea:
Add an extra Button:
„Generate random code“
After that the QR-Code and plain-text (the Code) is displayed....

Eventually this could help for the QR-Code generating:
https://developers.google.com/chart/infographics/docs/qr_codes

cduvenhorst, 2020-10-22:
> My idea:
> 
> Add an extra Button:
> 
> „Generate random code“
> 
> After that the QR-Code and plain-text (the Code) is displayed....
> 
> 
> 
> Eventually this could help for the QR-Code generating:
> 
> https://developers.google.com/chart/infographics/docs/qr_codes

That's the app side. The homekit-adk should implement an rpc service for setup information where the current payload and code is shown. Could even include an svg element for the qr code.

andyblac, 2020-10-25:
i come across this https://www.studiopieters.nl/esp-homekit-sdk-qr-code/ if it helps

rojer, 2020-10-25:
we need a JS library to render QR codes

andyblac, 2020-10-25:
> we need a JS library to render QR codes

like this https://davidshimjs.github.io/qrcodejs/ ?

cduvenhorst, 2020-10-26:
> we need a JS library to render QR codes

I used https://github.com/nayuki/QR-Code-generator ...

rojer, 2021-06-03:
alright, this is done and seems to be working. will test in the next release cycle.

rojer, 2021-06-03:
[image]

closed 2021-06-03 by rojer

andyblac, 2021-06-03:
nice one 👍