mongoose-os-apps/shelly-homekit #128: I've tried different versions of software, resets, HomeKit doesn't show it.

michaelnozbe, 2020-09-05:
Hello, after hours on this, I'm giving up. I have a Shelly 2.5 device.

I managed to flash it to 2.0.12 version of HomeKit. I've assigned it a static IP address.

Web interface works, both switches work when I click on them.

When I set up HomeKit, I put code 00000001 which got accepted.

It says: paired:NO, provisioned:YES, Connections: 0/0/9 pending/active/max

But when I open the HomeKit app and tap to add device, it won't show.

Here's what I've tried:

- I've reset the homeKit connection several times, I've set up different codes, I've rebooted the device several times via the web interface.
- I've also uploaded different firmware versions, tried 2.0.3 as the oldest one. Works on web, but my HomeKit setup won't discover the new device.

I'm giving up for now. I've spent hours trying to get it to work with no avail.

Any way I can debug it for you or help you figure out why in my case the device won't show up in the list of available devices to add?

Any help or guidance would be appreciated. I've tried to read all the past issues to figure it out but no avail.

andyblac, 2020-09-05:
the HomeKit setup code is too basic, HomeKit it self will not allow it, please try a more random code

michaelnozbe, 2020-09-06:
It wasn't that. For some reason it just didn't work. I left it there. Waited a day and after a day it showed up, it got added and it's now in my homekit. Dunno why after a day it just started to work.

closed 2020-09-06 by michaelnozbe