mongoose-os-apps/shelly-homekit #129: HAP HTTP server drops connections

rojer, 2020-09-06:
Reported by @andyblac in https://github.com/mongoose-os-apps/shelly-homekit/issues/30#issuecomment-687673015, confirmed by other users.
Apparently, after a while (~30 days) the device becomes unavailable in the Home app.
 * still available on port 80, web ui works fine
 * RPC works fine
 * free ram is normal, so not a memory leak
 * HAP connection counts are all at 0
HAP HTTP server drops connection without sending a response - https://github.com/mongoose-os-apps/shelly-homekit/issues/30#issuecomment-687847662

pivale, 2020-09-09:
Hey, 

Having the same problem:

0/0/9 pending/active/max 
Uptime: 25:16:33:46

Noticed just because automation on babycam stopped working.

Anything other information I can provide before rebooting?

andyblac, 2020-09-09:
nope, if you want to load the debug version, so if you get it again, it will allow @rojer to login to the device to see why it is happening download from below

https://rojer.me/files/shelly/2.0.12-dbg/

pivale, 2020-09-09:
> nope, if you want to load the debug version, so if you get it again, it will allow @rojer to login to the device to see why it is happening download from below
> 
> https://rojer.me/files/shelly/2.0.12-dbg/

2 quick questions, as I would like to help:
- what do you mean connecting to the device? as an ssh connection there? or just having more info I could then provide?
- when changing/loading/updating the firmware, does the connection with homekit needs to be redone and all other configurations?

Thank you!

andyblac, 2020-09-09:
> 2 quick questions, as I would like to help:
>
> * what do you mean connecting to the device? as an ssh connection there? or just having more info I could then provide?
> * when changing/loading/updating [...]

you'll have to ask @rojer about question 1, ut as for Q2 no just flash it, all settings are saved.

rojer, 2020-09-10:
i ran a "connection stress test" overnight, with about 25000 connection attempts total - everything worked as expected.

> what do you mean connecting to the device? as an ssh connection there? or just having more info I could then provide?

debug firmware sends serial console logs to my server and opens an MQTT connection back to my server, so i can send RPC requests to the device remotely, to reconfigure it and gather debug information. it also gives me ability to reflash the device remotely, so pretty much full control over a device on your network. i totally understand if you are not willing to provide that kind of access, but it's really invaluable in terms of debugging and being able to figure out what's wrong directly.

rojer, 2020-09-10:
i have one suspicion: just over 24 days is the overflow value for a signed 32-bit counter of milliseconds. i need to check if we use 32-bit uptime somewhere in HAP code.

andyblac, 2020-09-10:
@rojer only had 4 devices do this atm, other devices still going strong, if it helps, it seems to be devices that are used in HomeKit automations either as a trigger or an action.

ie my bathroom SW2 is a fan that is controlled via an automation from a Humidity Sensor if humidity gets over 75% it kicks in fan and turns off once it drops below 65%

andyblac, 2020-09-10:
> i have one suspicion: just over 24 days is the overflow value for a signed 32-bit counter of milliseconds. i need to check if we use 32-bit uptime somewhere in HAP code.

hmm, now you mention counter, could it be the auto off feature ? these same devices are using that too ?

billcalle, 2020-09-10:
Both of mine died and I do not use auto off or automation. I think that is a red herring.

rebuss0507, 2020-09-17:
Hi, just wanted to confirm the exact same issue for me. I was also having the issue reported in #30 (comment) and where upgrading. All of my Shelly actually did a couple of days ago already (but not exactly sure when). Rebooted them now after being online for 29 days, now they are all working again......

Except two where one of it is indeed included in an automation.....But I will check on these extra separately

andyblac, 2020-10-02:
@rojer on day 25, so we should be seeing something soon

andyblac, 2020-10-02:
@rojer it happened again, can you check logs ?? `shellyswitch25-8CAAB505AD13` was ok this morning but now 'no response` in HomeKit

andyblac, 2020-10-02:
every now and then log shows 1 pending but it does not get a new connection.

rojer, 2020-10-02:
```
[Oct  2 14:47:29.773] shellyswitch25-8CAAB505AD13 497165 1601650049.912 2|shelly_main.cpp:276     Uptime: 2236751.25, conns 0/3/9, RAM: 41200, 24664 free                                                                                                                                             
[Oct  2 14:47:30.389] shellyswitch25-8CAAB505AD13 497166 1601650050.472 2|HAPPlatformTCPStrea:120 0x3fff4834 192.168.3.24:50151 Incoming HAP connection, ns 0/3/9                                                                                                                                     
[Oct  2 14:47:30.389] shellyswitch25-8CAAB505AD13 497167 1601650050.482 2|HAPPlatformTCPStrea:274 0x3fff4834 192.168.3.24:50151 Accepted HAP connection, ns 0/4/9 ts 0x3fff68ec                                                                                                                       
[Oct  2 14:47:30.389] shellyswitch25-8CAAB505AD13 497168 1601650050.502 2|HAPPlatformLog.c:65     IPAccessoryServer Failed to allocate session. (Number of supported accessory server sessions should be consistent with the maximum number of concurrent streams supported by TCP stream manager.)   
```

hm, that's odd. something is off with session management. i'll need to take a closer look

andyblac, 2020-10-02:
you want me to leave it in this state for now ?

rojer, 2020-10-02:
there isn't much use, unfortunately, as core dump functionality is broken.
i'll review the relevant code, for now feel free to reboot and upgrade this device.

rojer, 2020-10-02:
clearly HAP sessions get out of sync with TCP connections. i'll add a bunch of debug output around that code.

andyblac, 2020-10-02:
if you want me test another firmware thats no problem, but it will take another 25 or so till it happens again.

andyblac, 2020-10-02:
hmm rebooted, and still showing no response.

andyblac, 2020-10-02:
see anything ?

andyblac, 2020-10-02:
ahh it back now, never mind.

andyblac, 2020-10-02:
let me know when this is built, I'll update and enable debug logging, will that open port for you ?

rojer, 2020-10-02:
it's out with 2.2.0, please install it on the device.
it will provide additional information i can correlate with logs to figure out when the discrepancy occurs.
i should be able to notice it earlier, once first "phantom" session appears instead of when we run out of session slots.
this will manifest as number of sessions getting out of sync with number of active connections.

rojer, 2020-10-02:
...aaaand it went silent. install the 2.2.0-debug build please - http://rojer.me/files/shelly/2.2.0-dbg/

andyblac, 2020-10-02:
> ...aaaand it went silent. install the 2.2.0-debug build please - http://rojer.me/files/shelly/2.2.0-dbg/

done.

rojer, 2020-10-02:
yep, it's back. ok, will keep an eye on it.

andyblac, 2020-10-02:
cheers.

andyblac, 2021-02-23:
started getting this again on 3 devices 30 days online.

[shelly.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6031402/shelly.log)

andyblac, 2021-02-23:
how can i do core dump for you.

freez1001, 2021-02-28:
Hi
Have the same issue with version 2.7.3.
Would say after 26 days Shelly 1 devices are show as not responding in HomeKit

jsiegenthaler, 2021-02-28:
I found last week my 2 x Shelly1s with 2.7.3 loaded had stopped responding after 28 days
Web interface worked, could interrogate device and display settings and control the device
Physical switch connected to SW input worked
But HomeKit reported no response from the accessory
After a reboot, the accessory started responding again in HomeKit

rojer, 2021-02-28:
HAP server gets stuck after 25-28 days, the reason is not yet clear and being investigated

chstoll, 2021-03-08:
Is there a way to just remotly restart all shelly devices in the network (maybe with the update-script)? I have about 40 shellys in my home network and it is somewhat painfull to connect to each webinterface an hit restart if they get unresponsive in Homekit.

andyblac, 2021-03-08:
i could add a reboot option to script, but we prefer to find why it doing it, atm in 2.8.0 alpha we are trying to track down some network issues.

rojer, 2021-03-08:
i think it makes sense as a script action. we do have to use it sometimes, unfortunately.

andyblac, 2021-03-08:
> i think it makes sense as a script action. we do have to use it sometimes, unfortunately.

ok i'll sort it, `--reboot` ok ?

rojer, 2021-03-08:
SGTM

andyblac, 2021-03-08:
ok here a beta version @chstoll @GHanthorn please test, report any issues, if all good i'll create a pull request.
[flash-shelly-reboot.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6100955/flash-shelly-reboot.zip)

freez1001, 2021-03-08:
...a reboot option via script is useful tool. THX

chstoll, 2021-03-08:
Thanks for the quick response. Works perfect, rebootet all my 47 shellys at once as well as single selected ones.

andyblac, 2021-03-08:
ok cheers guy's i'll do a pull request.

jsiegenthaler, 2021-04-10:
I also have noted a connection issue which looks like a HAP server stuck issue

Shelly1 with v2.8.2, used in Switch mode, Toggle, Inverted input
Uptime 25:19:58:xx and counting
Web GUI responds normally
Physical switch controls normally

When switching to the room with the Shelly1, the Home app shows **Updating** and then **No Response**

During this period the log shows:

```
2232004849323 HAPPlatformTCPStrea:121 0x3fff6f2c 192.168.0.153:52892 Incoming HAP connection, ns 0/0/12
2232005981349 HAPPlatformTCPStrea:121 0x3fff6304 192.168.0.11:61107 Incoming HAP connection, ns 1/0/12
2232006731272 HAPPlatformTCPStrea:276 0x3fff6f2c 192.168.0.153:52892 Accepted HAP connection, ns 1/1/12 ts 0x3fff617c
2232006739403 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff617c nc 0x3fff6f2c 1/0/12
2232006749341 HAPPlatformTCPStrea:105 0x3fff6f2c 192.168.0.153:52892 HAP connection closed, ns 1/0/12 i 2
2232008312491 HAPPlatformTCPStrea:276 0x3fff6304 192.168.0.11:61107 Accepted HAP connection, ns 0/1/12 ts 0x3fff34dc
2232008320585 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff34dc nc 0x3fff6304 0/0/12
2232008331954 HAPPlatformTCPStrea:105 0x3fff6304 192.168.0.11:61107 HAP connection closed, ns 0/0/12 i 2
2232008363855 HAPPlatformTCPStrea:121 0x3fff70f4 192.168.0.11:61108 Incoming HAP connection, ns 0/0/12
2232010006467 HAPPlatformTCPStrea:276 0x3fff70f4 192.168.0.11:61108 Accepted HAP connection, ns 0/1/12 ts 0x3fff617c
2232010014599 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff617c nc 0x3fff70f4 0/0/12
2232010026349 HAPPlatformTCPStrea:105 0x3fff70f4 192.168.0.11:61108 HAP connection closed, ns 0/0/12 i 2
2232010732981 shelly_main.cpp:560     Up 2232010.72, HAP 0/0/12 ns 0, RAM: 20460/38552; st 0; 0.1: st:0 in_st:0 inm:1 ininv:1
2232011514750 HAPPlatformTCPStrea:121 0x3fff6f2c 192.168.0.10:60703 Incoming HAP connection, ns 0/0/12
2232011524671 HAPPlatformTCPStrea:276 0x3fff6f2c 192.168.0.10:60703 Accepted HAP connection, ns 0/1/12 ts 0x3fff29a4
2232011532603 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff29a4 nc 0x3fff6f2c 0/0/12
2232011542660 HAPPlatformTCPStrea:105 0x3fff6f2c 192.168.0.10:60703 HAP connection closed, ns 0/0/12 i 2
2232012577124 HAPPlatformTCPStrea:121 0x3fff6f2c 192.168.0.10:60704 Incoming HAP connection, ns 0/0/12
2232013730582 HAPPlatformTCPStrea:276 0x3fff6f2c 192.168.0.10:60704 Accepted HAP connection, ns 0/1/12 ts 0x3fff29a4
2232013738707 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff29a4 nc 0x3fff6f2c 0/0/12
2232013749719 HAPPlatformTCPStrea:105 0x3fff6f2c 192.168.0.10:60704 HAP connection closed, ns 0/0/12 i 2
2232018733231 shelly_main.cpp:560     Up 2232018.72, HAP 0/0/12 ns 0, RAM: 20824/38552; st 0; 0.1: st:0 in_st:0 inm:1 ininv:1
2232022008758 esp_main.c:137          SDK: ip:192.168.0.195,mask:255.255.255.0,gw:192.168.0.1

```

I will not reboot the device yet in case you need more info.... but I might have to reboot this afternoon as I need to do repairs on my bathroom lights anyway....

andyblac, 2021-04-10:
> I also have noted a connection issue which looks like a HAP server stuck issue
> 
> Shelly1 with v2.8.2, used in Switch mode, Toggle, Inverted input
> Uptime 25:19:58:xx and counting
> Web GUI responds normally
> Physical switch controls normally
> RSSI is -52, device is located 2m from Wifi access point.
> 
> When switching to the room with the Shelly1, the Home app shows **Updating** and then **No Response**
> 
> During this period the log shows:
> 
> I will not reboot the device yet in case you need more info.... but I might have to reboot this afternoon as I need to do repairs on my bathroom lights anyway....

we need a core dump, can do the following:
1) open debug log tab
2) start an update or try to reflash, you should see "stopping service" in the log, but update won't be starting (as server is not able to stop).
3) abort the update, close flash tab (because of the workaround the update will begin anyway after 10 seconds, we don't want that)
4) take a snapshot of `http://xxx.xxx.xxx.xxx/debug/info` in new tab
5) invoke `http://xxx.xxx.xxx.xxx/rpc/Shelly.Abort` in new tab, it will not load anything
6) close this tab, do not wait for it to load something
6) save /debug/log from the tab
7) when device comes back, go to `http://xxx.xxx.xxx.xxx/debug/core` save core dump and send it to me, along with the log and debug info tab contents

andyblac, 2021-04-10:
> I also have noted a connection issue which looks like a HAP server stuck issue
>
> Shelly1 with v2.8.2, used in Switch mode, Toggle, Inverted input
> Uptime 25:19:58:xx and counting
> Web GUI responds [...]

> I will not reboot the device yet in case you need more info.... but I might have to reboot this afternoon as I need to do repairs on my bathroom lights anyway....

we need a core dump, please do the following:
1) open debug log in new tab
2) start an update / reflash, you should see "stopping service" in the log, but update won't start (as server is not able to stop).
3) abort the update by closing the update / flash tab, (because of the workaround the update will begin anyway after 10 seconds, we don't want that), **DO NOT ABORT THE UPDATE / FLASH IF ONE IS ACUTALLY RUNNING. THIS WILL CAUSE YOUR DEVICE TO BE REQUIRED TO REFLASH VIA SERIAL LEAD** 
4) take a snapshot of `http://xxx.xxx.xxx.xxx/debug/info`
5) invoke `http://xxx.xxx.xxx.xxx/rpc/Shelly.Abort` in a new tab
6) close this tab, do not wait for the page to load something.
7) save `http://xxx.xxx.xxx.xxx/debug/log` from the tab
8) when device comes back, save core dump by doing `http://xxx.xxx.xxx.xxx/debug/core`save and send it to @rojer , along with the log and debug info tab contents.

jsiegenthaler, 2021-04-10:
Ooops... not sure if I did abort it in time, maybe not, but I got lucky.Device is still running, still has v2.8.2 installed.And after the process, it still shows No Response in the Home app. I won’t take the mains power off yet... or try and reboot it yet... I’ll await your response Here’s the data CoreDump and log file attached  Snapshop of debug/info at the starthttp://192.168.0.195/debug/info App: switch1 2.8.2 20210315-093656/2.8.2-gc0f7426-masterUptime: 2242265.00RAM: 20124 free, 9032 min freeHAP server port: 9000HAP config number: 20HAP connection stats: 0/0/12HAP connections:Total: 0HAP sessions:Total: 0 Snapshop of debug/info after the abort attempt, looks like the Shelly1 restartedhttp://192.168.0.195/debug/info App: switch1 2.8.2 20210315-093656/2.8.2-gc0f7426-masterUptime: 311.59RAM: 22364 free, 20160 min freeHAP server port: 9000HAP config number: 20HAP connection stats: 0/0/12HAP connections: Total: 0HAP sessions: Total: 0  Jochen   Sent from Mail for Windows 10 From: Andy BlackburnSent: Saturday, 10 April 2021 13:05To: mongoose-os-apps/shelly-homekitCc: jsiegenthaler; CommentSubject: Re: [mongoose-os-apps/shelly-homekit] HAP HTTP server drops connections (#129) I also have noted a connection issue which looks like a HAP server stuck issueShelly1 with v2.8.2, used in Switch mode, Toggle, Inverted inputUptime 25:19:58:xx and countingWeb GUI responds normallyPhysical switch controls normallyRSSI is -52, device is located 2m from Wifi access point.When switching to the room with the Shelly1, the Home app shows Updating and then No ResponseI will not reboot the device yet in case you need more info.... but I might have to reboot this afternoon as I need to do repairs on my bathroom lights anyway....we need a core dump, please do the following:open debug log in new tabstart an update / reflash, you should see "stopping service" in the log, but update won't start (as server is not able to stop).abort the update by closing the update / flash tab, (because of the workaround the update will begin anyway after 10 seconds, we don't want that), DO NOT ABORT THE UPDATE / FLASH IF ONE IS ACUTALLY RUNNING. THIS WILL CAUSE YOUR DEVICE TO BE REQUIRED TO REFLASH VIA SERIAL LEADtake a snapshot of http://xxx.xxx.xxx.xxx/debug/infoinvoke http://xxx.xxx.xxx.xxx/rpc/Shelly.Abort in a new tabclose this tab, do not wait for the page to load something.save http://xxx.xxx.xxx.xxx/debug/log from the tabwhen device comes back, save core dump by doing http://xxx.xxx.xxx.xxx/debug/coresave and send it to @rojer , along with the log and debug info tab contents.

andyblac, 2021-04-10:
please posts as files.

jsiegenthaler, 2021-04-10:
Done. Github doesn't handle email responses from Windows Mail very well :(

andyblac, 2021-04-10:
unfortunately this is no good, you did not close the flash tab quick enough
```
2242180530860 shelly_main.cpp:789     Server failed to stop, proceeding with the update anyway
2242180535555 shelly_main.cpp:797     Starting firmware update
2242180547438 esp_ota_backend.c:185   Slot 0, FW: switch1.bin -> 0x8000, FS fs.bin -> 0xbb000
```

jsiegenthaler, 2021-04-10:
Happy to try again, but I was unclear how to abort the flash

I tried to reflash by downgrading the Shelly1 to 2.8.1. I did this by uploading the 2.8.1 zip file
once that starts I see no way to stop it.

You mention "abort the update by closing the update / flash tab, " but I see no update tab...??

jsiegenthaler, 2021-04-10:
And anyway, it did not load v2.8.1, it stayed at v2.8.2....

andyblac, 2021-04-10:
> Happy to try again, but I was unclear how to abort the flash
> 
> I tried to reflash by downgrading the Shelly1 to 2.8.1. I did this by uploading the 2.8.1 zip file
> once that starts I see no way to stop it.
> 
> You mention "abort the update by closing the update / flash tab, " but I see no update tab...??

close the browser tab

andyblac, 2021-04-10:
> Happy to try again, but I was unclear how to abort the flash
> 
> I tried to reflash by downgrading the Shelly1 to 2.8.1. I did this by uploading the 2.8.1 zip file
> once that starts I see no way to stop it.
> 
> You mention "abort the update by closing the update / flash tab, " but I see no update tab...??
if the device has restarted you can;t try again, as homekit should be reponding again.

jsiegenthaler, 2021-04-10:
Sorry, I lost the opportunity. Looks like the reboot fixed it.

Device is now responsive again, and debug/info now shows HAP sessions:

http://192.168.0.195/debug/info

App: switch1 2.8.2 20210315-093656/2.8.2-gc0f7426-master
Uptime: 1880.74
RAM: 23148 free, 16064 min free
HAP server port: 9000
HAP config number: 20
HAP connection stats: 0/3/12
HAP connections:
  192.168.0.181:62822 nc 0x3fff6a54f 0x200000 io 268 ts 0x3fff605c rd 267
  192.168.0.12:61051 nc 0x3fff55b4f 0x200000 io 349 ts 0x3fff6084 rd 348
  192.168.0.153:53096 nc 0x3fff5fa4f 0x200000 io 1086 ts 0x3fff6394 rd 1085
 Total: 3
HAP sessions:
  0: s 0x3ffef4c0 ts 0x3fff6394 o 1 st 1 ts 795204
  1: s 0x3ffef800 ts 0x3fff605c o 1 st 1 ts 1613327
  3: s 0x3ffefe80 ts 0x3fff6084 o 1 st 1 ts 1532240
 Total: 3

andyblac, 2021-04-10:
no worries i did as well on my 1st tempt 😄 , i'm now waiting for another device to stall.

jsiegenthaler, 2021-04-10:
My other Shelly1 with Mongoose OS has been up for 11 days. In two weeks time it'll hit 25 days and I'll expect it to become non-responsive... and will try again then

pivale, 2021-04-10:
Hey, 

I think I got it...
Can you check @rojer?

[no_hap_connections_logs+dump.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6290829/no_hap_connections_logs%2Bdump.zip)

andyblac, 2021-04-10:
> Hey,
> 
> I think I got it...
> Can you check @rojer?
> 
> [no_hap_connections_logs+dump.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6290829/no_hap_connections_logs%2Bdump.zip)

unfortunately no, you did not close flash tab soon enough 

```
2282420026801 mgos_ota_core.c:587     begin: waiting for user
2282420037938 shelly_main.cpp:789     Server failed to stop, proceeding with the update anyway
2282420042754 shelly_main.cpp:797     Starting firmware update
```
the server was force stoped, we need core dump BEFORE the server gets forced stopped.

next time as soon as you click upload, close the tab.

rojer, 2021-04-10:
> DO NOT ABORT THE UPDATE / FLASH IF ONE IS ACUTALLY RUNNING. THIS WILL CAUSE YOUR DEVICE TO BE REQUIRED TO REFLASH VIA SERIAL LEAD

this is not correct, you can abort an update at any point and it won't brick the device. during update image is written to an alternative location and boot config is modified only if the image has been fully written and verified.

pivale, 2021-04-10:
> > Hey,
> > I think I got it...
> > Can you check @rojer?
> > [no_hap_connections_logs+dump.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6290829/no_hap_connections_logs%2Bdump.zip)
> 
> unfortunately no, you did not close flash tab soon enough
> 
> ```
> 2282420026801 mgos_ota_core.c:587     begin: waiting for user
> 2282420037938 shelly_main.cpp:789     Server failed to stop, proceeding with the update anyway
> 2282420042754 shelly_main.cpp:797     Starting firmware update
> ```
> 
> the server was force stoped, we need core dump BEFORE the server gets forced stopped.
> 
> next time as soon as you click upload, close the tab., then proceed to do core dump.

It took me only the time to get to the X on the tab to close it after the upload... If that is not fast enough, not sure how to do it (keyboard shortcut might do it, but I suspect it will result in the same problem).

And it didn't update, I'm still on the same version...

This is really odd...

andyblac, 2021-04-10:
> this is not correct, you can abort an update at any point and it won't brick the device. during update image is written to an alternative location and boot config is modified only if the image has [...]

well it did mind using script, i did CTRL-C as flashing, left the device unbootable.

[3 more replies]

closed 2021-04-10 by rojer