mongoose-os-apps/shelly-homekit #132: [Flash Script] features....

andyblac, 2020-09-14:
Hi guys,

For those of you that use my script would it be beneficial to add an option to flash a specific version, so you can move between versions when @rojer is releasing test versions ?

If  there is anything you would like to see please comment below.

andyblac, 2020-09-14:
@rojer can you tweak the model id in the ShelllyInfo so it matches the Firmware file names ie

Model: Shelly2.5 to Model: Shelly25

andyblac, 2020-09-14:
@rojer never mind I work around it.

andyblac, 2020-09-14:
https://github.com/mongoose-os-apps/shelly-homekit/pull/134

andyblac, 2020-09-18:
@rojer Thinking of adding some text formatting, what you think Y/N ?

[image]

rojer, 2020-09-19:
flashy flasher :)
yeah, why not

andyblac, 2020-09-20:
what you think ?
[image]

rojer, 2020-09-20:
looks ok so far

andyblac, 2020-09-20:
@rojer thinking of adding a exclude option to the script ie `./flash_script -ae hostname` is -e ok with you ?

this would basically skip the referenced hostname when using -a. 

ie atm my shelly25 is in debug mode for you to keep an eye one with the HAP drop issue, so I don't want it to update it's fw.

rojer, 2020-09-20:
SGTM

rojer, 2020-09-20:
i think you can also remove DURL from output, for most users it's just noise.

andyblac, 2020-09-20:
> i think you can also remove DURL from output, for most users it's just noise.

ye I have in the formatting commit  :smile:

andyblac, 2020-09-20:
just working on -e, will push both soon

andyblac, 2020-09-21:
To much ?

[image]

rojer, 2020-09-21:
yeah, i'd say it's a bit much.
highlighting just the version numbers / status is enough, i think.
final result can be in default color

andyblac, 2020-09-21:
ok np

andyblac, 2020-09-21:
exclude and text formatting PR done https://github.com/mongoose-os-apps/shelly-homekit/pull/143.

andyblac, 2020-09-21:
what do you think ?, until we can name devices in webUI, it should help identify the device. I would prefer to be able to edit device name.

[image]

andyblac, 2020-09-28:
@rojer been thinking about all these betas, would be possible when you release beta's to release it on your server in a beta folder, then I could add a -b flag to script for 'betas' it could search if there is a beta to test?

ie
`http://rojer.me/files/shelly/beta/$lfw/shelly-homekit-$model.zip`

andyblac, 2020-09-28:
and maybe a txt file in the root beta folder with release tag info.

ie
`http://rojer.me/files/shelly/beta/release.info`

rojer, 2020-09-28:
yeah, i've been thinking about it myself.
i've been putting some groundwork for it, maybe will get to it for the next release cycle (i.e. after 2.2)

andyblac, 2020-10-08:
we now have python script, it should work on all OS's requires python 3.6

closed 2020-10-22 by andyblac