mongoose-os-apps/shelly-homekit #133: HomeKit Connection stuck at "0/4/9 pending/active/max" in build 2.1.0 and 2.1.2

vladbisceanu, 2020-09-14:
@rojer @andyblac Opening this issue as a follow-up to the discussion on Issue #9 .

With the new 2.1.0 - 2.1.2 firmware, HomeKit connections seems to be stuck on pending indefinitely. Not sure what the downsides of this are, as I'm not experiencing any problems right now, but my understanding is that it shouldn't behave like that, so I'm opening the issue so we can investigate.
[image]

rojer, 2020-09-14:
this is completely normal, actually. 0 pending connections, 4 active out of 9 maximum supported.

vladbisceanu, 2020-09-14:
Understood. My bad for the fake alarm. Will close the issue now.

closed 2020-09-14 by vladbisceanu