mongoose-os-apps/shelly-homekit #138: Support for Web Hooks.

andyblac, 2020-09-19:
Now we have detached mode which supports single, double and long press, would be possible to add web hooks support to these actions when in normal HAP mode for "Double / Long Press", leaving single press linked to the physical switch?

What I mean by this that if enabled "Double and Long Press", would post a url request that is setup in web UI on the action of the physical switch.

rojer, 2020-09-19:
this makes me think that we should expand functionality of inputs.
we have inputs that generate events: single, double and long press. these events can be tied to one (or more) actions, one of them would be to toggle a physical switch, turn it it on or off, another could be send a hap even (a stateless switch service will be created), or send a web hook.

andyblac, 2020-09-19:
sounds good to me.

rojer, 2020-09-23:
i filed https://github.com/mongoose-os-apps/shelly-homekit/issues/147 to track that and will close this one

closed 2020-09-23 by rojer