mongoose-os-apps/shelly-homekit #139: Flash script not working

yuvalbeck, 2020-09-19:
[image]

No command is working keep getting same help screen @andyblac

andyblac, 2020-09-19:
hmm did it not ask you to install the tools ?

andyblac, 2020-09-19:
leave with I'll look thx

andyblac, 2020-09-19:
ok please this version if fixed I'll submit pull request

[flash_shelly.sh.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5250387/flash_shelly.sh.zip)

yuvalbeck, 2020-09-19:
Firstly, thank you.
Yes, I would expect have a list of all devices with an update available and have an option to agree to update all at once and not press "y" each time.

I also see some issues - will send you over

andyblac, 2020-09-19:
> Firstly, thank you.
> Yes, I would expect have a list of all devices with an update available and have an option to agree to update all at once and not press "y" each time.
> 
> I also see some issues - will send you over

Hi np, You can -y option in command line to bypass the question

andyblac, 2020-09-21:
can I close this now ?, for any addition feature please here https://github.com/mongoose-os-apps/shelly-homekit/issues/132

yuvalbeck, 2020-09-21:
Thank you for your help the file you sent worked

andyblac, 2020-09-21:
thanks, has now been committed.

closed 2020-09-21 by andyblac