mongoose-os-apps/shelly-homekit #142: HAP Typ Garagedoor

Pixel-Chris, 2020-09-20:
It would be nice if the garagedoor HAP Typ could be integrated for a Shelly 2.5, then the second input can be used for detecting a closed garagedoor (with an magnetic sensor).

Ludelag, 2020-09-21:
You can do this already with the Shelly 1, just use the L for mass, the N for the neutral, connect I and O to the garage door, the switch goes to the magnetic sensor and feed also the magnetic sensor with L.
configure the Shelly 1 as detached

ursusHU, 2020-09-21:
https://github.com/mongoose-os-apps/shelly-homekit/issues/43

Pixel-Chris, 2020-09-23:
@Ludelag 
Thanks!!
That works for now!

The most Hörmann Garagedoor openers have an implemented door closed output!

So when HAP type „Garage“ is implemented, the user should be able to use 1 sensor with Shelly 1 and 2 sensors with shelly 2.5
(If you also want a sensor for an completely opened door)

Ludelag, 2020-09-23:
> @Ludelag 
> 
> Thanks!!
> 
> That works for now!
> 
> 
> 
> So when HAP type „Garage“ is implemented, the user should be able to use 1 sensor with Shelly 1 and 2 sensors with shelly 2.5
> 
> (If you also want a sensor for an completely opened door)
> 
> 
> 
> Some Tips :
> 
> The most Hörmann Garagedoor openers have an implemented door closed output! And the Shelly can be powered with 24V from the board.

@Pixel-Chris good to read it has worked! Thanks for the tips as well, I will check it out.

Pixel-Chris, 2020-09-23:
@rojer 
Is it possible to use the „serial port“ gpios for a second sensor/Input ?

[image]

rojer, 2020-09-23:
in principle yes, in practice - unlikely. once you go up to any significant length of wires, noise becomes an issue and you will get false triggering all the time. you are better off just buying shelly 2.5.
correct me if i'm wrong, but garage door functionality sounds more like #5, no?

andyblac, 2020-09-23:
> in principle yes, in practice - unlikely. once you go up to any significant length of wires, noise becomes an issue and you will get false triggering all the time. you are better off just buying shelly 2.5.
> correct me if i'm wrong, but garage door functionality sounds more like #5, no?

yes, bug normally garage door, only operate on one button that  does (OPEN, STOP, CLOSE, STOP, OPEN, etc)

rojer, 2020-09-23:
ok. i don't own and have never used a motorized garage door, i'll need detailed description of the operation and available inputs and outputs.

andyblac, 2020-09-23:
> ok. i don't own and have never used a motorized garage door, i'll need detailed description of the operation and available inputs and outputs.

on mine, and I think the vast majority operate on a single push button type, each time you press the button it cycles through the above function list.

Pixel-Chris, 2020-09-23:
The problem is, that most garagedoor openers have only one button (direction 1 -> stop -> direction 2 -> stop -> direction 1 and so on)
So no separate buttons for up an down
And for the end positions normally magnetic sensors are used, Some of them have an door closed sensor included, some of them have also the open sensor included) 

For my Hörmann promatic 2 (with included garage door closed sensor/output)

andyblac, 2020-09-23:
> The problem is, that most garagedoor openers have only one button (direction 1 -> stop -> direction 2 -> stop -> direction 1 and so on)
> So no separate buttons for up an down
> And for the end [...]

I use iSmartGate on mine, but the original opener, did not even have sensors, just end points where if no button was press to stop the motor it just stopped by itself as it hit the endpoint.

rojer, 2020-09-23:
can you find and link here some documentation for popular garage door systems that we'd be integrating with?

Pixel-Chris, 2020-09-23:
Hörmann promatic 2

https://www.tor7.de/media/pdf/e2/59/13/hoermann-promatic-2-anleitung.pdf

rojer, 2020-09-23:
um... in english, if possible :)

andyblac, 2020-09-23:
https://www.auto-roll.com/sites/default/files/AR433-%20plus%20instructions%20-%20updated.pdf

Pixel-Chris, 2020-09-23:
In English Hörmann Promatic 2
https://cdn.hoermann-cloud.de/fileadmin/_country/dok/Promatic_v2_Fitting_Instructions.pdf?v=1520493487

Pixel-Chris, 2020-09-23:
So for my Hörmann garagedoor opener I connected the Shelly as follows:

L(-) goes to 20
N(+) goes to 5
So the Shelly is powered with 24V

0 and I goes to the momentary switch input

SW to the 0V R- (where normally the optional Relais is connected)
(Detached mode, Auto off after 2 seconds)

Pixel-Chris, 2020-09-23:
[image]

Pixel-Chris, 2020-09-24:
And if the Gpio pins can be used:

L(-) goes to 20
N(+) goes to 5

0 and I goes to the momentary switch input

SW to a magnetic door sensor (fully opened)

GPIO to the 0V R- (where normally the optional Relais is connected) (door closed)

Pixel-Chris, 2020-09-24:
Found a thread where they used the gpio pins:

https://community.openhab.org/t/setting-up-a-garage-door-opener-with-shelly-and-espeasy/98749

vkingxxx, 2020-09-24:
> Found a thread where they used the gpio pins:
> 
> https://community.openhab.org/t/setting-up-a-garage-door-opener-with-shelly-and-espeasy/98749

Why do you need 2 sensors? One magnetic or inside signal for closed door is sufficient. Sensor is active = door is close, inactive door is open.

Pixel-Chris, 2020-09-24:
The HAP Type „Garagedoor“ can show the actually states:
Door opened
Door closing
Door closed
Door opening

So both end positions would be nice
And when the opener provides an internal output for one of them why not use the gpio for that and an external magnetic switch for the other end position.

And when there are two recognised endpositions, and after x seconds none of them is the actual state - you know that there is a problem
(Maybe there is a state „door failure“ or something like that)

The Hörmann supramatic Series has two build in (internal) enposition outputs. So why not use these?

Pixel-Chris, 2020-09-24:
And I found something:

let HMCharacteristicTypeObstructionDetected: String
An indicator of whether an obstruction is detected, as when something prevents a garage door from closing.

(https://developer.apple.com/documentation/homekit/hmcharacteristic/characteristic_types)

andyblac, 2020-09-24:
@Pixel-Chris my HomeKit opener only has 1 sensor and all 4 conditions are supported, when Closed and it start opening it show "Opening.." till the motor stops (no current) then it shows "Open" (from as the sensor is open, and same in reverse. So only 1 sensor is required. IMO.

Pixel-Chris, 2020-09-24:
But, Yes you are right 
One sensor is normally enough 

But when you have to sensors... why not use both? The Shelly can do it (an when not the gpio pins of an Shelly one - then an Shelly 2.5 can do it)

Motor stops because of an obstacle? Or what after an power loss? In which position is the door (open? Or in the middle?)

vkingxxx, 2020-09-24:
I have too 4 states with one sensor with another firmware. In this firmware is defined time to close XY seconds, when door is not in XY seconds closed/sensor activated = obstruction. I don't need detect obstruction for opening.

Is possible with @rojer firmware switch door status on SW contact?

andyblac, 2020-09-24:
> Yes you are right
> One sensor is normally enough
> 
> But when you have two sensors... why not use both? The Shelly can do it (an when not the gpio pins of an Shelly one - then an Shelly 2.5 can do it)
> 
> Motor stops because of an obstacle? Or what after an power loss? In which position is the door (open? Or in the middle?)

shelly 2.5 is not dry contact, so can't always be used. ie I couldn't use it on mine, as NO VOLTAGE can go through to garage circuit.

vkingxxx, 2020-09-24:
@Pixel-Chris https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door

Pixel-Chris, 2020-09-24:
IMO
It should be possible to choose
- no sensor
- one sensor
- two sensors
and where there are connected
SW, GPIO x

Maybe rojer can include the new calibration mode form the roller shutter #5 type (https://github.com/mongoose-os-apps/shelly-homekit/issues/5#issuecomment-698174092)
to define the times

rojer, 2020-09-24:
i'm not super excited about the idea of making header gpios available for stuff, because they are not properly isolated.

Pixel-Chris, 2020-09-24:
> i'm not super excited about the idea of making header gpios available for stuff, because they are not properly isolated.

Shelly 2.5 ?

rojer, 2020-09-24:
i mean the additional GPIOs available on the programming header on any type of device.
shelly 2.5 provides 2 proper inputs, so yes, those can be used for remote sensing.

Pixel-Chris, 2020-09-24:
Ok,
Perfect !

hfagelnour, 2020-09-27:
> You can do this already with the Shelly 1, just use the L for mass, the N for the neutral, connect I and O to the garage door, the switch goes to the magnetic sensor and feed also the magnetic sensor with L.
> configure the Shelly 1 as detached

The reed switch (magnetic) .... does it have to be on by default or off?

Pixel-Chris, 2020-09-28:
> The reed switch (magnetic) .... does it have to be on by default or off?

The sensor contact should be closed, when the garagedoor is closed / when the sensor is near the magnet.

Pixel-Chris, 2020-09-28:
NC = normally close

juancitomadrid, 2020-09-28:
Can tests be done with current firmware for Shelly 1 ? (I have a garage door with 1 sensor, open/close).
What setup should I use?

Pixel-Chris, 2020-09-28:
Yes
Testing is possible with the stock firmware.
The closed door is visualised with a blue line in the power button:
[image]

Pixel-Chris, 2020-09-28:
My garagedoor opener can power the Shelly with 24v
And the closed sensor is included....

https://github.com/mongoose-os-apps/shelly-homekit/issues/142#issuecomment-697941414

Stundenplan7b, 2020-09-28:
> You can do this already with the Shelly 1, just use the L for mass, the N for the neutral, connect I and O to the garage door, the switch goes to the magnetic sensor and feed also the magnetic sensor with L.
> 
> configure the Shelly 1 as detached

Hello Ludelag, if i do this what happens in the HomeKit App? Will i see a Counter or a garagedoor? If i see the normal Counter how does it work? Is the Counter active while Door is open and inactive while the door is closed?

Pixel-Chris, 2020-09-29:
Actually it works only with the stock firmware, no HomeKit!

andyblac, 2020-09-30:
@rojer should I close this ticket https://github.com/mongoose-os-apps/shelly-homekit/issues/105 in favour of this one ?

Ludelag, 2020-09-30:
> Yes
> 
> Testing is possible with the stock firmware.
> 
> The closed door is visualised with a blue line in the power button:
> 
> [image]
> 
> 

When I just updated to the last version of the Shelly 1 immediately came to my mind this blue bottom that works when the switch Shelly is detached, with the new version you actually got more o less the same when detached you get the input 1 which could be use to connect a door sensor abs detect if the door is open or close. Even can be coded to appear in HomeKit as garage door showing the open or close position. 
Like shower on this video: https://youtu.be/P5mQADnZXOI

Pixel-Chris, 2020-10-01:
With the stock firmware the sensor is connected to the detached switch.
And the input and output is wired the the momentary switch input of the garage door opener.
So the switch must be in detached mode and the auto-off needs to be set (for example „2 seconds“).

So you can control the garagedoor over the Shelly App or an url
And you can see the state of the door in the Shelly app

Now the HAP Type „garage door“ must be implemented in the HomeKit Firmware, so that it is displayed as an garage door in the home app and an external sensor wired to the detached switch port must be recognized.
Optional, when there are two sensors for opened and closed state, the Shelly 2.5 must be supported (I think this will only work, when the Shelly is powered over the garage door opener with 24V).

Pixel-Chris, 2020-10-01:
@rojer 
Sorry I have to ask again ;-) (GPIOs)

When you add an option to the Shelly 1 webui... for example:
„Has the opener an integrated sensor?“
„Is the Shelly powered with 24V from the garage door opener“
or can the Shelly recognise on its own, how it is powered? 12v/24v/230v ???

When that would be possible... and the Shelly 1 is powered over 12v or 24v - and only then - the option to use the gpios is given to use one of them for an additional sensor.
What would you say about this?

As additional info:
When the opener has an integrated sensor and the Shelly can be powered over the opener,
The Shelly can be integrated in the opener housing - so cables can be very short for the gpios.

rojer, 2020-10-01:
@Pixel-Chris we'll have some UI options for sure.
wrt power - i don't think it matters how shelly is powered, as long as it works and input works as well.
again, i am not a fan of allowing pin header GPIOs to be used as inputs, it can damage internal circuitry and absence of proper input noise suppression will cause issues with false triggers that will be difficult to debug.
as far as i'm concerned, if you need more than one input - get a shelly 2.5.

please keep updating this issue with relevant information, i'm not focusing on it right now but it will come in handy when the time comes.

in particular, i'd like to know how much garage door control is different from roller shutter control that we're discussing on #5, and whether it makes sense to share some code internally.

Pixel-Chris, 2020-10-02:
IMO
It is not the same like an Roller shutter, because you only send an signal to the motor that it should start or stop (the complete cycling process is controlled by the opener itself)
the garagedoor opener is not powered over the Relais of the Shelly.

Pixel-Chris, 2020-10-02:
or the Shelly Door Window sensor ?

Ludelag, 2020-10-03:
@rojer the garage door needs most of them usually needs only trigger for 1 or 2 seconds to operate is quiet simple. The same action wired all the same action the door to open trigger again stops trigger again opens. If the door is open when trigger food closes or vicevers

rojer, 2020-10-03:
so what additional support is needed? isn't a switch with auto-off after 1-2 seconds enough?

Ludelag, 2020-10-03:
To open and close the garage door is perfect ! Thank you, only that in detached mode with the input 1 switch could be use to detect the position of the door, if it is open or closed using a wire sensor door detector like for example this one: 

Gebildet 5pcs Cableado Empotrado Seguridad de Ventana/Puerta Sensor de Contacto Alarma,Interruptor de Láminas Magnético (Blanco),Normalmente Cerrado Sensor de Puerta Magnético https://www.amazon.es/dp/B082PNPWFZ/ref=cm_sw_r_cp_api_fab_-eeEFb4362THW

Whiterabbit64, 2020-10-03:
The switch with auto off is enough to drive the garage door through cycles (open, stop, close, open, etc.)
But this does not allow you to get the garage door status (opened, closed). To achieve this some of us add a wireless sensor. 
But to make all of this working perfectly, Shelly 1 and wireless sensor should work together in order to:
- «  Hey Siri open the garage door » will open it only if it’s closed
- «  Hey Siri close the garage door » will close it only if it’s already opened.

And the we also know the garade door status, opend or closed.

Pixel-Chris, 2020-10-03:
IMO
It should be possible to choose how many sensors

0, 1 or 2

and if there are NO or NC

0 and I goes to the momentary switch input

Pixel-Chris, 2020-10-03:
With two sensors:
You can exactly say in which status the door is:
Closed sensor is active: the door is closed
Open sensor is active: the door is open
No sensor is active: The door is moving
No sensor is active for a longer period: Door failure/obstruction 

So the drive times should be editable or there is a calibration mode for this (like roller shutters #5 )

I think the minimum is one sensor, because you can control the door with your remote or an extra push/key/fingerprint button (connected to the opener board). If someone uses one of these, HomeKit would not know, that there is an status change.

hfagelnour, 2020-10-04:
> You can do this already with the Shelly 1, just use the L for mass, the N for the neutral, connect I and O to the garage door, the switch goes to the magnetic sensor and feed also the magnetic sensor with L.
> configure the Shelly 1 as detached

@Ludelag I used the above but i am using 12V to power the shelly. i thus connected L to -ve and N to +ve, I and O to the garage door. the NC magnetic sensor is connected to the SW and L (-ve) Detached
The sensor status on Stock shows the little I inside the power logo as it should but no matter what i do in HomeKit using the firmware here it only flickers the Input icon for second while it doesn't show if the magnetic circuit is closed or opened!
any thoughts pls

Pixel-Chris, 2020-10-04:
@hfagelnour 
Actually the HomeKit Firmware is not compatible with garagedoor openers!

hfagelnour, 2020-10-04:
Oh ok thx a million, I believe I will go for the meross solution then. It does all what we want and more. It is not expensive and compatible with HomeKit, Amazon Alexa and smart things. Will feed back here when I try it.

Pixel-Chris, 2020-10-05:
@rojer 
What will be you next project ? ;-)

Ludelag, 2020-10-05:
> @Ludelag I used the above but i am using 12V to power the shelly. i thus connected L to -ve and N to +ve, I and O to the garage door. the NC magnetic sensor is connected to the SW and L (-ve) [...]

Yes, exactly this is what we have been talking about and trying to explain to @rojer, there is no real use for the detached Shelly 1, is just input and useless for garage door. If you get back to the stock firmware you get better results in detached mode for garage door.

Probably in future @rojer will have the will to get a look to the detached in Shelly 1 make it work with a wire sensor which will show the real state of the door, if open or close as the Shelly you can find in Amazon 

https://www.amazon.es/HomeKit-kompatibel-Shelly-Garage-Door/dp/B07WN8YXGP/ref=sr_1_1?__mk_es_ES=ÅMÅŽÕÑ&dchild=1&keywords=shelly+garage+door&qid=1601933229&sr=8-1

Ludelag, 2020-10-05:
@rojer @Pixel-Chris @hfagelnour this what you were writing about? https://www.youtube.com/watch?v=yDa-hZJ6dlQ

[311 more replies]

closed 2020-10-12 by rojer