mongoose-os-apps/shelly-homekit #144: Reboot function

Elsandro12, 2020-09-22:
Hello 
I have the problem that some of my devices are shown in the HomeKit app as ,,no Answer,, . Not the same every week another one. 

When I go to the device directly with the ip address I can switch it normally . I also can reboot the device and the problem with HomeKit  is solved.

perhaps you can make a option that the device automatically reboots it`s self at night.

andyblac, 2020-09-22:
1st I presume your running 2.1.7 ?, if not please update, if you are this sounds like an on going issue being discussed here https://github.com/mongoose-os-apps/shelly-homekit/issues/129, 'reboot' has been deceased before and was ruled out. 

therefore @rojer I think this ticket can be closed?

Elsandro12, 2020-09-22:
thank you I will try to understand #129 you can close it

closed 2020-09-22 by Elsandro12

andyblac, 2020-09-22:
you can reopen if is not same issue.