mongoose-os-apps/shelly-homekit #146: Physical Switch state in Edge Mode

Marfre888, 2020-09-23:
Hi @rojer , thank you so much for the latest update, been waiting for it for a while.

I have two small suggestions that would benefit my setup, and I wonder what you think about them;

Would it be possible to extend the new detached mode switch feature to Edge mode (obviously with the possibility of disabling the switch). Reason being, if shelly is currently functioning in detached mode, it adds more reliance on homekit doing the work. With both shelly and homekit still not being 100% reliable, this means that if either had to fail, you would lose the ability to turn on your light. So in edge mode, the light would still turn on via the physical switch, but you gain the benefit of being able to assign a shortcut to the switch.

Instead of exposing a stateless switch, could a stateful switch be exposed. ie: in edge mode, this would be exposed as, ON when physical switch is in the on position, and OFF, when in the off position. This would mean that you could define the state of a room based on the state of the switch;

IE: If motion sensor triggers, and switch is in the off position, turn on lights, otherwise nothing.

This is of course now possible by adding an additional physical switch, and assigning a single press, to toggle the state of the switch, but it adds another layer of complexity that isn't really needed.

Would be great if we could get these bundled on in there. Again, thanks so much for this update

andyblac, 2020-09-23:
just so I understand you, you basically want Double and Long press exposed then, as single press would always be attached to the switch ?

andyblac, 2020-09-23:
> IE: If motion sensor triggers, and switch is in the off position, turn on lights, otherwise nothing.
> 
thats a relay state, not a switch state surely ?

andyblac, 2020-09-23:
If you want the switch attached in Edge mode, and you only want you sensor to only turn on the light if OFF, then just simply add a 'Condition' to that automation. Is that what you're after ?

Marfre888, 2020-09-23:
Sorry, I'm doing a horrible job explaining this.

Basically I'd like to be able to get the position of my physical switch (the one attached to my wall) into homekit. ie: UP/DOWN.

I'd like it in Edge or Toggle mode for purely reliability reasons, as I'm getting instances a couple of times a month where automations don't go through. So in edge/toggle mode, the physical switch is still able to turn on or off the light, but also reports the state of the physical switch.

No, it's a switch state, not a relay state, because the light itself is on a timer, but the physical switch is not. So my hoping is that if I turn on the physical switch is disables the motion sensor/timer, but if I just want to quickly walk into a room, the motion sensor turns on the light (and runs the minute timer to turn it back off). As a result the physical switch would be untouched.

Regarding single, double, and long presses. In single press for on, double press for off mode, they could serve this purpose, however you would have to create a new toggle switch tile, and offload to process to homekit, which isn't something I'm too keen on. That said, the single presses are very useful if you want to add a smart bulbs/accessories to activate with the physical switch, so if they don't need to be disabled, I don't see any reason why they need to be

andyblac, 2020-09-23:
ok, I think I get you now, so you are wanting a to add a new Tile for the Switch State, that would be a dummy Tile only. there is a request for that here. https://github.com/mongoose-os-apps/shelly-homekit/issues/74

Add your support to that request, I think this is a duplicate of that request, if so please close this one, if not I'll let @rojer advise what is / is not possible.

Marfre888, 2020-09-23:
Give me a second, I've come up with a way to show you

rojer, 2020-09-23:
ok, so you want input to control a physical switch as well as send an event to a stateless switch tile in homekit?

i think this will have to wait for the "advanced input actions configuration". i filed https://github.com/mongoose-os-apps/shelly-homekit/issues/147 for that purpose.

Marfre888, 2020-09-23:
So this is what I'm trying to do.

But this process involves;

Shelly switch ->> homekit ->> homebridge dummy switch
Shelly switch ->> homekit ->> shelly light

But in reality the shelly already knows all this information and shouldn't need to speak to homekit, other than to send it the result. Plus, more joint means that there is always higher chance that it will fail.

PS: It would also be nice if you could group the dummy switch/stateless switch, to keep homekit decluttered.

[https://cvws.icloud-content.com/B/AfR5DsghcvRsmlJdt2uk6BtcKcyIAcJzaOilttHx6feIqbC4DuN0lxSy/RPReplay_Final1600853678.mp4?o=AsaLCt2UuH1_-OyO0OyiZWcm_kybOBabcFCzlZMfyswd&v=1&x=3&a=CAog-P2BrLsiwH9EIO2zjzFDN1oEl5BItPelHnMhT3glQ-0SaxDUtOPSyy4Y9Kua08suIgEAUgRcKcyIWgR0lxSyaiWvpHm9Q9-XjZKXlwBYnd4NE98H2cElKsEbsDm_x8fx4hXjb6bJciXnqOyZ291Za-iej9rrJxhbREgNP1hq3rqGGVE6gRe5GlOr8-uI&e=1600855119&fl=&r=8a51f166-ceb6-4bc3-b314-611fdfb38d75-1&k=79AVAmIDa1AsNNzI4FbbhQ&ckc=com.apple.clouddocs&ckz=com.apple.CloudDocs&p=26&s=kojAoEcO1UkMsTer38LK58A8ICI&cd=i](url)

andyblac, 2020-09-23:
video link does not work .

Marfre888, 2020-09-23:
@andyblac This should work now;

https://drive.google.com/file/d/1qs60wSDOJqni01WFzykXjBQn_wE4t6Su/view?usp=sharing

andyblac, 2020-09-23:
ye, so I was right, it just a dummy switch you're after. #74

Marfre888, 2020-09-23:
Yes, but it would be even better, if when physical switch is turn on, Shelly could turn on the relay (light) without needing to ask homekit to do it, as it does in toggle/edge mode.

andyblac, 2020-09-23:
updated slightly read above.

andyblac, 2020-09-23:
> PS: It would also be nice if you could group the dummy switch/stateless switch, to keep homekit decluttered.

unfortunately you can't group different switch types in HomeKit.

Marfre888, 2020-09-23:
Not exactly; because the 'dummy switch' is only intended to show physical switch state.

So if you manually turn the light off in homekit, using the light tile, the 'dummy switch' should still show up as on (as it physically hasn't been turn off from the wall)

So in reality on switch click; shelly would execute two separate actions. 1. to turn the light on, 2. to turn the dummy switch on. In a perfect world, dummy switch cannot be manipulated unless physical switch state is changed, so I'm not sure if exposing it as an occupancy sensor would make more sense.

Marfre888, 2020-09-23:
> > PS: It would also be nice if you could group the dummy switch/stateless switch, to keep homekit decluttered.
> 
> unfortunately you can't group different switch types in HomeKit.

Sorry, by group I meant, the 'separate tiles' feature. I have accessories that have different types of tiles grouped together, so surely there is a way to do that..

andyblac, 2020-09-23:
> Sorry, by group I meant, the 'separate tiles' feature. I have accessories that have different types of tiles grouped together, so surely there is a way to do that..

ye I know, not possible.

Marfre888, 2020-09-23:
Light/Dummy Switch group together, with possibility of disabling stateless switch?

andyblac, 2020-09-23:
> Light/Dummy Switch grouped together, with possibility of disabling stateless switch?

@rojer can confirm, but I am pretty the feature is only for multiple thing of the same type ie switches and sensors, they can't be different types, this why when you enable Detached mode, even though it is part of the same accessory it is not show in the single tile.

Marfre888, 2020-09-23:
Ahh, I have definitely had a fan slider group with a switch before. I think homekit only lets users group the same type of tile, but a developer can choose how they want the tile to be exposed, but I'm obviously no expert.

Either way I'm not dying for that option, but would be nice to have

andyblac, 2020-09-23:
> Ahh, I have definitely had a fan slider group with a switch before. I think homekit only lets users group the same type of tile, but a developer can choose how they want the tile to be exposed, but I'm obviously no expert.
> 
> Either way I'm not dying for that option, but would be nice to have

ye, If both are switches (switch set fan in homekit) just you can't have a stateless and a state switch in same tile, as they are 2 different types.

Marfre888, 2020-09-23:
Haha, thank you @andyblac.

That took way too much time to explain, but thanks for your help

timoschilling, 2023-02-10:
I reed this thread 3 times in a row and didn't find the point what this feature would improve.

So I try to write down what I understand:
1. You want that the wall switch and the relay works like it does now in edge mode, moving the switch from on to off or from off to on inverts the relay state. But additional to that you want a switch tile in HomeKit which represents the physical state of the wall switch, wall switch on = HomeKit switch on and off = off. Is that right?
2. You want to solve some kind of problem with that. But I'm not sure which one. It sounds like you would improve automatisations by reducing roundtrips between HomeKit and Shelly, right?

Marfre888, 2023-02-13:
> I reed this thread 3 times in a row and didn't find the point what this feature would improve.
>
> So I try to write down what I understand:
>
> 1. You want that the wall switch and the relay works like [...]

Yes that seems standard with anything I write it seems.

It's very simple. The ticket focuses on detached mode and the overreliance on homekit that this firmware has created over the years, through the implementation of the feature. (Although incredibly useful)

It is very dangerous, because (and now in 2023 we can finally say that we've seen it with the new architecture rollout) if your home were to go into a no-response state, or your homehub were to fail, then you suddenly have no control of your devices at all. You can't even turn on your lights.

My suggestion was that the shelly functions normally, so the physical switch is able to turn on the relay without interacting with Homekit, but the switch also reports its state in homekit as a stateless switch (as it does now in detached mode)

My second suggestion was that instead of a stateless switch, you expose a stateful switch, which turns on when a switch is turn on, but after reading the homekit documentation, I now know that this wouldn't work very well, and there are easy ways around it.

timoschilling, 2023-02-13:
You already have the state change in homekit as a change on the switch.

Marfre888, 2023-02-13:
> You already have the state change in homekit as a change on the switch.

Yes but it requires that you decouple the switch from the relay

timoschilling, 2023-02-13:
If you push the switch, you get a event in HomeKit.

Marfre888, 2023-02-13:
> If you push the switch, you get a event in HomeKit.

Yes, BUT in detached mode, to turn on the relay, it then needs to wait to receive an instruction from homekit.

What if the network/wifi is offline cause of a failure? Do I turn on a torch and work in the dark?

timoschilling, 2023-02-13:
You don't need a detached mode, you get a event in hk if the switch changes his state!

Marfre888, 2023-02-13:
> You don't need a detached mode, you get a event in hk if the switch changes his state!

Yes, the relay (light) turns on. BUT then for homekit there is no distinction between if the switch triggered the light on or say a motion sensor

timoschilling, 2023-02-13:
🤷‍♂️ Maybe you should simply use a Shelly with two inputs and wire the switch to both inputs, then you have two different behaviours at once. Your request didn't get any +1 in 2,5 years, so I will close it as edge case.

closed 2023-02-13 by timoschilling

Marfre888, 2023-02-13:
> 🤷‍♂️ Maybe you should simply use a Shelly with two inputs and wire the switch to both inputs, then you have two different behaviours at once. Your request didn't get any +1 in 2,5 years, so I will close it as edge case.

Lol.. up to you.. you don't care about improving the firmware so why should I

Hope your homekit fails and you're left fixing it in the dark like an idiot!

And for the record this was moved to a different ticket, you're about 2 and a half years too late

Good day!

timoschilling, 2023-02-13:
Improving open source software doesn't mean to implement every feature to support every edge case, it mean to make a maintainable good average of feature requests.

But any way I hope I give you a place to 🤮.

Marfre888, 2023-02-13:
> Improving open source software doesn't mean to implement every feature to support every edge case, it mean to make a maintainable good average of feature requests.
> 
> But any way I hope I give you a place to 🤮.

It’s not an edge case. This software works for me as is, but I care to improve it!

You’re just stubborn and refuse to see the usefulness of a feature because you didn’t think of it yourself.

For some wild reason you think detached mode is saviour of all your problems, probably because you came up with that yourself. The second you run into an issue, and I have a house with 180devices so I’ve done enough troubleshooting, your lights stop turning on and you’re left in the dark until you fix the problem. The implementation is flawed, and detached mode should be an edge case.

but whatever, you don’t care, and neither do as.. as long as you get the ticket count down