mongoose-os-apps/shelly-homekit #147: Advanced input actions configuration

rojer, 2020-09-23:
It should be possible to specify multiple actions for a physical device input. Among them:
 * Control an output on the same device directly, in toggle/momentary/edge mode
 * Send a HAP event - through a Stateless Switch accessory
 * Call a web hook
 * ... more can be added later

andyblac, 2020-09-26:
should we include #51 into this as well ?

rojer, 2020-09-26:
that's more like advanced _output_ configuration. let's keep it separate.

andyblac, 2020-09-26:
> that's more like advanced _output_ configuration. let's keep it separate.

ok

rojer, 2020-09-29:
additional relevant HAP services for an input:

* contact sensor
* motion sensor
* occupancy sensor

andyblac, 2020-09-29:
https://github.com/mongoose-os-apps/shelly-homekit/issues/124

andyblac, 2020-11-30:
and now that we have an API this could be used to link Shelly devices together.

timoschilling, 2021-05-29:
@rojer do I see it right that every thing in this ticket is already implement, except "Call a web hook" which already requested in https://github.com/mongoose-os-apps/shelly-homekit/issues/70#issuecomment-735776717, so I think we can close this issue as a double of #70

timoschilling, 2022-02-19:
Summery of #934 and #829, some users (including me) have the wish to have an mixed mode of switch and detached mode. We want to control the output for example with a singe push and exposing the long and double as stateless input to HK.

timoschilling, 2023-01-26:
Closed because of https://github.com/mongoose-os-apps/shelly-homekit/issues/147#issuecomment-850909617
> rojer do I see it right that every thing in this ticket is already implement, except "Call a web hook" which already requested in [...]

closed 2023-01-26 by timoschilling