mongoose-os-apps/shelly-homekit #149: Flash Script

gregDEweb, 2020-09-24:
@andyblac 

Sorry for asking but how do i use the script to flash multiple shellys. 
Not sure how to start it by the console.

andyblac, 2020-09-24:
in the tools folder type
`./flash_script -la` to list all devices
`./flash_script -a` to flash all devices
`./flash_script -h`for more help on what the script can do.

gregDEweb, 2020-09-24:
Tools folder on a shelly device? How to access the folder?
Bot familiar with the console so far.

andyblac, 2020-09-24:
tools for in git here https://github.com/mongoose-os-apps/shelly-homekit/blob/master/tools/flash_shelly.sh you need to download or clone the git first

freddair, 2020-09-24:
What's wrong?

Installing coreutils...
You may be asked for your password...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

Installing jq...
You may be asked for your password...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

OS: Darwin
Scanning for Shelly devices...
/Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly.sh: line 387: timeout: command not found
Frederiks-MacBook-Pro:tools frederikmueller$ /Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly.sh -a
Installing coreutils...
You may be asked for your password...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
[... 26 lines cut ...]
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
[... cut ...]
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

OS: Darwin
Scanning for Shelly devices...
/Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly.sh: line 387: timeout: command not found

andyblac, 2020-09-25:
ok, it seems git command line is required, please try the attached version, and report back if it fixes issue.

Thanks 
[flash_shelly.sh.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5281342/flash_shelly.sh.zip)

freddair, 2020-09-25:
Installing coreutils...
You may be asked for your password...
Updating Homebrew...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

Installing jq...
You may be asked for your password...
Updating Homebrew...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

OS: Darwin
Scanning for Shelly devices...
/Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly 2.sh: line 411: timeout: command not found

andyblac, 2020-09-25:
Is this from zip file above ?

andyblac, 2020-09-25:
Also please I need to see full output including command you used to start

andyblac, 2020-09-25:
@freddair please try the version in the zip above that should install git before it tries to install coreutils

freddair, 2020-09-26:
@andyblac yes I used the zip above and replaced the old .sh in the tools folder. Just did it again and here is the full output:

Frederiks-MacBook-Pro:~ frederikmueller$ /Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly.sh -ay
Installing coreutils...
You may be asked for your password...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

Installing jq...
You may be asked for your password...
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)

OS: Darwin
Scanning for Shelly devices...
/Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly.sh: line 411: timeout: command not found
Frederiks-MacBook-Pro:~ frederikmueller$

andyblac, 2020-09-26:
hmm strange, it seems git is still missing, can you run `which git` in terminal please.

should see something like
```
Andrews-MacBook-Pro:~ andrewblackburn$ which git
/usr/bin/git
```

freddair, 2020-09-26:
looking good:

`Frederiks-MacBook-Pro:~ frederikmueller$ which git
/usr/bin/git
`

andyblac, 2020-09-26:
hmm so why does brew not see it. can you try to reboot, and try script again

andyblac, 2020-09-26:
could also try the following please

```
cd /usr/local/Homebrew
git pull origin master
brew update && brew upgrade
```

let me know.

andyblac, 2020-09-26:
it might have something to do with upgrading to Big Sur, brew might need a update, try the above commands anyway let me know.

freddair, 2020-09-26:
I already tried rebooting before, nothing changed. Here is the outcome of your commands above:

Frederiks-MacBook-Pro:~ frederikmueller$ cd /usr/local/Homebrew
Frederiks-MacBook-Pro:Homebrew frederikmueller$ git pull origin master
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
Frederiks-MacBook-Pro:Homebrew frederikmueller$ brew update && brew upgrade
Traceback (most recent call last):
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `<main>'
	 9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:23:in `require_relative'
	 8: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `<top (required)>'
	 7: from /usr/local/Homebrew/Library/Homebrew/global.rb:28:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:3:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:21:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:58:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:24:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:25:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:368:in `initialize': Version value must be a string; got a NilClass () (TypeError)
Error: Git must be installed and in your PATH!

andyblac, 2020-09-26:
hmm, did you have Xcode installed at any time ? is there an update available for it?

andyblac, 2020-09-26:
If that doesn't show you an update run:

```
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install
```

freddair, 2020-09-26:
[image]

hmmm... Software can't be installed because it's currently not available on the software update server... maybe I really should wait until Big Sur is released.

andyblac, 2020-09-26:
try opening Xcode app, it may be require to complete Big Sur update.

freddair, 2020-09-26:
haha. xcode was not installed. now loading...
thanks @andyblac

andyblac, 2020-09-26:
Xcode should not be needed for script, but Big Sur thinks you previously had it installed, so it must have need to be updated. anyway let me know if script works when installed.

freddair, 2020-09-26:
now it worked! But it only found 2 of 3 shelly devices in my network - why could this be?

andyblac, 2020-09-26:
hmmm, strange what does 
`timeout 2 dns-sd -B _http . `

report ?

freddair, 2020-09-26:
Now I see: Could not resolve host: shelly1-8CAAB505BEF3

```
Frederiks-MacBook-Pro:Homebrew frederikmueller$ /Users/frederikmueller/Documents/GitHub/shelly-homekit/tools/flash_shelly.sh -a
OS: Darwin
Scanning for Shelly devices...
Could not resolve host: shelly1-8CAAB505BEF3

Host: shelly1-8CAAB505BF22
Model: Shelly1
Current: HomeKit 2.1.8
Latest: HomeKit 2.1.8
Does not need updating...

Host: shelly1-8CAAB5061CC9
Model: Shelly1
Current: HomeKit 2.1.8
Latest: HomeKit 2.1.8
Does not need updating...
Frederiks-MacBook-Pro:Homebrew frederikmueller$ timeout 2 dns-sd -B _http .
Browsing for _http._tcp
DATE: ---Sat 26 Sep 2020---
15:31:53.115  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
15:31:53.116  Add        3   4 local.               _http._tcp.          shelly1-8CAAB505BEF3
15:31:53.116  Add        3   4 local.               _http._tcp.          shelly1-8CAAB505BF22
15:31:53.116  Add        2   4 local.               _http._tcp.          shelly1-8CAAB5061CC9

```

freddair, 2020-09-26:
I use pi-hole as DNS - but strangely 2 of 3 shellys do work.

andyblac, 2020-09-26:
> I use pi-hole as DNS - but strangely 2 of 3 shellys do work.

can access it via web browser ?

andyblac, 2020-09-26:
could be related to this https://github.com/mongoose-os-apps/shelly-homekit/issues/108

freddair, 2020-09-26:
ahh no it's not accessible via web browser and also not responding in HomeKit... but my router shows it being connected. I will need to wait for the night when everybody sleeps so that I can cut the power of the living room ;-)

andyblac, 2020-09-26:
at least it working now, let me know if you have any more issues, or would like anything adding to it.

andyblac, 2020-09-26:
@gregDEweb did you get it working ?

gregDEweb, 2020-09-27:
@andyblac 
Sorry could not will rk on it yet. Hab to leave town for a couple of days, will try it next weekend.

andyblac, 2020-09-27:
np, no rush, I'll push the fix in the mean time, let me if works when your return.

rojer, 2020-09-28:
@andyblac FYI, in https://github.com/mongoose-os-apps/shelly-homekit/commit/ccc07452f04dec9e1fab1471dde6780faf2c08f5 i added `stock_model` to Shelly.GetInfo (now that web ui also needs it), so you can use that directly when constructing stock firmware URLs.

andyblac, 2020-09-29:
added a wiki for flash script here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Flashing-script

juancitomadrid, 2020-09-30:
I am trying to run the script, and [this](https://www.dropbox.com/s/xc5ujcsg5lrrnzk/Captura%20de%20pantalla%202020-09-30%20a%20las%2011.07.10.png?dl=0) is what I get.
I am doing something wrong?

juancitomadrid, 2020-09-30:
> I am trying to run the script, and [this](https://www.dropbox.com/s/xc5ujcsg5lrrnzk/Captura%20de%20pantalla%202020-09-30%20a%20las%2011.07.10.png?dl=0) is what I get.
> I am doing something wrong?

Forget about this.
Working!

andyblac, 2020-09-30:
> I am trying to run the script, and [this](https://www.dropbox.com/s/xc5ujcsg5lrrnzk/Captura%20de%20pantalla%202020-09-30%20a%20las%2011.07.10.png?dl=0) is what I get.
> I am doing something wrong?

for others, yeah, it just installing the tools required for script to work.

gregDEweb, 2020-10-04:
So far i do not have access to a Linux System anymore just Windows and could not get it to work.

andyblac, 2020-10-04:
> So far i do not have access to a Linux System anymore just Windows and could not get it to work.

Good news working on a total rewrite of script in python that should make it compatible with windows 🤣

andyblac, 2020-10-05:
@gregDEweb would you be willing to help test my python version on windows?

andyblac, 2020-10-08:
@gregDEweb I have completed my python version of the script please give it a test on Windows. It requires Python 3.6 or later.

gregDEweb, 2020-10-08:
@andyblac 
Yes i will try if it works. 
Send me a link and i'll try. I'm pretty busy right now so it might take some days.

andyblac, 2020-10-08:
ye, np.

andyblac, 2020-10-08:
script in git in tools folder.

gregDEweb, 2020-10-08:
yeah found it sorry. did you try it before? not working so far

andyblac, 2020-10-08:
not on windows, can you run it with a -D option as well, and post results.

pivale, 2020-10-13:
Hey @andyblac,

Thank you for creating this script. I find it really useful and I only have 2 shellys installed (for now, I have 5 more for when I have the time!)

When I run the script, I always get duplicates for my devices.
Any idea?

`nb-pvale:tools pedro.vale$ ./flash_shelly.sh -la
OS: Darwin
Scanning for Shelly devices...

Host: shelly1pm-extensao
Model: Shelly1PM
Current: HomeKit 2.2.1
Latest: HomeKit 2.3.0

Host: shelly1pm-extensao
Model: Shelly1PM
Current: HomeKit 2.2.1
Latest: HomeKit 2.3.0

Host: shellyswitch25-sala-persiana
Model: Shelly25
Current: HomeKit 2.3.0
Latest: HomeKit 2.3.0

Host: shellyswitch25-sala-persiana
Model: Shelly25
Current: HomeKit 2.3.0
Latest: HomeKit 2.3.0
nb-pvale:tools pedro.vale$ ./flash_shelly.sh -am keep
OS: Darwin
Scanning for Shelly devices...

Host: shellyswitch25-sala-persiana
Model: Shelly25
Current: HomeKit 2.3.0
Latest: HomeKit 2.3.0
Does not need updating...

Host: shellyswitch25-sala-persiana
Model: Shelly25
Current: HomeKit 2.3.0
Latest: HomeKit 2.3.0
Does not need updating...

Host: shelly1pm-extensao
Model: Shelly1PM
Current: HomeKit 2.2.1
Latest: HomeKit 2.3.0
Do you wish to flash shelly1pm-extensao to firmware version 2.3.0 ? y
Downloading Firmware...
Now Flashing...
  Update applied, rebooting
Successfully flashed shelly1pm-extensao to 2.3.0

Host: shelly1pm-extensao
Model: Shelly1PM
Current: HomeKit 2.3.0
Latest: HomeKit 2.3.0
Does not need updating...
nb-pvale:tools pedro.vale$ `

andyblac, 2020-10-13:
hmm. can you run it with `-v=0` please and post results 

ie `./flash_shelly.sh -la -v=0`

pivale, 2020-10-14:
@andyblac it's working with the python version!
I was just trying the bash one since I was being lazy and didn't want to install/update phython :P

Thank you!

andyblac, 2020-10-14:
> @andyblac it's working with the python version!
> I was just trying the bash one since I was being lazy and didn't want to install/update phython :P
> 
> Thank you!

haha, np, glad it working 😄

Ryf81, 2020-10-15:
I would love to use the script for flashing all of my devices. But to be honest, i'm not really a computer freak.
I downloaded the zip and unpacked the file. Then i run phyton launcher and open in the tools folder the "flash_shelly.py". After that a Terminal window was opened. But nothing happen if i add any of the command in the wiki like ./flash_shelly.py -la . (Xcode is also installed.)

What did i wrong?

 A short video tutorial would be great with a step by step manual.

andyblac, 2020-10-15:
ok try `python3 --version`

Ryf81, 2020-10-15:
stefan-ryfs-imac:tools stefanryf$ python3 --version
Python 3.8.3
stefan-ryfs-imac:tools stefanryf$

andyblac, 2020-10-15:
try this one should tell me which url is failing
[flash_shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5386103/flash_shelly.py.zip)

Ryf81, 2020-10-15:
stefan-ryfs-imac:tools stefanryf$ ./flash_shelly.py -la -v=0
app
OS: Darwin
action: list
mode: homekit
do_all: True
dry_run: False
silent_run: False
mode: homekit
version: False
exclude: None
variant: False
verbose: 0
hosts: []
Failed to lookup Shelly.Cloud version information
https://api.shelly.cloud/files/firmware
stefan-ryfs-imac:tools stefanryf$ python3 --version
Python 3.8.3
stefan-ryfs-imac:tools stefanryf$

andyblac, 2020-10-15:
does url work in you web browser ?

Ryf81, 2020-10-15:
yes. if i klick on the usb the following appears

{"isok":true,"data":{"SHPLG-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHPLG-1.zip","version":"20200827-065544\/v1.8.3@4a8bc427"},"SHPLG-S":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHPLG-S.zip","version":"20200827-070306\/v1.8.3@4a8bc427"},"SHPLG-U1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHPLG-U1.zip","version":"20200827-070340\/v1.8.3@4a8bc427"},"SHPLG2-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHPLG2-1.zip","version":"20200827-070415\/v1.8.3@4a8bc427"},"SHSW-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSW-1.zip","version":"20200827-065344\/v1.8.3@4a8bc427"},"SHSW-21":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSW-21.zip","version":"20200827-065420\/v1.8.3@4a8bc427"},"SHSW-25":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSW-25.zip","version":"20200827-065456\/v1.8.3@4a8bc427"},"SHSW-PM":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSW-PM.zip","version":"20200827-070450\/v1.8.3@4a8bc427"},"SHAIR-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHAIR-1.zip","version":"20200812-092044\/v1.8.0@8acf41b0"},"SHUNI-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHUNI-1.zip","version":"20200812-092537\/v1.8.0@8acf41b0"},"SHEM":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHEM.zip","version":"20200813-140346\/v1.8.1@1b2a49be"},"SHEM-3":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHEM-3.zip","version":"20200813-140420\/v1.8.1@1b2a49be"},"SHSEN-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSEN-1_build.zip","version":"20200717-112934\/v1.7.7@a02db0b8"},"SHGS-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHGS-1.zip","version":"20200812-092502\/v1.8.0@8acf41b0"},"SHSM-01":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSM-01.zip","version":"20200812-091144\/v1.8.0@8acf41b0"},"SHHT-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHHT-1.zip","version":"20200812-091311\/v1.8.0@8acf41b0"},"SHWT-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHWT-1.zip","version":"20200812-091355\/v1.8.0@8acf41b0"},"SHDW-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHDW-1.zip","version":"20200818-120325\/v1.8.2@36539b0b","beta_url":"http:\/\/api.shelly.cloud\/firmware\/rc\/SHDW-1.zip","beta_ver":"20200814-174756\/v1.8.2-rc1@59a25feb"},"SHDW-2":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHDW-2.zip","version":"20200818-120407\/v1.8.2@36539b0b","beta_url":"http:\/\/api.shelly.cloud\/firmware\/rc\/SHDW-2.zip","beta_ver":"20200814-174839\/v1.8.2-rc1@59a25feb"},"SHSPOT-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHSPOT-1.zip","version":"20200812-090754\/v1.8.0@8acf41b0"},"SHCL-255":{"url":"http:\/\/api.shelly.cloud\/firmware\/compatibility\/SHCL-255\/0.2.0-73a40eef.zip","version":"20180116-161402\/v0.2.0@73a40eef"},"SHBLB-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHBLB-1.zip","version":"20200812-090503\/v1.8.0@8acf41b0"},"SHCB-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHCB-1.zip","version":"20200812-090429\/v1.8.0@8acf41b0"},"SHRGBW2":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHRGBW2.zip","version":"20200827-070122\/v1.8.3@4a8bc427"},"SHRGBWW-01":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHRGBWW-01.zip","version":"20200827-065125\/v1.8.3@4a8bc427"},"SH2LED-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SH2LED-1.zip","version":"20200812-090719\/v1.8.0@8acf41b0"},"SHDM-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHDM-1.zip","version":"20201007-085347\/v1.8.5@5fbcbaa3"},"SHDM-2":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHDM-2.zip","version":"20201007-085431\/v1.8.5@5fbcbaa3"},"SHDIMW-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHDIMW-1.zip","version":"20200812-092244\/v1.8.0@8acf41b0"},"SHVIN-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHVIN-1.zip","version":"20200827-065016\/v1.8.3@4a8bc427"},"SHBDUO-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHBDUO-1.zip","version":"20200827-065050\/v1.8.3@4a8bc427"},"SHBTN-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHBTN-1.zip","version":"20200827-070047\/v1.8.3@4a8bc427"},"SHIX3-1":{"url":"http:\/\/api.shelly.cloud\/firmware\/SHIX3-1.zip","version":"20200827-070910\/v1.8.3@4a8bc427"}}}

andyblac, 2020-10-15:
ok try `pip3 install requests` and `pip3 install zeroconf`in terminal

Ryf81, 2020-10-15:
stefan-ryfs-imac:tools stefanryf$ pip3 install requests
Requirement already satisfied: requests in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (2.24.0)
Requirement already satisfied: chardet<4,>=3.0.2 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests) (3.0.4)
Requirement already satisfied: idna<3,>=2.5 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests) (2.10)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests) (1.25.10)
Requirement already satisfied: certifi>=2017.4.17 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests) (2020.6.20)
WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
stefan-ryfs-imac:tools stefanryf$ pip3 install zeroconf
Requirement already satisfied: zeroconf in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (0.28.6)
Requirement already satisfied: ifaddr>=0.1.7 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from zeroconf) (0.1.7)
WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
stefan-ryfs-imac:tools stefanryf$ ./flash_shelly.py -la -v=0
app
OS: Darwin
action: list
mode: homekit
do_all: True
dry_run: False
silent_run: False
mode: homekit
version: False
exclude: None
variant: False
verbose: 0
hosts: []
Failed to lookup Shelly.Cloud version information
https://api.shelly.cloud/files/firmware
stefan-ryfs-imac:tools stefanryf$

[95 more replies]

closed 2020-12-27 by andyblac