mongoose-os-apps/shelly-homekit #150: Shelly 1 Momentary Switch issue

freddair, 2020-09-27:
Hi, since updating from 2.0.12 to 2.1.7 (and now also 2.1.8) my momentary light switch occasionally needs to be tapped twice to initiate the switch action. After some switching, and after reboot, it again switches on the first tap as expected. After a while it again needs to be tapped twice. I guess that's a bug in the new 2.1.x Firmware?

andyblac, 2020-09-27:
I have had this too, but even before 2.0.12 I thought it was just my momentary switches, or releasing too fast.

freddair, 2020-09-27:
Now that you say that, I remind that I also had this issue some time earlier (I guess it was around April). It is for sure not a hardware issue. @rojer ?

rojer, 2020-09-28:
@freddair please install [2.2.0-beta3](http://rojer.me/files/shelly/2.2.0-beta3/) and capture logs after such misfires (add `/debug/log` to the address in the web UI).

freddair, 2020-09-29:
48118481865 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48118500912 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48119365402 shelly_input.cpp:72     Input 1: long (state 0)
48120814812 shelly_main.cpp:464     Uptime: 48120.81, conns 0/7/9, RAM: 41400, 21396 free
48128814863 shelly_main.cpp:464     Uptime: 48128.81, conns 0/7/9, RAM: 41400, 21396 free
48136416807 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000007 firmware.revision] Calling read handler.
48136435224 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000008 hardware.revision] Calling read handler.
48136489434 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000009 adk-version] Calling read handler.
48136580087 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000001012 version] Calling read handler.
48136639975 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000101 name] Calling read handler.
48136691064 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48136815029 shelly_main.cpp:464     Uptime: 48136.81, conns 0/7/9, RAM: 41400, 21420 free
48144814881 shelly_main.cpp:464     Uptime: 48144.81, conns 0/7/9, RAM: 41400, 22088 free
48152814807 shelly_main.cpp:464     Uptime: 48152.81, conns 0/7/9, RAM: 41400, 22088 free
48160814976 shelly_main.cpp:464     Uptime: 48160.81, conns 0/7/9, RAM: 41400, 22088 free
48168814787 shelly_main.cpp:464     Uptime: 48168.81, conns 0/7/9, RAM: 41400, 22088 free
48176814868 shelly_main.cpp:464     Uptime: 48176.81, conns 0/7/9, RAM: 41400, 22088 free
48184814881 shelly_main.cpp:464     Uptime: 48184.81, conns 0/7/9, RAM: 41400, 22088 free
48192814701 shelly_main.cpp:464     Uptime: 48192.81, conns 0/7/9, RAM: 41400, 22088 free
48200815176 shelly_main.cpp:464     Uptime: 48200.81, conns 0/7/9, RAM: 41400, 22088 free
48208814840 shelly_main.cpp:464     Uptime: 48208.81, conns 0/7/9, RAM: 41400, 22088 free
48216814840 shelly_main.cpp:464     Uptime: 48216.81, conns 0/7/9, RAM: 41400, 22088 free
48224814877 shelly_main.cpp:464     Uptime: 48224.81, conns 0/7/9, RAM: 41400, 22088 free
48232815028 shelly_main.cpp:464     Uptime: 48232.81, conns 0/7/9, RAM: 41400, 22088 free
48240814864 shelly_main.cpp:464     Uptime: 48240.81, conns 0/7/9, RAM: 41400, 22088 free
48241600370 shelly_input.cpp:72     Input 1: change (state 0)
48242328463 shelly_input.cpp:72     Input 1: change (state 1)
48242337551 shelly_output.cpp:52    Output 1: on -> off (button)
48242370734 mgos_sys_config.c:232   Loading conf2.json
48242512294 mgos_sys_config.c:174   Saved to conf9.json
48242539953 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48242563670 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48242616308 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48242636762 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48242684312 shelly_input.cpp:72     Input 1: change (state 0)
48243027193 shelly_input.cpp:72     Input 1: single (state 0)
48243986572 shelly_input.cpp:72     Input 1: change (state 1)
48243995675 shelly_output.cpp:52    Output 1: off -> on (button)
48244027734 mgos_sys_config.c:232   Loading conf2.json
48244174686 mgos_sys_config.c:174   Saved to conf9.json
48244239668 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48244259241 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48244279612 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48244299172 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48245225750 shelly_input.cpp:72     Input 1: long (state 0)
48246270267 shelly_input.cpp:72     Input 1: change (state 0)
48246952633 shelly_input.cpp:72     Input 1: change (state 1)
48246962696 shelly_output.cpp:52    Output 1: on -> off (button)
48246996183 mgos_sys_config.c:232   Loading conf2.json
48247169737 mgos_sys_config.c:174   Saved to conf9.json
48247199675 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48247220089 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48247239678 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48247260081 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48247305192 shelly_input.cpp:72     Input 1: change (state 0)
48247685049 shelly_input.cpp:72     Input 1: single (state 0)
48248591640 shelly_input.cpp:72     Input 1: change (state 1)
[... cut ...]
48254177555 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
48254262056 shelly_input.cpp:72     Input 1: change (state 0)
48254583669 shelly_input.cpp:72     Input 1: single (state 0)
48256814720 shelly_main.cpp:464     Uptime: 48256.81, conns 0/7/9, RAM: 41400, 22088 free
48262704825 mgos_http_server.c:180  0x3fff6db4 HTTP connection from 10.0.0.37:51923
48262719302 mgos_http_server.c:180  0x3fff587c HTTP connection from 10.0.0.37:51924

Ludelag, 2020-09-29:
what are the changes in beta 3 2.2.0?

rojer, 2020-09-29:
@freddair ok, now please describe what you were doing, what was happening and what you expected to happen.
one weird thing i see is that events always seem to start with 1 -> 0 transition end with 0 -> 1.
do you have a switch that is normally closed and opens when pressed?

@Ludelag major new feature is roller shutter mode (#5), but it's also an almost complete rewrite of all other functionality.

freddair, 2020-09-30:
one more try. I find it difficult to read/understand the log. What I did:
- load log web page
- tap switch
- reload log web page
- tap switch
and so on.

The Shelly switches not every time when I tap the wall switch. I don't know if it is normally closed and opens when pressed. But I do know that it should work as expected anyway: The light should switch on/off every time when the switch is pressed. The only difference would be if it switches the light either on press or on release of the switch.

```
141010088329 mgos_http_server.c:180  0x3fff76a4 HTTP connection from 10.0.0.37:55730
141016346164 shelly_input.cpp:72     Input 1: change (state 1)
141016355284 shelly_output.cpp:52    Output 1: off -> on (button)
141016387862 mgos_sys_config.c:232   Loading conf2.json
141016572475 mgos_sys_config.c:174   Saved to conf9.json
141016599455 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141016619200 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141016639762 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141016659499 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141017534179 shelly_main.cpp:464     Uptime: 141017.53, conns 0/8/9, RAM: 41400, 18328 free
141017589111 shelly_input.cpp:72     Input 1: long (state 0)
141018392920 mgos_http_server.c:180  0x3fff76a4 HTTP connection from 10.0.0.37:55731
141025533945 shelly_main.cpp:464     Uptime: 141025.53, conns 0/8/9, RAM: 41400, 18372 free```
141025595409 shelly_input.cpp:72     Input 1: change (state 0)
141027513315 mgos_http_server.c:180  0x3fff76a4 HTTP connection from 10.0.0.37:55732
141030659652 shelly_input.cpp:72     Input 1: change (state 1)
141030668778 shelly_output.cpp:52    Output 1: on -> off (button)
141030701222 mgos_sys_config.c:232   Loading conf2.json
141030844747 mgos_sys_config.c:174   Saved to conf9.json
141030878790 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141030932248 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141030952819 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141030972563 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141031864932 shelly_input.cpp:72     Input 1: long (state 0)
141031958278 mgos_http_server.c:180  0x3fff76a4 HTTP connection from 10.0.0.37:55733
141033533894 shelly_main.cpp:464     Uptime: 141033.53, conns 0/8/9, RAM: 41400, 18180 free
141035737111 mgos_http_server.c:180  0x3fff7874 HTTP connection from 10.0.0.37:55734
141034721702 shelly_input.cpp:72     Input 1: change (state 0)
141037795700 shelly_input.cpp:72     Input 1: change (state 1)
141037805883 shelly_output.cpp:52    Output 1: off -> on (button)
141037874279 mgos_sys_config.c:232   Loading conf2.json
141038022305 mgos_sys_config.c:174   Saved to conf9.json
141038053299 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141038073946 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141038093761 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141038114384 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141039038170 shelly_input.cpp:72     Input 1: long (state 0)
141039454216 mgos_http_server.c:180  0x3fff7874 HTTP connection from 10.0.0.37:55735
141041534063 shelly_main.cpp:464     Uptime: 141041.53, conns 0/8/9, RAM: 41400, 17988 free
141041943629 shelly_input.cpp:72     Input 1: change (state 0)
141043777823 mgos_http_server.c:180  0x3fff7874 HTTP connection from 10.0.0.37:55736
141045260414 shelly_input.cpp:72     Input 1: change (state 1)
141045269570 shelly_output.cpp:52    Output 1: on -> off (button)
141045303226 mgos_sys_config.c:232   Loading conf2.json
141045446549 mgos_sys_config.c:174   Saved to conf9.json
141045507340 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141045527111 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141045547695 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141045568288 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141046496993 shelly_input.cpp:72     Input 1: long (state 0)
141048421892 mgos_http_server.c:180  0x3fff7874 HTTP connection from 10.0.0.37:55737
141049534045 shelly_main.cpp:464     Uptime: 141049.53, conns 0/8/9, RAM: 41400, 17988 free
141057533934 shelly_main.cpp:464     Uptime: 141057.53, conns 0/8/9, RAM: 41400, 18368 free
141065534130 shelly_main.cpp:464     Uptime: 141065.53, conns 0/8/9, RAM: 41400, 18560 free
141065977079 shelly_input.cpp:72     Input 1: change (state 0)
[... cut ...]
141082027976 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141082048585 HAPPlatformLog.c:65     Characteristic [0000000000000001 shelly1-8CAAB5061CC9] [0000000000000102 on] Calling read handler.
141082974391 shelly_input.cpp:72     Input 1: long (state 0)
141084844957 mgos_http_server.c:180  0x3fff7874 HTTP connection from 10.0.0.37:55741
141089534220 shelly_main.cpp:464     Uptime: 141089.53, conns 0/8/9, RAM: 41400, 18180 free
141095032601 mgos_http_server.c:180  0x3fff7334 HTTP connection from 10.0.0.37:55742
```

freddair, 2020-10-08:
@rojer can you see any hints in my logs? Otherwise I would switch to HAA for now with this momentary switching Shelly1.

rojer, 2020-10-08:
yes, i think your switch is wired so that when you press it, it interrupts the circuit and thus goes from 1 to 0.
this confuses shelly, since it expects the opposite.
i can add an option to make that adjustable.

rojer, 2020-10-08:
```
141049534045 shelly_main.cpp:464     Uptime: 141049.53, conns 0/8/9, RAM: 41400, 17988 free
141057533934 shelly_main.cpp:464     Uptime: 141057.53, conns 0/8/9, RAM: 41400, 18368 free
141065534130 shelly_main.cpp:464     Uptime: 141065.53, conns 0/8/9, RAM: 41400, 18560 free
141065977079 shelly_input.cpp:72     Input 1: change (state 0)
141068296340 mgos_http_server.c:180  0x3fff7334 HTTP connection from 10.0.0.37:55738
141071505384 shelly_input.cpp:72     Input 1: change (state 1)
```

there's no activity for some time, then input changes from 1 to 0, then after 5 seconds from 0 to 1 again - this looks like your press, and it's the opposite of what we expect (we expect 0->1 when pressed, then 1->0 when released).

freddair, 2020-10-09:
I verified that the momentary switch is wired normally so that it is normally interrupted (0) and only closed when pressed: in toggle mode the Relais is switched on when I press the switch, and off when released. When I switch it on via Home app, it only switches off after pressing and with releasing the switch.

freddair, 2020-10-10:
When reverting to stock firmware, it just works fine. I think there is a bug that in momentary switch setting detects "single" or "long" states which should only be used for detached stateless button type if I'm right. have a look at these logs:

that's how it should be:
```
48241600370 shelly_input.cpp:72     Input 1: change (state 0)
48242328463 shelly_input.cpp:72     Input 1: change (state 1)
48242337551 shelly_output.cpp:52    Output 1: on -> off (button)
```

But here it detects "Input 1: **long** (state 0)" when pressing the switch and nothing happens. It should have detected "Input 1: change (state 1)" instead. At the second push, it switches.
```
48245225750 shelly_input.cpp:72     Input 1: long (state 0)
48246270267 shelly_input.cpp:72     Input 1: change (state 0)
48246952633 shelly_input.cpp:72     Input 1: change (state 1)
48246962696 shelly_output.cpp:52    Output 1: on -> off (button)
```

here is another moment and this time it detects Input 1: **single** (state 0)" when pressing the switch and nothing happens. It should have detected "Input 1: change (state 1)" instead. At the second push, it switches.
```
48242684312 shelly_input.cpp:72     Input 1: change (state 0)
48243027193 shelly_input.cpp:72     Input 1: single (state 0)
48243986572 shelly_input.cpp:72     Input 1: change (state 1)
48243995675 shelly_output.cpp:52    Output 1: off -> on (button)
```

andyblac, 2020-10-18:
any update on this?

freddair, 2020-10-18:
I reverted to stock where it works - but I would love to have it back in HomeKit... I think it is clear that it's a firmware issue. I also tested it with another Shelly and it had the same bug - so no hardware issue.

andyblac, 2020-10-18:
i get it few times a week, it seems totally random.

andyblac, 2020-10-18:
@rojer how can we help track this bug down ?

rojer, 2020-10-18:
i doubt that it's the same issue. @andyblac i'll need a log from you when the misfire happens.
@freddair's seems to be a clear-cut case of inverted switch to me. i'll prepare a test firmware with inverted action and let's see if that helps.

andyblac, 2020-10-18:
@rojer ok, i'll enable debug on all my devices. thx.

rojer, 2020-10-18:
@freddair here's a quick hack to invert input states - https://rojer.me/files/shelly/2.3.1-beta2-b150/
please install and let me know if it works better for you.

andyblac, 2020-10-18:
@rojer here is a log, turned on light it did come on, press switch to turn off, nothing happened, press switch again light turned off.

[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5397266/log.txt)

freddair, 2020-10-18:
Hi rojer , thank you, I am just a little hesitant to install this "quick hack" Beta for inverted input states (is it safe?) as I am still sure it can't be the issue as described before. To put it another way, see my photo of the Eltako relay that I replaces with the Shelly - it only works with power impulses (= input state 1).

Before:
[image]

After:
[image]

I'll install it anyway and report back...

andyblac, 2020-10-18:
@freddair does my issue seem like yours?

andyblac, 2020-10-18:
There is deffinatly something strange going on.

andyblac, 2020-10-18:
just pressed button again, nothing happened

```
430029742579 shelly_main.cpp:471     Uptime: 430029.69, conns 0/7/9 ns 7, RAM: 38912, 20620 free
430037700210 shelly_main.cpp:471     Uptime: 430037.69, conns 0/7/9 ns 7, RAM: 38912, 19048 free
430045700063 shelly_main.cpp:471     Uptime: 430045.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430053700041 shelly_main.cpp:471     Uptime: 430053.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430061700083 shelly_main.cpp:471     Uptime: 430061.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430069700183 shelly_main.cpp:471     Uptime: 430069.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430077700005 shelly_main.cpp:471     Uptime: 430077.69, conns 0/7/9 ns 7, RAM: 38912, 19048 free
430085700007 shelly_main.cpp:471     Uptime: 430085.69, conns 0/7/9 ns 7, RAM: 38912, 19048 free
430093699999 shelly_main.cpp:471     Uptime: 430093.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430101700092 shelly_main.cpp:471     Uptime: 430101.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430109700102 shelly_main.cpp:471     Uptime: 430109.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430117700233 shelly_main.cpp:471     Uptime: 430117.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430125700092 shelly_main.cpp:471     Uptime: 430125.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430133700077 shelly_main.cpp:471     Uptime: 430133.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430141700005 shelly_main.cpp:471     Uptime: 430141.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430149700051 shelly_main.cpp:471     Uptime: 430149.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430157699988 shelly_main.cpp:471     Uptime: 430157.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430165700019 shelly_main.cpp:471     Uptime: 430165.69, conns 0/7/9 ns 7, RAM: 38912, 20628 free
430173700067 shelly_main.cpp:471     Uptime: 430173.69, conns 0/7/9 ns 7, RAM: 38912, 20588 free
430173713000 shelly_input.cpp:72     Input 1: change (state 0)
430181700199 shelly_main.cpp:471     Uptime: 430181.69, conns 0/7/9 ns 7, RAM: 38912, 19052 free
430186797476 mgos_http_server.c:180  0x3fff5c34 HTTP connection from 192.168.3.20:56571
```

rojer, 2020-10-18:
@freddair yes, it should be fine. giev it a go, the only change is the inverted state of input processing and even if it's wrong, you shouldn't see any adverse effects, at most the button will not be working (and it's already wonky, so...). you can install 2.3.0 back immediately.

@andyblac ok, let's pull apart your log:

first 0 -> 1 transition, turns output on.
```
429248725754 shelly_input.cpp:72     Input 1: change (state 1)
429248733637 shelly_output.cpp:56    Output 1: off -> on (button)
429248759422 mgos_sys_config.c:232   Loading conf2.json
429248867210 mgos_sys_config.c:174   Saved to conf9.json
429248880862 shelly_switch.cpp:185   1: Set auto-off timer for 1200.000
```
1.1 second later - long press detection, though output is already 0.
```
429249889385 shelly_input.cpp:72     Input 1: long (state 0)
```

2 seconds later - change of state to 0.
```
429251888675 shelly_input.cpp:72     Input 1: change (state 0)
```

3 seconds later - another press
```
429254865622 shelly_input.cpp:72     Input 1: change (state 1)
429254872943 shelly_output.cpp:56    Output 1: on -> off (button)
429254898340 mgos_sys_config.c:232   Loading conf2.json
429255006142 mgos_sys_config.c:174   Saved to conf9.json
429255020441 shelly_switch.cpp:185   1: Set auto-off timer for 1200.000
429256028314 shelly_input.cpp:72     Input 1: long (state 0)
```

rojer, 2020-10-18:
@andyblac , we don't seem to be detecting return from 1 to 0. these are short presses you make, right?

andyblac, 2020-10-18:
@rojer i only pressed twice, with no long preses.

andyblac, 2020-10-18:
> @andyblac , we don't seem to be detecting return from 1 to 0. these are short presses you make, right?

yes, i am noy doing any long presses at all.

rojer, 2020-10-18:
ok, i think i know what might be wrong.
@andyblac please try this one - https://rojer.me/files/shelly/2.3.1-beta2-b150_nosave/
here i turned off saving switch state on every change, i think it might be why we miss state transitions.

andyblac, 2020-10-18:
> ok, i think i know what might be wrong.
> @andyblac please try this one - https://rojer.me/files/shelly/2.3.1-beta2-b150_nosave/
> here i turned off saving switch state on every change, i think it might be why we miss state transitions.

ok, flashing now.

andyblac, 2020-10-18:
log. i did 3 short press, 2nd one did nothing.

[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5397291/log.txt)

andyblac, 2020-10-18:
does seem better, maybe that one was a switch issue, and not a shelly issue. i as i do only see 2 presses in log.

rojer, 2020-10-18:
yeah, i only see 2 transitions and the 1 -> 0 detections does seem a lot better.
try some more, let me know if it's better on average.

andyblac, 2020-10-18:
yeah it deffinatly better
[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5397296/log.txt)

andyblac, 2020-10-18:
i'll flash the others. report if any issue.

freddair, 2020-10-18:
> @freddair here's a quick hack to invert input states - https://rojer.me/files/shelly/2.3.1-beta2-b150/
> please install and let me know if it works better for you.

@rojer I installed your beta and it now worked flawlessly. I am puzzled. Here is the log:

```
472316173 shelly_main.cpp:471     Uptime: 472.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
474783017 shelly_input.cpp:72     Input 1: change (state 0)
474946097 shelly_input.cpp:72     Input 1: change (state 1)
474956050 shelly_output.cpp:56    Output 1: on -> off (button)
474988802 mgos_sys_config.c:232   Loading conf2.json
475012736 mgos_sys_config.c:232   Loading conf3.json
475141130 mgos_sys_config.c:174   Saved to conf9.json
476157773 shelly_input.cpp:72     Input 1: long (state 1)
477881601 shelly_input.cpp:72     Input 1: change (state 0)
478042548 shelly_input.cpp:72     Input 1: change (state 1)
478051525 shelly_output.cpp:56    Output 1: off -> on (button)
478085008 mgos_sys_config.c:232   Loading conf2.json
478108952 mgos_sys_config.c:232   Loading conf3.json
478237519 mgos_sys_config.c:174   Saved to conf9.json
479253370 shelly_input.cpp:72     Input 1: long (state 1)
480316468 shelly_main.cpp:471     Uptime: 480.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
482157155 shelly_input.cpp:72     Input 1: change (state 0)
482347874 shelly_input.cpp:72     Input 1: change (state 1)
482356861 shelly_output.cpp:56    Output 1: on -> off (button)
482389310 mgos_sys_config.c:232   Loading conf2.json
482414229 mgos_sys_config.c:232   Loading conf3.json
482542793 mgos_sys_config.c:174   Saved to conf9.json
483558487 shelly_input.cpp:72     Input 1: long (state 1)
486245672 shelly_input.cpp:72     Input 1: change (state 0)
486453009 shelly_input.cpp:72     Input 1: change (state 1)
486462006 shelly_output.cpp:56    Output 1: off -> on (button)
486494369 mgos_sys_config.c:232   Loading conf2.json
486518330 mgos_sys_config.c:232   Loading conf3.json
486648048 mgos_sys_config.c:174   Saved to conf9.json
487663875 shelly_input.cpp:72     Input 1: long (state 1)
488316068 shelly_main.cpp:471     Uptime: 488.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
496316001 shelly_main.cpp:471     Uptime: 496.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
503454205 shelly_input.cpp:72     Input 1: change (state 0)
503619702 shelly_input.cpp:72     Input 1: change (state 1)
503628692 shelly_output.cpp:56    Output 1: on -> off (button)
503660980 mgos_sys_config.c:232   Loading conf2.json
503685910 mgos_sys_config.c:232   Loading conf3.json
503814665 mgos_sys_config.c:174   Saved to conf9.json
504319684 shelly_main.cpp:471     Uptime: 504.31, conns 0/8/9 ns 8, RAM: 38912, 17472 free
504836297 shelly_input.cpp:72     Input 1: long (state 1)
504869445 shelly_input.cpp:72     Input 1: change (state 0)
505067327 shelly_input.cpp:72     Input 1: change (state 1)
505076323 shelly_output.cpp:56    Output 1: off -> on (button)
505108524 mgos_sys_config.c:232   Loading conf2.json
505133446 mgos_sys_config.c:232   Loading conf3.json
505262356 mgos_sys_config.c:174   Saved to conf9.json
506252016 shelly_input.cpp:72     Input 1: change (state 0)
506260965 shelly_input.cpp:72     Input 1: single (state 0)
506447112 shelly_input.cpp:72     Input 1: change (state 1)
506457096 shelly_output.cpp:56    Output 1: on -> off (button)
506489226 mgos_sys_config.c:232   Loading conf2.json
506513187 mgos_sys_config.c:232   Loading conf3.json
506642113 mgos_sys_config.c:174   Saved to conf9.json
507589436 shelly_input.cpp:72     Input 1: change (state 0)
507598385 shelly_input.cpp:72     Input 1: single (state 0)
507769783 shelly_input.cpp:72     Input 1: change (state 1)
507778790 shelly_output.cpp:56    Output 1: off -> on (button)
507811815 mgos_sys_config.c:232   Loading conf2.json
507835781 mgos_sys_config.c:232   Loading conf3.json
507964888 mgos_sys_config.c:174   Saved to conf9.json
508981007 shelly_input.cpp:72     Input 1: long (state 1)
512316495 shelly_main.cpp:471     Uptime: 512.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
520316247 shelly_main.cpp:471     Uptime: 520.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
528316054 shelly_main.cpp:471     Uptime: 528.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
535157788 shelly_input.cpp:72     Input 1: change (state 0)
535373329 shelly_input.cpp:72     Input 1: change (state 1)
535382344 shelly_output.cpp:56    Output 1: on -> off (button)
535414327 mgos_sys_config.c:232   Loading conf2.json
535438309 mgos_sys_config.c:232   Loading conf3.json
535568414 mgos_sys_config.c:174   Saved to conf9.json
536316216 shelly_main.cpp:471     Uptime: 536.31, conns 0/8/9 ns 8, RAM: 38912, 21232 free
536584110 shelly_input.cpp:72     Input 1: long (state 1)
544316112 shelly_main.cpp:471     Uptime: 544.31, conns 0/8/9 ns 8, RAM: 38912, 21272 free
```

rojer, 2020-10-18:
yep, that does seem much better in the log too.
flash writes are expensive... but we do need to save state.
ok, let me come up with a proper solution.

@freddair please try the -nosave firmware too, let me know if it solves your issue.

rojer, 2020-10-18:
@andyblac i cleaned up the nosave hack and pushed https://rojer.me/files/shelly/2.3.1-beta3/ please test
@freddair please also test beta3, let me know if it solves the issue

andyblac, 2020-10-18:
> @andyblac i cleaned up the nosave hack and pushed https://rojer.me/files/shelly/2.3.1-beta3/ please test
> @freddair please also test beta3, let me know if it solves the issue

yup, seems ok.
[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5397537/log.txt)

andyblac, 2020-10-19:
no issues so far on 2.3.1, thanks @rojer

kzizi, 2020-10-21:
Same here, works perfect! Thanks!!

freddair, 2020-10-21:
No problems on my side, too - thanks @rojer !

I'm curious - it wasn't inverted switches, was it?

andyblac, 2020-10-21:
> No problems on my side, too - thanks @rojer !
> 
> I'm curious - it wasn't inverted switches, was it?

see here 
> ok, i think i know what might be wrong.
> @andyblac please try this one - https://rojer.me/files/shelly/2.3.1-beta2-b150_nosave/
> here i turned off saving switch state on every change, i think it might be why we miss state transitions.

rojer, 2020-10-21:
@freddair yeah, it looked a lot like it but ended up being something different.

thanks for confirming, closing the issue.

closed 2020-10-21 by rojer