mongoose-os-apps/shelly-homekit #151: Blinking problem

MilanBacaCz, 2020-09-27:
Hi, i have a problem with two of my Shellys 1. Sometime it can't turn off. Instead of permanently OFF I see attempt  to ON every ~2s interval. Is there any way how to see logs of shelly if this behavior occurs?

andyblac, 2020-10-02:
logging info has been adding to 2.2.0 betas, will be released soon, but you can try beta8 here https://rojer.me/files/shelly/2.2.0-beta8/shelly-homekit-Shelly25.zip
enable it in UI then go to http://yourip/debug/log

andyblac, 2020-10-02:
fixed link above for shelly1

MilanBacaCz, 2020-10-03:
Blinking is back. Lopg is attached. Can anyone help?
[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5322057/log.txt)

rojer, 2020-10-03:
```
80001760727 shelly_output.cpp:52    Output 1: on -> off (HAP)
... 2.9 s later ...
80004653456 shelly_output.cpp:52    Output 1: off -> on (HAP)
... 0.5 s later ...
80005184154 shelly_output.cpp:52    Output 1: on -> off (HAP)
```

these are state transitions requested by HAP (i.e. by HomeKit controller). not sure which one, you have 4 connected at the time.
i can add some debugging to log which controller requested the change

MilanBacaCz, 2020-10-03:
Please do it. My wife is angry with me about this blinking.

S pozdravem,
Milan Bača

MilanBacaCz, 2020-10-03:
Now this shelly one works only from its web ui. Not works from apple Homekit. 
Log added.
[log2.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5322140/log2.txt)

rojer, 2020-10-03:
192.168.0.119 is not happy, constantly reconnects. can you try rebooting the device?

MilanBacaCz, 2020-10-03:
rebooted. not helped
[log3.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5322146/log3.txt)

rojer, 2020-10-03:
wait for things to settle a bit and for all the devices to reconnect.

MilanBacaCz, 2020-10-03:
waited and still there is a issue. Web ui works perfectly. But Homekit all times loose connection.
[log4.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5322166/log4.txt)
I am thinking we need more log points to see what is happened.

MilanBacaCz, 2020-10-04:
This is log from my second shelly1 which has this issue.
[log5.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5323458/log5.txt)

MilanBacaCz, 2020-10-07:
We can close this issue. Ii is happening even with stock software. So it is not issue of this firmware.

rojer, 2020-10-07:
i see. ok, thanks for letting me know!

closed 2020-10-07 by rojer