mongoose-os-apps/shelly-homekit #153: Returning to a stock - test and check?

macguru341, 2020-09-28:
Hello
Used the mos way with terminal in windows to flash the original stock firmware and I want to ask if there's a way to check if it's working before I'm selling my Shelly. I can't reconnect it into the wall.

Thanks

rojer, 2020-09-28:
with serial connection you should be able to manage it on wifi, one thing that will not work is relays, so you won't hear it clicking when controlling from the  web interface for example.
i'd say, if it's available on wifi and you can access the stock web UI, you're good.

closed 2020-09-28 by rojer

macguru341, 2020-09-28:
After I've used the mos way to install the stock firmware and when trying to check my WiFi to see that I was succeed and didn't find nothing I see in the wiki that I must use the web interface as this firmware does not support pulling from remote URL like I did in mos.
How do I access the web i interface? What is the script I see and how do I use it? 
Thanks for the support

macguru341, 2020-09-28:
Also trying to use mos with link to the downloaded file and I get "all done" but I still can't see the Shelly 1 in my iPhone WiFi

macguru341, 2020-09-28:
[image]

[image]

Flashing blue

andyblac, 2020-09-28:
script is only for macOS or Linux, Windows is not supported.

andyblac, 2020-09-28:
if the device has been refreshed with stock, it should keep all settings from HomeKit firmware (wifi etc), do you see it connected to your Router. does it have an ip ?

if not try connecting to it as an Access Point from your computer. you should see 'shelly-xxxxx' as a AP.

macguru341, 2020-09-28:
1. I also have a Mac, should I use the script ? Is it better to use this method and not the mos way?
2. How do I trigger the access point ? Because I don't see the Shelly WiFi address in my iPhone WiFi list.

Thanks for the support

andyblac, 2020-09-28:
> 1. I also have a Mac, should I use the script ? Is it better to use this method and not the mos way?
> 2. How do I trigger the access point ? Because I don't see the Shelly WiFi address in my iPhone WiFi list.
> 
> Thanks for the support

script only works if it is attached to your network, but you could try it

https://github.com/mongoose-os-apps/shelly-homekit/blob/master/tools/flash_shelly.sh

in terminal try `./flash_script.sh -la` this should list all found shelly devices on the network, and report what firmware is loaded. to revert to stock to `./flash_script -m revert hostname`. hostname is the one listed shown in the list.