mongoose-os-apps/shelly-homekit #154: HomeKit section in webUI

andyblac, 2020-09-29:
in beta3 this section no longer shows ***** in the homekit code if one is set, and you can enter a new one and save button can be clicked.

IMO I think this text field should be showing **** and unable to change text, and Save button should be disabled, until reset button has been clicked.

andyblac, 2020-09-29:
[image]

schemhad, 2020-09-29:
I cloud disable the Save button and textfield when "Paired" is "yes"... 

If it isn't a bug, I could set the textfield to ***-**-*** too.

andyblac, 2020-09-29:
I'll leave this to @rojer to decide.

rojer, 2020-09-29:
@schemhad i'll fix this in the next 2.2 beta, don't worry about it.
if you are looking for a UI challenge, https://github.com/mongoose-os-apps/shelly-homekit/issues/50 is the one to tackle. will make UI more dynamic and improve UX for sure.

andyblac, 2020-09-30:
fixed in beta4

closed 2020-09-30 by andyblac