mongoose-os-apps/shelly-homekit #155: Turn AP off and after holding switch for 10sec turn AP on

maf-o, 2020-09-29:
Hello,
I use the Shelly 1 in my stairwell to trun off the lights after 50sec when button was pushed.
The main power distributor, where shelly hangs, is out of range of Wifi so get no connection.
Would be nice if i can turn on AP after holding switch/button for 10 sec, than i can adjust behavior and after 5min automatically turn AP off until i hold for next 10sec.

andyblac, 2020-09-29:
if you can't connect it to a WiFi AP as signal is too weak, why not just leave the device in AP mode all the time, ie don't set wifi settings, then you can just walk up to it with your phone connect to it's own AP and configure via 192.168.33.1 ?

maf-o, 2020-09-29:
Because of environmental polution (energy and frequenzes)
I dont need an accespoint in the basement.
I dont need an accespoint where someone tries to hack in and switching my lights.

andyblac, 2020-09-29:
I'll leave this for rojer to comment on, but imo the range of AP is very limited, and regardless if it is AP or wifi mode it still emitting as you say pollution, so sorry but that makes no sense.

maf-o, 2020-09-29:
In Arduino IDE i can switch of the Wifi of ESP. Then there is no AP or Wifi transmission.

andyblac, 2020-09-29:
so you basically want to disable the Wif totally. tbh I don't see this that something most people would use. but again I'll revert to @rojer on this matter.

rojer, 2020-09-29:
i agree, this seems too niche of an application

closed 2020-09-29 by rojer

maf-o, 2020-09-29:
can you point me where i can do the magic myself?

andyblac, 2020-09-29:
https://github.com/mongoose-os-apps/shelly-homekit/tree/master/src