mongoose-os-apps/shelly-homekit #156: Input 1

Ludelag, 2020-09-29:
I just update to version 2.1.8 and I see in the Home app input 1 not configured
Someone could please tell me how to configure input 1? and what and how to use it.

Thanks

andyblac, 2020-09-29:
long press on the tile you should something like

[image]

look at actions, tap on the action you want to add, and set as a normal homekit action.

Ludelag, 2020-09-29:
Thank you for the info, So is where I can for example wire the door magnetic sensor?, but it wont appear in the homekit as a door sensor, isn´t it?

andyblac, 2020-09-29:
> Thank you for the info, So is where I can for example wire the door magnetic sensor?, but it wont appear in the homekit as a door sensor, isn´t it?

thats not support yet, iirc.

Ludelag, 2020-09-29:
What is not supported? the door sensor wiring or how will it be showed in the apple homekit?

andyblac, 2020-09-29:
Both iirc

rojer, 2020-09-29:
the only way to represent an independent input in homekit that is currently supported is a stateless switch.

other possibilities are:
 - contact sensor
 - motion sensor 
 - occupancy sensor

this work will happen some time in the future, https://github.com/mongoose-os-apps/shelly-homekit/issues/147 will track that

closed 2020-09-29 by rojer

Ludelag, 2020-09-29:
If I got it, now is possible to represent it as:

contact sensor
motion sensor
occupancy sensor

rojer, 2020-09-29:
not yet, but it might be

Ludelag, 2020-09-29:
So, it means right now Input 1 is useless, I mean there is any action to be applied?

andyblac, 2020-09-29:
> So, it means right now Input 1 is useless, I mean there is any action to be applied?, 

yes, any homekit device to control, that is not the shelly, ie group of lights, power socket, etc etc, ie you use the physical switch to control something else.