mongoose-os-apps/shelly-homekit #161: webUI beta5 inconsistencies.

andyblac, 2020-10-01:
see below drop boxes for Shelly1 vs Shelly2.5

Shelly1
[image]

Shelly2.5
[image]

andyblac, 2020-10-01:
also Model is missing on both Shelly1 / 2.5 
[image]

andyblac, 2020-10-01:
hmm, OK ignore issue 1 , seems was a browser issue, looks ok now after a reboot.

rojer, 2020-10-01:
i think (1) is due to stale style.css, eventually it gets reloaded.

re: model - oops, yeah, added a field and forgot to populate it.

rojer, 2020-10-01:
fixed in beta6

closed 2020-10-01 by rojer

andyblac, 2020-10-01:
release notes for beta6 are linked to https://github.com/mongoose-os-apps/shelly-homekit/issues/5#issuecomment-701662473.  ie about beta5 ??

rojer, 2020-10-01:
https://github.com/mongoose-os-apps/shelly-homekit/issues/5#issuecomment-702269738

forgot to update the url, fixed now