mongoose-os-apps/shelly-homekit #166: shelly 2.5 support for rolling blinds

Tomer51m, 2020-10-02:
Hi! is there any chance shelly 2.5 will support rolling blinds? This is such a big feature for home kit users. thanks!

rojer, 2020-10-02:
it's in the final stages of beta, see #5 and please help testing.

closed 2020-10-02 by rojer

Tomer51m, 2020-10-02:
I’ll be happy to help. BTW, I’m a front end developer, and would love to help with UI for the shelly software.

andyblac, 2020-10-02:
maybe take on this issue if you could  ? https://github.com/mongoose-os-apps/shelly-homekit/issues/50

rojer, 2020-10-02:
oh, that would be handy. i am very much not a web developer and do the barest minimum necessary.
i added a section to the developemnt wiki on how to upload files without rebuilding entire firmware - https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#web-ui-development

as @andyblac mentioned, #50 is a good start.