mongoose-os-apps/shelly-homekit #169: webUI

andyblac, 2020-10-03:
just thinking about webUI and wiki, do you think it would be helpful to have a "Help" button in each section of the UI that would open a new page to the wiki of that section.

ie. 
in Switch section the word "Help" (or something else) in the top right corner that when clicked it opens a new web page the shows the "Switch Settings" section of the wiki?

rojer, 2020-10-03:
yep, had exactly the same idea. we should have "Help" links in each section.
@schemhad can you help with that, perhaps?

schemhad, 2020-10-03:
This should be easy, I'll take a look at it ...

schemhad, 2020-10-03:
[image]

Like this?

andyblac, 2020-10-04:
yeah, but maybe move dot so it inside white area.

schemhad, 2020-10-04:
[image]

andyblac, 2020-10-04:
for me personally 1t one.

schemhad, 2020-10-04:
I could customize and place it any way you like.

rojer, 2020-10-04:
i like no. 3

andyblac, 2020-10-04:
go 3 then, thats ok with me :)

andyblac, 2020-10-05:
@rojer in `Shelly 1` could tweak ui for switch setting from `Switch 1 (Shelly SW)` to `Switch (Shelly SW)`, as there is only 1 relay.

rojer, 2020-10-05:
integrated and released in 2.2.1

closed 2020-10-05 by rojer

andyblac, 2020-10-05:
> @rojer in `Shelly 1` could tweak ui for switch setting from `Switch 1 (Shelly SW)` to `Switch (Shelly SW)`, as there is only 1 relay, and same for `Input 1 (Input)` when detached is enabled.

@rojer what about this ?