mongoose-os-apps/shelly-homekit #170: Shelly Shutters - opposite

macguru341, 2020-10-03:
Hello
First of all I want to say thanks for your hard work to make it happen and use Shelly 2.5 as roller shutters .
I've installed on all of my shutters and in some strange way (even after a calibration I did) the status is opposite , when the shutter is open it's says in HomeKit that it's close , only in one of my shutters it shows the right status , how can I fix it?

rojer, 2020-10-03:
sounds like you have open on output 2 and close on output 1 then. try swapping outputs - enable "swap outputs" in the web ui.

macguru341, 2020-10-03:
Both swap are off and the actual switch is working fine. If I use the swap the actual switch will be opposite

rojer, 2020-10-03:
so you mean physical switches operate in the right direction but home app status is reversed?

macguru341, 2020-10-03:
Yep , exactly

rojer, 2020-10-03:
interesting. does partial positioning fron the app work? e.g is 25% open really 25 or more like 75?

macguru341, 2020-10-03:
Also the control in HomeKit is opposite

macguru341, 2020-10-03:
> interesting. does partial positioning fron the app work? e.g is 25% open really 25 or more like 75?
25% open is 75%

macguru341, 2020-10-03:
Maybe it's the calibration? In what position I need to start the calibration?

rojer, 2020-10-03:
ok. well, then you need to swap both outputs and inputs.

rojer, 2020-10-03:
no, you really have the device wired the opposite way. inputs are currently in sync with outputs because they are also wired backwards, so you just need to swap both.

macguru341, 2020-10-03:
Tried it , if I swap both inputs and outputs to on position the status in HomeKit stays the same and now the actual switch isn't working

rojer, 2020-10-03:
re-run calibration

macguru341, 2020-10-03:
1. So it doesn't matter in what position is the shutter when I start the calibration?
2. Also off topic, if the firmware have an update it will say in HomeKit ?

rojer, 2020-10-03:
1. yeah, you can start calibration from anywhere
2. no, it won't. there isn't (or at least i'm not aware) of any way to do that. you need to check for update manually in the web ui.

macguru341, 2020-10-03:
Did a new calibration and still in the homekit web it's says 89% and also in HomeKit Aldo it's 11% open 🤷🏻‍♂️[image]

rojer, 2020-10-04:
weird. ok, when calibrating, does it go to open position first, then close? this is how it's supposed to be.

macguru341, 2020-10-04:
No
It's first close and than open
Also the buttons in the web interface are opposite

rojer, 2020-10-04:
what happens if you enable both swap inputs and swap outputs? don't forget to press "save" button.

macguru341, 2020-10-05:
Well, one of my shutters was fixes and now it's ok if I swap both input and output but my second one lost connection now and I can't reach it , the web isn't loading and the actual switch isn't working as well, what can I do?

macguru341, 2020-10-05:
Hope you can help retrieve connection to my porch shutters as it's seems offline and my WiFi is going to kill me!!! She always says that all of this smart home things will kill one of the shutters ...please help 😏

rojer, 2020-10-05:
hm, that's the 3rd case reported of this, see #171 
unfortunately, i currently have no idea what the cause is...
it looks like i will need to have one of these devices sent to me so i can examine.

rojer, 2020-10-05:
can you tell how it happened? was it moving at the time it became unreachable?

macguru341, 2020-10-05:
I played with the swap on and off and pressed save and than it become like that

macguru341, 2020-10-05:
Ao it means the Shelly is dead ?

rojer, 2020-10-05:
i take it you tried turning power off and on for it, right?
after power is applied, you should see LED blinking. do you see any activity on the LED?

andyblac, 2020-10-05:
to me it sounds like they are caught in a boot loop, but as @rojer says without connecting one to serial it hard to tell.

rojer, 2020-10-05:
yeah. i suspect filesystem corruption, but no way to confirm or fix without a serial connection, unfortunately

macguru341, 2020-10-05:
I will replace the unit today

rojer, 2020-10-05:
can you send the bad one to me?

macguru341, 2020-10-05:
Sure , send me your address to my email : ofir341@gmail.com

rojer, 2020-10-05:
ok, i believe we have this resolved. 2.2.1 was released that fixes #171 and swapping addressed your initial issue.

closed 2020-10-05 by rojer