mongoose-os-apps/shelly-homekit #171: Shelly 2.5 no response / dead

Tigerstryke, 2020-10-04:
First of all thank you very much for the newest update. The rollershutter mode is so great. Yesterday I had a problem with my shelly 2.5 with the newest software version. Suddenly it has not responded. In my opinion its dead, because the led is not doing anything. I also tried to use it with the normal switches. Is there a way to recover it?

Montardit, 2020-10-04:
I have the same problem. I have 5 Shelly 2.5 and after the upgrade one of them is dead. I configured all of them as rolling shutters. Yesterday worked but today has no response, no web UI, no nothing.

rojer, 2020-10-04:
ok, let's try to triage this.
so, not all devices are experiencing this, correct?
second, does restarting a device help? is it available at least on the web UI after a power cycle?

Tigerstryke, 2020-10-04:
I only have one in use, so I can´t say.
A restart does not help, there is no web UI and no clicking from the relay if you use the normal switches.

Montardit, 2020-10-04:
Restarting does nothing. No web UI. Not shown in LAN devices list

rojer, 2020-10-04:
do you see any activity on the built-in LED at all?

Tigerstryke, 2020-10-04:
No Led activity.

rojer, 2020-10-05:
ok, i'm afraid next step is obtaining a flash dump of one of the shellies.
for that, a serial connection needs to be set up - see [here](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection).
alternatively, if anyone is willing to send me one for examination, reach out by email and i'll give you address.

Montardit, 2020-10-05:
Mine is in an inaccessible place. I'll try to reset it and reinstall firmware. Hope it works. But I need my time... The strange thing is it worked ok for serveral hours after installing 2.2.0. The other 4 shelly 2.5 were bought at the same time and are working fine. I think some of them had the same configuration as the ko one.

rojer, 2020-10-05:
yeah, this is a complete mystery at this point...
@Tigerstryke will be sending me one for examination.
my best guess at this point is filesystem corruption, but it's just a guess.

rojer, 2020-10-05:
i got a flash dump from a user and found the bug. it occurs when you swap inputs and set input mode to single.
in this case we should use input 2 for control, but instead we just crash. oops.
i can fix your device if you send me dump obtained over a serial connection as described here: https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection
do not flash anything yourself, once you get a dump just send it to rojrer@rojer.me and i'll send you back a fixed one.
sorry about that, i really need to think how to avoid getting device into such unrecoverable state...

Montardit, 2020-10-05:
I can't reset the device, not with the 5 consecutive switch on SW1 or SW2 nor pressing button for 10 seconds. Shelly Wifi doesn't appear anymore... :-(

rojer, 2020-10-05:
@Montardit that's right, unfortunately in this state device is completely unresponsive. you will have to connect it via serial cable as described on the wiki. i can also fix it for you if you mail it to me.
apologies again, it's a nasty bug, i know.

Montardit, 2020-10-05:
I've just ordered one usb to serial cable. Tomorrow i'll try. 
Next bug try not to be so bad with us :-p
Thanks for all, good work!!

andyblac, 2020-10-05:
> I've just ordered one usb to serial cable. Tomorrow i'll try.
> Next bug try not to be so bad with us :-p
> Thanks for all, good work!!

pins are tiny, if you have trouble finding a suitable male plug, use a female one and a short CAT5 cable with both ends striped, and push copper wire in to pin header.

andyblac, 2020-10-05:
@Montardit also make sure USB to Serial can do 3v and some are only 5v, and it will blow your shelly device.

rojer, 2020-10-05:
released 2.2.1 with the fix.

closed 2020-10-05 by rojer

wdylag, 2022-12-02:
> released 2.2.1 with the fix.

hmmm, today I upgraded my devices to the newest version but it was 2.11.2, not 2.2.1.... with the bug! no idea how can I fix my device now because is dead.