mongoose-os-apps/shelly-homekit #172: Roller shutter power selection

masafra86, 2020-10-04:
First of all, this is the greatest release since the beginning in my opinion, so, thank for that.
Then, I installed the new FW in my shelly 2.5 and set it as roller shutter. During the calibration, the window moved few millimetres and stopped, I saw on the browser that the calibration power was 2 W. So I decided to revert to the stock FW to measure how much power the motor needed for moving the windows and the shelly app said it was 11 w to close and 5.4W to close.
My question is: would it be possible to set a standard calibration power to be used? Alternatively, to set the time (in seconds for opening and closing? Thanks

rojer, 2020-10-04:
wow, that's a very low power motor. current threshold for idle is 10W, so no wonder it doesn't work properly.
thing is, i had to raise it in response to user feedback - they had a motor that uses 8W when idle.
we need a smarter calibration procedure, to figure out idle power somewhow.

meanwhile, if you have a Mac, please open terminal an enter the following command:

```
curl 'http://shelly25-test.local/rpc/Config.Set' -d '{"config": {"wc1": {"idle_power_thr": 1.0}}, "reboot": true}'
```
replace `shelly25-test` with the name of your shelly device.
this will set idle cutoff to 1W. you should get back
```
{"saved": true}
```

device will reboot, try calibrating again.

masafra86, 2020-10-04:
It worked, but partially. I mean, the motor started moving and the calibration procedure started fine. I started it when the window was fully opened. The window closed, arrived at full close position, reverted its run then, at full open position it continued saying "calibrating" on the status. I waited some minutes, then I understood it didn't recognised the position. It looked like it was continuing to open. I had to reboot it to stop the calibration process. 
After trying some times, I tried modifying the threshold in the string you wrote. 2.0 ; 5.0; 10.0; 20.0; 0.5. The result was always the same: at full open position, the calibration was still in process.

rojer, 2020-10-04:
please post log of the process.

masafra86, 2020-10-05:
I'm having some problems on using mos. I'm not an expert (I'm studying on it) and I can't make the UDP logging... (embarrassed)

masafra86, 2020-10-07:
Is this the log you were asking for?

14462119324:1602101424334274 mgos_sys_config.c:232   Loading conf2.json
14462143256 mgos_sys_config.c:232   Loading conf3.json
14462276204 mgos_sys_config.c:174   Saved to conf9.json
14462315407 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60121
14462335557 mgos_mongoose.c:66      New heap free LWM: 17568
14464623028 shelly_main.cpp:471     Uptime: 14464.60, conns 0/0/9 ns 0, RAM: 38760, 23024 free
14472613450 shelly_main.cpp:471     Uptime: 14472.60, conns 0/0/9 ns 0, RAM: 38760, 21980 free
14480613425 shelly_main.cpp:471     Uptime: 14480.60, conns 0/0/9 ns 0, RAM: 38760, 22176 free
14488613422 shelly_main.cpp:471     Uptime: 14488.60, conns 0/0/9 ns 0, RAM: 38760, 23432 free
14496613387 shelly_main.cpp:471     Uptime: 14496.60, conns 0/0/9 ns 0, RAM: 38760, 23452 free
14504613386 shelly_main.cpp:471     Uptime: 14504.60, conns 0/0/9 ns 0, RAM: 38760, 23004 free
14507957083 mgos_http_server.c:180  0x3fff5f34 HTTP connection from 192.168.1.6:60131
14507971859 mg_rpc.c:292            Shelly.SetConfig via HTTP 192.168.1.6:60131
14508006310 mgos_sys_config.c:232   Loading conf2.json
14508030249 mgos_sys_config.c:232   Loading conf3.json
14508163098 mgos_sys_config.c:174   Saved to conf9.json
14508188999 shelly_hap_window_c:293 WC 1: State: idle -> precal0 (0 -> 10)
14508288665 shelly_output.cpp:52    Output 1: on -> off (precal0)
14508296585 shelly_hap_window_c:386 Begin calibration
14508328706 mgos_sys_config.c:232   Loading conf2.json
14508352633 mgos_sys_config.c:232   Loading conf3.json
14508486452 mgos_sys_config.c:174   Saved to conf9.json
14508501643 shelly_output.cpp:52    Output 1: off -> on (precal0)
14508511687 shelly_hap_window_c:293 WC 1: State: precal0 -> cal0 (10 -> 11)
14508527331 shelly_hap_window_c:402 WC 1: P0 = 0.000
14509327502 shelly_hap_window_c:402 WC 1: P0 = 1.122
14509377426 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14509598977 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14509927925 shelly_output.cpp:52    Output 1: on -> off (cal0)
14509939030 shelly_hap_window_c:293 WC 1: State: cal0 -> postcal0 (11 -> 12)
14510028804 shelly_hap_window_c:293 WC 1: State: postcal0 -> precal1 (12 -> 13)
14510127443 shelly_output.cpp:52    Output 2: off -> on (precal1)
14510138465 shelly_hap_window_c:293 WC 1: State: precal1 -> cal1 (13 -> 14)
14510227757 shelly_hap_window_c:433 WC 1: P1 = 2.805
14510632557 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14510653890 mgos_mongoose.c:66      New heap free LWM: 17528
14511027557 shelly_hap_window_c:433 WC 1: P1 = 1.360
14511654298 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14511827785 shelly_hap_window_c:433 WC 1: P1 = 6.110
14512599449 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14512625161 shelly_main.cpp:471     Uptime: 14512.62, conns 0/0/9 ns 0, RAM: 38760, 21788 free
14512636924 shelly_hap_window_c:433 WC 1: P1 = 5.049
14513427595 shelly_hap_window_c:433 WC 1: P1 = 5.091
14513601918 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14514227532 shelly_hap_window_c:433 WC 1: P1 = 5.445
14514606689 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14515027683 shelly_hap_window_c:433 WC 1: P1 = 5.037
14515604294 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14515827530 shelly_hap_window_c:433 WC 1: P1 = 4.537
14516606036 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14516629303 shelly_hap_window_c:433 WC 1: P1 = 4.134
14517427543 shelly_hap_window_c:433 WC 1: P1 = 4.024
14517615225 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14518227547 shelly_hap_window_c:433 WC 1: P1 = 4.024
14518633438 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14519027617 shelly_hap_window_c:433 WC 1: P1 = 4.232
14519611972 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14519827676 shelly_hap_window_c:433 WC 1: P1 = 4.250
14520610920 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14520638734 shelly_main.cpp:471     Uptime: 14520.63, conns 0/0/9 ns 0, RAM: 38760, 21896 free
14520650608 shelly_hap_window_c:433 WC 1: P1 = 7.244
14521427581 shelly_hap_window_c:433 WC 1: P1 = 1.396
14521609272 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14522227730 shelly_hap_window_c:433 WC 1: P1 = 1.494
14522633779 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14523027637 shelly_hap_window_c:433 WC 1: P1 = 1.366
14523622204 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60131
14523827558 shelly_hap_window_c:433 WC 1: P1 = 1.360
[... 43 lines cut ...]
14552613623 shelly_main.cpp:471     Uptime: 14552.60, conns 0/0/9 ns 0, RAM: 38760, 23444 free
14552627705 shelly_hap_window_c:433 WC 1: P1 = 1.384
14553079785 mgos_http_server.c:180  0x3fff5eec HTTP connection from 192.168.1.6:60191
14553098177 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
14553427571 shelly_hap_window_c:433 WC 1: P1 = 1.366
14554067144 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
14554227607 shelly_hap_window_c:433 WC 1: P1 = 1.348
14555027369 shelly_hap_window_c:433 WC 1: P1 = 1.354
14555052988 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
14555827589 shelly_hap_window_c:433 WC 1: P1 = 1.445
14556050095 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
[... cut ...]
14559141255 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
14559827469 shelly_hap_window_c:433 WC 1: P1 = 1.323
14560091169 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
14560613544 shelly_main.cpp:471     Uptime: 14560.60, conns 0/0/9 ns 0, RAM: 38760, 22936 free
14560627662 shelly_hap_window_c:433 WC 1: P1 = 1.488
14561057208 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:60191
14561427572 shelly_hap_window_c:433 WC 1: P1 = 1.268
14562227812 shelly_hap_window_c:433 WC 1: P1 = 1.427

rojer, 2020-10-07:
yes, indeed this is the log. what i see here is that calibration starts in the fully open position (cal0 state is very short), then we move to closing but power draw behaves somewhat erratically: 2.8 -> 1.3 -> 6 -> 5 -> 4 -> 7 (?) and then stabilizes at 1.3.
i take it that 1.3 is idle power consumption then). sound liek you should set threshold at 1.5. except this blip at the beginning of cal1 phase (2.8 -> 1.3) would terminate calibration.
can you record a few more of these? also, getting type of the motor would be very helpful.
are these tiny shutters or something? 5W is very little consumption.

masafra86, 2020-10-07:
Ok. Tomorrow I'll do other tests. 
The shelly operates a window opener motor similar to this http://www.specialvetroserramenti.it/catalogo/links/154_Manuale_attuatore_Comunello_Smart.pdf
I admit it's not very powerful, but it's installed in a small place, so it can't be more powerful.
Anyway it's also quite noisy, so I can't make other tests (it's late here in Italy).

masafra86, 2020-10-08:
Ok so. This is from closed window:

9326193 mgos_mongoose.c:66      New heap free LWM: 18320
9341680 mgos_mongoose.c:66      New heap free LWM: 17784
9604346 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
11856406 mg_rpc.c:292            Shelly.SetConfig via HTTP 192.168.1.6:61576
11889920 mgos_sys_config.c:232   Loading conf2.json
11913849 mgos_sys_config.c:232   Loading conf3.json
12047463 mgos_sys_config.c:174   Saved to conf9.json
12073530 shelly_hap_window_c:293 WC 1: State: idle -> precal0 (0 -> 10)
12173011 shelly_output.cpp:52    Output 2: on -> off (precal0)
12181811 shelly_hap_window_c:386 Begin calibration
12213233 mgos_sys_config.c:232   Loading conf2.json
12237136 mgos_sys_config.c:232   Loading conf3.json
12370741 mgos_sys_config.c:174   Saved to conf9.json
12386891 shelly_output.cpp:52    Output 1: off -> on (precal0)
12396896 shelly_hap_window_c:293 WC 1: State: precal0 -> cal0 (10 -> 11)
12446099 shelly_hap_window_c:402 WC 1: P0 = 3.665
13246079 shelly_hap_window_c:402 WC 1: P0 = 1.451
13272391 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
13596416 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
14046172 shelly_hap_window_c:402 WC 1: P0 = 7.878
14427438 esp_main.c:158          SDK: pm open,type:0 0
14605726 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
14846184 shelly_hap_window_c:402 WC 1: P0 = 10.573
15598361 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
15646248 shelly_hap_window_c:402 WC 1: P0 = 8.689
16445991 shelly_hap_window_c:402 WC 1: P0 = 8.244
16600042 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
16664427 shelly_main.cpp:471     Uptime: 16.66, conns 0/0/9 ns 0, RAM: 38760, 21404 free
17246180 shelly_hap_window_c:402 WC 1: P0 = 6.726
17595930 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
18046039 shelly_hap_window_c:402 WC 1: P0 = 7.037
18622804 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
18845970 shelly_hap_window_c:402 WC 1: P0 = 5.787
19623004 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
19646830 shelly_hap_window_c:402 WC 1: P0 = 7.878
20446125 shelly_hap_window_c:402 WC 1: P0 = 6.372
20631117 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61576
21246134 shelly_hap_window_c:402 WC 1: P0 = 7.780
21358048 mg_rpc.c:292            Sys.Reboot via HTTP 192.168.1.6:61576
21403327 mgos_system.c:58        Rebooting in 500 ms
21426372 HAPPlatformTCPStrea:104 0x3fff58fc 0.0.0.0:9000 HAP connection closed, ns 0/0/9 i 1
21437432 shelly_main.cpp:354     HAP server state: 0
482233 mgos_rpc_channel_mq:203 0x3fff4214 FinestraBagno/rpc
492572 mgos_rpc_channel_ua:313 0x3fff4cc4 UART0
515142 mgos_ade7953.c:96       ADE7953 silicon version: 0x02 (2)
549609 shelly_main.cpp:304     === Creating accessories
560958 shelly_hap_window_c:167 WC 1: not calibrated
571419 shelly_main.cpp:328     === Starting HAP server (CN 11)
639416 HAPPlatformTCPStrea:185 Listening on 9000
650033 mgos_mdns.c:82          Listening on udp://:5353
666421 mgos_init.c:33          Init done, RAM: 38760 total, 24764 free, 24768 min free
715246 shelly_main.cpp:354     HAP server state: 1
723978 mgos_mongoose.c:66      New heap free LWM: 24512
732580 mgos_mdns.c:55          mDNS socket closed
743135 mgos_mdns.c:82          Listening on udp://:5353
757405 mgos_ota_core.c:1335    UID: 03d105f19e7b6c97, license: none
771614 mgos_mongoose.c:66      New heap free LWM: 23688
780250 mgos_mdns.c:55          mDNS socket closed
789851 mgos_mdns.c:82          Listening on udp://:5353
799779 mgos_net.c:86           WiFi STA: connecting
3453573 esp_main.c:158          SDK: scandone
4348192 esp_main.c:158          SDK: state: 0 -> 2 (b0)
4378587 esp_main.c:158          SDK: state: 2 -> 3 (0)
4404779 esp_main.c:158          SDK: state: 3 -> 5 (10)
4447124 esp_main.c:158          SDK: add 0
4454566 esp_main.c:158          SDK: aid 1
4462198 esp_main.c:158          SDK: cnt 
4478942 esp_main.c:158          SDK: 
[... 57 lines cut ...]
17312986 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
17905632 shelly_hap_window_c:433 WC 1: P1 = 6.963
18315274 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
18705562 shelly_hap_window_c:433 WC 1: P1 = 4.652
19322745 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
19505819 shelly_hap_window_c:433 WC 1: P1 = 5.000
20305581 shelly_hap_window_c:433 WC 1: P1 = 5.872
20323498 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
21105555 shelly_hap_window_c:433 WC 1: P1 = 5.183
21315386 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
21905569 shelly_hap_window_c:433 WC 1: P1 = 5.000
22328288 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
22705564 shelly_hap_window_c:433 WC 1: P1 = 3.902
23313946 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
23505661 shelly_hap_window_c:433 WC 1: P1 = 4.159
24305568 shelly_hap_window_c:433 WC 1: P1 = 4.018
24324640 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
24666315 shelly_main.cpp:471     Uptime: 24.66, conns 0/0/9 ns 0, RAM: 38760, 22664 free
25105531 shelly_hap_window_c:433 WC 1: P1 = 3.963
25324841 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
25905580 shelly_hap_window_c:433 WC 1: P1 = 4.665
26328750 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
26705581 shelly_hap_window_c:433 WC 1: P1 = 6.841
27318432 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61581
27505586 shelly_hap_window_c:433 WC 1: P1 = 1.439

masafra86, 2020-10-08:
Before calibrating again, this is the log....

109170341 shelly_hap_window_c:433 WC 1: P1 = 1.354
109905431 shelly_hap_window_c:433 WC 1: P1 = 1.494
110705319 shelly_hap_window_c:433 WC 1: P1 = 1.463
111505485 shelly_hap_window_c:433 WC 1: P1 = 1.409
112305411 shelly_hap_window_c:433 WC 1: P1 = 1.463
112666347 shelly_main.cpp:471     Uptime: 112.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
113105779 shelly_hap_window_c:433 WC 1: P1 = 1.415
113905666 shelly_hap_window_c:433 WC 1: P1 = 1.378
114705574 shelly_hap_window_c:433 WC 1: P1 = 1.256
115505734 shelly_hap_window_c:433 WC 1: P1 = 1.451
116305411 shelly_hap_window_c:433 WC 1: P1 = 1.378
117105427 shelly_hap_window_c:433 WC 1: P1 = 1.433
117905581 shelly_hap_window_c:433 WC 1: P1 = 1.427
118705398 shelly_hap_window_c:433 WC 1: P1 = 1.439
119505418 shelly_hap_window_c:433 WC 1: P1 = 1.317
120305505 shelly_hap_window_c:433 WC 1: P1 = 1.488
120666389 shelly_main.cpp:471     Uptime: 120.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
121105508 shelly_hap_window_c:433 WC 1: P1 = 1.500
121905369 shelly_hap_window_c:433 WC 1: P1 = 1.415
122705584 shelly_hap_window_c:433 WC 1: P1 = 1.482
123505317 shelly_hap_window_c:433 WC 1: P1 = 1.280
124305433 shelly_hap_window_c:433 WC 1: P1 = 1.457
125105458 shelly_hap_window_c:433 WC 1: P1 = 1.433
125905457 shelly_hap_window_c:433 WC 1: P1 = 1.311
126705532 shelly_hap_window_c:433 WC 1: P1 = 1.439
127505393 shelly_hap_window_c:433 WC 1: P1 = 1.329
128305477 shelly_hap_window_c:433 WC 1: P1 = 1.451
128666347 shelly_main.cpp:471     Uptime: 128.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
129105389 shelly_hap_window_c:433 WC 1: P1 = 1.463
129905478 shelly_hap_window_c:433 WC 1: P1 = 1.433
130705361 shelly_hap_window_c:433 WC 1: P1 = 1.323
131505384 shelly_hap_window_c:433 WC 1: P1 = 1.329
132305466 shelly_hap_window_c:433 WC 1: P1 = 1.470
133105555 shelly_hap_window_c:433 WC 1: P1 = 1.549
133905398 shelly_hap_window_c:433 WC 1: P1 = 1.451
134705503 shelly_hap_window_c:433 WC 1: P1 = 1.402
135505539 shelly_hap_window_c:433 WC 1: P1 = 1.415
136305294 shelly_hap_window_c:433 WC 1: P1 = 1.537
136666444 shelly_main.cpp:471     Uptime: 136.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
137105459 shelly_hap_window_c:433 WC 1: P1 = 1.409
137905460 shelly_hap_window_c:433 WC 1: P1 = 1.274
138705565 shelly_hap_window_c:433 WC 1: P1 = 1.476
139505541 shelly_hap_window_c:433 WC 1: P1 = 1.433
140305459 shelly_hap_window_c:433 WC 1: P1 = 1.378
141105416 shelly_hap_window_c:433 WC 1: P1 = 1.390
141905467 shelly_hap_window_c:433 WC 1: P1 = 1.445
142705492 shelly_hap_window_c:433 WC 1: P1 = 1.402
143505377 shelly_hap_window_c:433 WC 1: P1 = 1.543
144305429 shelly_hap_window_c:433 WC 1: P1 = 1.341
144666349 shelly_main.cpp:471     Uptime: 144.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
145105406 shelly_hap_window_c:433 WC 1: P1 = 1.335
145905435 shelly_hap_window_c:433 WC 1: P1 = 1.451
146705617 shelly_hap_window_c:433 WC 1: P1 = 1.427
147505498 shelly_hap_window_c:433 WC 1: P1 = 1.573
148305600 shelly_hap_window_c:433 WC 1: P1 = 1.396
149105240 shelly_hap_window_c:433 WC 1: P1 = 1.476
149905489 shelly_hap_window_c:433 WC 1: P1 = 1.445
150705546 shelly_hap_window_c:433 WC 1: P1 = 1.421
151505456 shelly_hap_window_c:433 WC 1: P1 = 1.451
152305480 shelly_hap_window_c:433 WC 1: P1 = 1.494
152666365 shelly_main.cpp:471     Uptime: 152.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
153105461 shelly_hap_window_c:433 WC 1: P1 = 1.415
153905471 shelly_hap_window_c:433 WC 1: P1 = 1.360
154705387 shelly_hap_window_c:433 WC 1: P1 = 1.396
155505422 shelly_hap_window_c:433 WC 1: P1 = 1.354
156305458 shelly_hap_window_c:433 WC 1: P1 = 1.348
157105395 shelly_hap_window_c:433 WC 1: P1 = 1.457
157905490 shelly_hap_window_c:433 WC 1: P1 = 1.384
[... 75 lines cut ...]
213105312 shelly_hap_window_c:433 WC 1: P1 = 1.390
213905386 shelly_hap_window_c:433 WC 1: P1 = 1.415
214705505 shelly_hap_window_c:433 WC 1: P1 = 1.317
215505716 shelly_hap_window_c:433 WC 1: P1 = 1.335
216306267 shelly_hap_window_c:433 WC 1: P1 = 1.488
216666345 shelly_main.cpp:471     Uptime: 216.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
217105416 shelly_hap_window_c:433 WC 1: P1 = 1.427
217905413 shelly_hap_window_c:433 WC 1: P1 = 1.341
218705339 shelly_hap_window_c:433 WC 1: P1 = 1.201
219505342 shelly_hap_window_c:433 WC 1: P1 = 1.348
220305395 shelly_hap_window_c:433 WC 1: P1 = 1.341
221105351 shelly_hap_window_c:433 WC 1: P1 = 1.366
221905358 shelly_hap_window_c:433 WC 1: P1 = 1.591
222705391 shelly_hap_window_c:433 WC 1: P1 = 1.372
223505565 shelly_hap_window_c:433 WC 1: P1 = 1.512
224305387 shelly_hap_window_c:433 WC 1: P1 = 1.409
224666328 shelly_main.cpp:471     Uptime: 224.66, conns 0/0/9 ns 0, RAM: 38760, 23660 free
225105498 shelly_hap_window_c:433 WC 1: P1 = 1.390
225905742 shelly_hap_window_c:433 WC 1: P1 = 1.518
226287491 mgos_http_server.c:180  0x3fff61c4 HTTP connection from 192.168.1.6:61610
226342355 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61610
226705367 shelly_hap_window_c:433 WC 1: P1 = 1.354
226986125 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61610
227505468 shelly_hap_window_c:433 WC 1: P1 = 1.384
228305447 shelly_hap_window_c:433 WC 1: P1 = 1.402

masafra86, 2020-10-08:
Clicked on 'calibration', this is what i get:

291217563 shelly_hap_window_c:402 WC 1: P0 = 6.567
291952250 shelly_hap_window_c:402 WC 1: P0 = 6.451
291990207 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
292752167 shelly_hap_window_c:402 WC 1: P0 = 7.787
292962027 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
293552209 shelly_hap_window_c:402 WC 1: P0 = 6.866
293967267 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
294352207 shelly_hap_window_c:402 WC 1: P0 = 7.707
294979803 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
295152300 shelly_hap_window_c:402 WC 1: P0 = 4.902
295952187 shelly_hap_window_c:402 WC 1: P0 = 5.165
296014295 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
296666535 shelly_main.cpp:471     Uptime: 296.66, conns 0/0/9 ns 0, RAM: 38760, 22472 free
296752495 shelly_hap_window_c:402 WC 1: P0 = 18.994
297552391 shelly_hap_window_c:402 WC 1: P0 = 1.232
297652882 shelly_output.cpp:52    Output 1: on -> off (cal0)
297662972 shelly_hap_window_c:293 WC 1: State: cal0 -> postcal0 (11 -> 12)
297753362 shelly_hap_window_c:293 WC 1: State: postcal0 -> precal1 (12 -> 13)
297852115 shelly_output.cpp:52    Output 2: off -> on (precal1)
297862136 shelly_hap_window_c:293 WC 1: State: precal1 -> cal1 (13 -> 14)
297879530 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
298052448 shelly_hap_window_c:433 WC 1: P1 = 1.567
298852317 shelly_hap_window_c:433 WC 1: P1 = 1.299
299652274 shelly_hap_window_c:433 WC 1: P1 = 6.037
300452270 shelly_hap_window_c:433 WC 1: P1 = 4.427
300495690 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
301252277 shelly_hap_window_c:433 WC 1: P1 = 5.049
301835715 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
302052289 shelly_hap_window_c:433 WC 1: P1 = 5.890
302852194 shelly_hap_window_c:433 WC 1: P1 = 5.232
303652281 shelly_hap_window_c:433 WC 1: P1 = 4.537
304452254 shelly_hap_window_c:433 WC 1: P1 = 4.171
304666350 shelly_main.cpp:471     Uptime: 304.66, conns 0/0/9 ns 0, RAM: 38760, 22472 free
305252284 shelly_hap_window_c:433 WC 1: P1 = 4.183
305830488 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
306052291 shelly_hap_window_c:433 WC 1: P1 = 4.238
306852276 shelly_hap_window_c:433 WC 1: P1 = 4.232
307501828 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
307652544 shelly_hap_window_c:433 WC 1: P1 = 4.323
308452206 shelly_hap_window_c:433 WC 1: P1 = 7.348
308504641 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
309252218 shelly_hap_window_c:433 WC 1: P1 = 1.372
309510960 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
310052226 shelly_hap_window_c:433 WC 1: P1 = 1.524
310537871 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.6:61613
310852263 shelly_hap_window_c:433 WC 1: P1 = 1.622
311652185 shelly_hap_window_c:433 WC 1: P1 = 1.415

window open and status calibration "in progress"

Sorbat, 2020-10-08:
Hi Rojer, 

i have the same problem. 

Here is my log:
123299825 mgos_config_util.c:108  Extra key: [relays]
123308120 mgos_config_util.c:108  Extra key: [relays.r0]
123318111 mgos_config_util.c:108  Extra key: [relays.r0.default_state]
123326799 mgos_config_util.c:108  Extra key: [relays.r0]
123335549 mgos_config_util.c:108  Extra key: [relays.r1]
123346524 mgos_config_util.c:108  Extra key: [relays.r1.default_state]
123355225 mgos_config_util.c:108  Extra key: [relays.r1]
123363614 mgos_config_util.c:108  Extra key: [relays]
123495383 mgos_sys_config.c:174   Saved to conf9.json
123512540 shelly_hap_window_c:293 WC 1: State: stop -> stopping (24 -> 25)
123529785 shelly_hap_window_c:293 WC 1: State: stopping -> idle (25 -> 0)
123627831 shelly_hap_window_c:293 WC 1: State: idle -> move (0 -> 20)
123727157 shelly_output.cpp:52    Output 2: off -> on (move)
123738155 shelly_hap_window_c:293 WC 1: State: move -> rampup (20 -> 22)
123828204 shelly_hap_window_c:486 P = 14.26 -> 10.29
123838257 shelly_hap_window_c:293 WC 1: State: rampup -> moving (22 -> 23)
123928491 shelly_output.cpp:52    Output 2: on -> off (moving)
123938509 shelly_hap_window_c:293 WC 1: State: moving -> stop (23 -> 24)
123956421 mgos_http_server.c:180  0x3fff5994 HTTP connection from 192.168.2.101:53068
123967418 mgos_http_server.c:187  0x3fff5994 GET /status
124048157 mgos_sys_config.c:232   Loading conf2.json
124072141 mgos_sys_config.c:232   Loading conf3.json
124082840 mgos_config_util.c:108  Extra key: [relays]
124091496 mgos_config_util.c:108  Extra key: [relays.r0]
124102481 mgos_config_util.c:108  Extra key: [relays.r0.default_state]
124111181 mgos_config_util.c:108  Extra key: [relays.r0]
124119944 mgos_config_util.c:108  Extra key: [relays.r1]
124129947 mgos_config_util.c:108  Extra key: [relays.r1.default_state]
124139617 mgos_config_util.c:108  Extra key: [relays.r1]
124148018 mgos_config_util.c:108  Extra key: [relays]
124279198 mgos_sys_config.c:174   Saved to conf9.json
124295384 shelly_hap_window_c:293 WC 1: State: stop -> stopping (24 -> 25)
124311873 shelly_hap_window_c:293 WC 1: State: stopping -> idle (25 -> 0)
124409865 shelly_hap_window_c:293 WC 1: State: idle -> move (0 -> 20)
124509205 shelly_output.cpp:52    Output 1: off -> on (move)
124519275 shelly_hap_window_c:293 WC 1: State: move -> rampup (20 -> 22)
124609956 shelly_hap_window_c:486 P = 20.37 -> 10.29
124620021 shelly_hap_window_c:293 WC 1: State: rampup -> moving (22 -> 23)
124710505 shelly_output.cpp:52    Output 1: on -> off (moving)
124720568 shelly_hap_window_c:293 WC 1: State: moving -> stop (23 -> 24)
124830180 mgos_sys_config.c:232   Loading conf2.json
124854159 mgos_sys_config.c:232   Loading conf3.json
124864869 mgos_config_util.c:108  Extra key: [relays]
124873536 mgos_config_util.c:108  Extra key: [relays.r0]
124884534 mgos_config_util.c:108  Extra key: [relays.r0.default_state]
124893246 mgos_config_util.c:108  Extra key: [relays.r0]
124902021 mgos_config_util.c:108  Extra key: [relays.r1]
124912035 mgos_config_util.c:108  Extra key: [relays.r1.default_state]
124921720 mgos_config_util.c:108  Extra key: [relays.r1]
124930137 mgos_config_util.c:108  Extra key: [relays]
125061427 mgos_sys_config.c:174   Saved to conf9.json
125077629 shelly_hap_window_c:293 WC 1: State: stop -> stopping (24 -> 25)
125093448 shelly_hap_window_c:293 WC 1: State: stopping -> idle (25 -> 0)
125191498 shelly_hap_window_c:293 WC 1: State: idle -> move (0 -> 20)
125290816 shelly_output.cpp:52    Output 2: off -> on (move)
125300871 shelly_hap_window_c:293 WC 1: State: move -> rampup (20 -> 22)
125391576 shelly_hap_window_c:486 P = 10.91 -> 10.29
125401653 shelly_hap_window_c:293 WC 1: State: rampup -> moving (22 -> 23)
125492147 shelly_output.cpp:52    Output 2: on -> off (moving)
125502186 shelly_hap_window_c:293 WC 1: State: moving -> stop (23 -> 24)
125611363 mgos_sys_config.c:232   Loading conf2.json
125635350 mgos_sys_config.c:232   Loading conf3.json
125646075 mgos_config_util.c:108  Extra key: [relays]
125654758 mgos_config_util.c:108  Extra key: [relays.r0]
125665769 mgos_config_util.c:108  Extra key: [relays.r0.default_state]
[... 20 lines cut ...]
127073783 shelly_hap_window_c:486 P = 1.15 -> 10.29
127173805 shelly_hap_window_c:486 P = 4.33 -> 10.29
127182082 shelly_hap_window_c:493 Failed to start moving
127192020 shelly_hap_window_c:293 WC 1: State: rampup -> stop (22 -> 24)
127272959 shelly_output.cpp:52    Output 1: on -> off (stop)
127303530 mgos_sys_config.c:232   Loading conf2.json
127327518 mgos_sys_config.c:232   Loading conf3.json
127338255 mgos_config_util.c:108  Extra key: [relays]
127346954 mgos_config_util.c:108  Extra key: [relays.r0]
127357014 mgos_config_util.c:108  Extra key: [relays.r0.default_state]
127366362 mgos_config_util.c:108  Extra key: [relays.r0]
127375166 mgos_config_util.c:108  Extra key: [relays.r1]
127385210 mgos_config_util.c:108  Extra key: [relays.r1.default_state]
127394939 mgos_config_util.c:108  Extra key: [relays.r1]
127403383 mgos_config_util.c:108  Extra key: [relays]
127535148 mgos_sys_config.c:174   Saved to conf9.json
127551382 shelly_hap_window_c:293 WC 1: State: stop -> stopping (24 -> 25)
127568483 shelly_hap_window_c:293 WC 1: State: stopping -> idle (25 -> 0)
127666669 shelly_hap_window_c:293 WC 1: State: idle -> error (0 -> 100)
127767345 shelly_hap_window_c:313 WC 1: Tgt pos 50.00 -> 46.43 (error)
127777277 shelly_hap_window_c:293 WC 1: State: error -> idle (100 -> 0)
128947485 mgos_http_server.c:180  0x3fff62ec HTTP connection from 192.168.2.101:53071
128958573 mgos_http_server.c:187  0x3fff62ec GET /status
129679746 shelly_main.cpp:471     Uptime: 129.67, conns 0/0/9 ns 0, RAM: 38760, 22292 free
130949380 mgos_http_server.c:180  0x3fff62ec HTTP connection from 192.168.2.101:53073

C-Git-C, 2020-10-19:
Hi Rojer, 

many, many thanks for the great project. i am using a shelly plug s and a shelly 2.5 so far.

My latest purchase are two shelly 2.5 for roller shutters. Both Shellys work with the delivered firmware. 
Open and close works immediately via the buttons on the window as well as via the web interface.

But of course I want to use homekit. :-)

My Details:
Device: Model: Shelly25, ID: shellyswitch25-76xxxx
Version: 2.3.1 (build 20201018-165618/2.3.1-gb647d9c-master)

In case of version: 2.3.1 (and earlier) I sometimes have to press the button on the window or button on the web interface up to 5 times until the shutter closes. In 1 of 10 times it seems to work immediately.

It seems that the motor starts with 1 second delay and then stops immediately.
I suspect that the current power is measured by the Shelly and the motor takes a little too long until the current power consumption is high enough so that the Shelly believes that no power is used and immediately stops.

If the motor is running and I trigger a directional change then it always works.

Could you help me? That would be very nice.
Regards
C

49505510789 shelly_hap_window_c:556 Moving to 100, p 145.63
49505544137 shelly_hap_window_c:314 WC 1: Cur pos 18.90 -> 19.52, P = 145.73
49506242654 shelly_hap_window_c:556 Moving to 100, p 145.54
49506344199 shelly_hap_window_c:314 WC 1: Cur pos 23.81 -> 24.43, P = 145.53
49506379883 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49507042715 shelly_hap_window_c:556 Moving to 100, p 145.82
49507144482 shelly_hap_window_c:314 WC 1: Cur pos 28.72 -> 29.34, P = 145.94
49507373151 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49507842925 shelly_hap_window_c:556 Moving to 100, p 145.57
49507944113 shelly_hap_window_c:314 WC 1: Cur pos 33.64 -> 34.25, P = 145.52
49508374355 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49508642551 shelly_hap_window_c:556 Moving to 100, p 145.66
49508744496 shelly_hap_window_c:314 WC 1: Cur pos 38.55 -> 39.16, P = 145.93
49509376558 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49509406432 shelly_main.cpp:482     Up 49509.40, HAP 0/2/9 ns 2, RAM: 19112/38712; 4.1: c:1 mp:130.15 mt_ms:16289 cp:42.84
49509442845 shelly_hap_window_c:556 Moving to 100, p 145.23
49509544553 shelly_hap_window_c:314 WC 1: Cur pos 43.46 -> 44.07, P = 145.03
49510242673 shelly_hap_window_c:556 Moving to 100, p 145.65
49510344228 shelly_hap_window_c:314 WC 1: Cur pos 48.37 -> 48.98, P = 145.23
49510379600 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49511042375 shelly_hap_window_c:556 Moving to 100, p 144.92
49511144340 shelly_hap_window_c:314 WC 1: Cur pos 53.28 -> 53.90, P = 145.16
49511381312 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49511842855 shelly_hap_window_c:556 Moving to 100, p 145.13
49511944206 shelly_hap_window_c:314 WC 1: Cur pos 58.19 -> 58.81, P = 144.96
49512372849 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49512642406 shelly_hap_window_c:556 Moving to 100, p 144.96
49512744228 shelly_hap_window_c:314 WC 1: Cur pos 63.10 -> 63.72, P = 144.82
49513374968 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49513442920 shelly_hap_window_c:556 Moving to 100, p 145.05
49513544249 shelly_hap_window_c:314 WC 1: Cur pos 68.02 -> 68.63, P = 144.70
49514242661 shelly_hap_window_c:556 Moving to 100, p 145.06
49514344225 shelly_hap_window_c:314 WC 1: Cur pos 72.93 -> 73.54, P = 144.49
49514372750 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49515042841 shelly_hap_window_c:556 Moving to 100, p 144.85
49515144128 shelly_hap_window_c:314 WC 1: Cur pos 77.84 -> 78.45, P = 145.04
49515382576 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49515842770 shelly_hap_window_c:556 Moving to 100, p 145.02
49515944091 shelly_hap_window_c:314 WC 1: Cur pos 82.75 -> 83.36, P = 144.95
49516378827 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49516642403 shelly_hap_window_c:556 Moving to 100, p 144.59
49516744156 shelly_hap_window_c:314 WC 1: Cur pos 87.66 -> 88.27, P = 144.98
49517375797 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49517405327 shelly_main.cpp:482     Up 49517.39, HAP 0/2/9 ns 2, RAM: 19004/38712; 4.1: c:1 mp:130.15 mt_ms:16289 cp:91.96
49517442775 shelly_hap_window_c:556 Moving to 100, p 144.73
49517544548 shelly_hap_window_c:314 WC 1: Cur pos 92.57 -> 93.19, P = 144.85
49518242471 shelly_hap_window_c:556 Moving to 100, p 1.20
49518252984 shelly_hap_window_c:314 WC 1: Cur pos 97.48 -> 100.00, P = 1.20
49518263097 shelly_output.cpp:56    Output 1: on -> off (moving)
49518273189 shelly_hap_window_c:305 WC 1: State: moving -> stop (23 -> 24)
49518362319 mgos_sys_config.c:232   Loading conf2.json
49518507757 mgos_sys_config.c:174   Saved to conf9.json
49518525821 shelly_hap_window_c:305 WC 1: State: stop -> stopping (24 -> 25)
49518544519 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580
49518570105 shelly_hap_window_c:305 WC 1: State: stopping -> idle (25 -> 0)
49519374102 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xxx.xxx:49580

lucaspinelli85, 2020-12-12:
Hi @rojer , I update firmware to 2.6.1 and I need to calibrate shelly2.5 with low power motor (about 10W).
Previously updating firmware I used this command by terminal:
> ```
> curl http://shelly25-test.local/rpc/Config.Set -d '{"config":{"wc1":{"idle_power_thr":1.0}}, "reboot": true}'
> ```
> replace `shelly25-test` with the name of your shelly device.
> 
> this will set idle cutoff to 1W. you should get back
> ```
> {"saved": true}
> ```
> device will reboot, try calibrating again.

Now with new firmware it don't work, it seems it don't reboot but in terminal I get back {"saved": true}.
If I try to calibrate it go up down various times with very short stroke...
[image]
8710419028 mgos_sys_config.c:232   Loading conf2.json
8710442729 mgos_sys_config.c:232   Loading conf3.json
8710621761 mgos_sys_config.c:174   Saved to conf9.json
8710770771 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.0.19:58236
8712171891 shelly_main.cpp:554     Up 8712.16, HAP 0/5/9 ns 5, RAM: 21500/41008; st 74; 4.1: c:1 mp:8.22 mt_ms:1045 cp:49.09 tp:49.09 lemd:0 lhmd:0
8713490756 mgos_http_server.c:180  0x3fff751c HTTP connection from 192.168.0.19:58238
8713533642 shelly_debug.cpp:203    Tailing
8719956572 mg_rpc.c:305            Shelly.SetConfig via HTTP 192.168.0.19:58236
8720055033 mgos_sys_config.c:232   Loading conf2.json
8720079491 mgos_sys_config.c:232   Loading conf3.json
8720219574 mgos_sys_config.c:174   Saved to conf9.json
8720243261 shelly_hap_window_c:309 WC 1: State: idle -> precal0 (0 -> 10)
8720269962 shelly_main.cpp:554     Up 8720.26, HAP 0/5/9 ns 5, RAM: 19504/41008; st 73; 4.1: c:1 mp:8.22 mt_ms:1045 cp:49.09 tp:49.09 lemd:0 lhmd:0
8720341748 shelly_hap_window_c:431 Begin calibration
8720376616 mgos_sys_config.c:232   Loading conf2.json
8720401257 mgos_sys_config.c:232   Loading conf3.json
8720541150 mgos_sys_config.c:174   Saved to conf9.json
8720557988 shelly_output.cpp:56    Output 1: off -> on (precal0)
8720573379 shelly_hap_window_c:309 WC 1: State: precal0 -> cal0 (10 -> 11)
8721229754 shelly_hap_window_c:447 WC 1: P0 = 7.988
8721452491 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.0.19:58236
8721630419 shelly_output.cpp:56    Output 1: on -> off (cal0)
8721641956 shelly_hap_window_c:309 WC 1: State: cal0 -> postcal0 (11 -> 12)
8721730933 shelly_hap_window_c:309 WC 1: State: postcal0 -> precal1 (12 -> 13)
8721829764 shelly_output.cpp:56    Output 2: off -> on (precal1)
8721840288 shelly_hap_window_c:309 WC 1: State: precal1 -> cal1 (13 -> 14)
8721970646 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.0.19:58236
8722030202 shelly_hap_window_c:478 WC 1: P1 = 2.091
8722829590 shelly_hap_window_c:478 WC 1: P1 = 6.921
8722930397 shelly_output.cpp:56    Output 2: on -> off (cal1)
8722943483 shelly_hap_window_c:482 WC 1: calibration done, move_time 1079, move_power 7.943
8723021803 shelly_hap_window_c:309 WC 1: State: cal1 -> postcal1 (14 -> 15)
8723057544 mgos_sys_config.c:232   Loading conf2.json
8723083108 mgos_sys_config.c:232   Loading conf3.json
8723228038 mgos_sys_config.c:174   Saved to conf9.json
8723246164 shelly_hap_window_c:329 WC 1: Tgt pos 49.09 -> 50.00 (postcal1)
8723311135 shelly_hap_window_c:309 WC 1: State: postcal1 -> idle (15 -> 0)
8723332480 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.0.19:58236
8723357598 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
8723457424 shelly_output.cpp:56    Output 1: off -> on (move)
8723468898 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
8723557566 shelly_hap_window_c:531 P = 8.51 -> 7.94
8723567906 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
8724159415 shelly_hap_window_c:318 WC 1: Cur pos 44.39 -> 53.66, P = 7.54
8724636807 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
8724735864 shelly_output.cpp:56    Output 2: off -> on (move)
8724797208 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
8724836476 shelly_hap_window_c:531 P = 17.55 -> 7.94
8724847107 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
8724937464 shelly_output.cpp:56    Output 2: on -> off (moving)
8724948075 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
8725060669 mgos_sys_config.c:232   Loading conf2.json
8725086311 mgos_sys_config.c:232   Loading conf3.json
8725279153 mgos_sys_config.c:174   Saved to conf9.json
[... 187 lines cut ...]
8737743766 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
8737843191 shelly_output.cpp:56    Output 2: off -> on (move)
8737853750 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
8737943955 shelly_hap_window_c:531 P = 17.58 -> 7.94
8737954562 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
8738046100 shelly_hap_window_c:329 WC 1: Tgt pos 50.00 -> 49.49 (fixup)
8738056800 shelly_output.cpp:56    Output 2: on -> off (moving)
8738067324 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
8738165498 mgos_sys_config.c:232   Loading conf2.json
8738190157 mgos_sys_config.c:232   Loading conf3.json
8738397433 mgos_sys_config.c:174   Saved to conf9.json
8738415245 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
8738436247 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
8744171378 shelly_main.cpp:554     Up 8744.16, HAP 0/5/9 ns 5, RAM: 20992/41008; st 74; 4.1: c:1 mp:7.94 mt_ms:1079 cp:49.49 tp:49.49 lemd:0 lhmd:0
8747600535 mg_rpc.c:305            Shelly.SetConfig via HTTP 192.168.0.19:58236
[... cut ...]
8768171229 shelly_main.cpp:554     Up 8768.16, HAP 0/5/9 ns 5, RAM: 22040/41008; st 74; 4.1: c:1 mp:7.94 mt_ms:1079 cp:49.49 tp:49.49 lemd:0 lhmd:0
8776171503 shelly_main.cpp:554     Up 8776.16, HAP 0/5/9 ns 5, RAM: 22040/41008; st 73; 4.1: c:1 mp:7.94 mt_ms:1079 cp:49.49 tp:49.49 lemd:0 lhmd:0
8784171380 shelly_main.cpp:554     Up 8784.16, HAP 0/5/9 ns 5, RAM: 22528/41008; st 74; 4.1: c:1 mp:7.94 mt_ms:1079 cp:49.49 tp:49.49 lemd:0 lhmd:0
8792171446 shelly_main.cpp:554     Up 8792.16, HAP 0/5/9 ns 5, RAM: 22528/41008; st 74; 4.1: c:1 mp:7.94 mt_ms:1079 cp:49.49 tp:49.49 lemd:0 lhmd:0

andyblac, 2021-03-03:
@rojer need anything further info ?, this one gone stale.

closed 2021-03-11 by github-actions[bot]

lucaspinelli85, 2021-11-06:
I also have a low power engine of 8W for the sofà.
May I ask what is the default value of the calibration power? 10W?
Is it possible to make this value editable in the webui instead of setting it via code from the terminal?
Or better, viceversa, leave the default low value which is good for many, and for those who have the engine that absorbs from a standstill, raises the value of what you want through the code from the terminal?
**Or better still** ... during calibration, the minimum and maximum absorption is detected before and the threshold value is set automatically!
Stupid example:
- if the motor stopped absorbs 0W and in motion 100W the threshold can also be, for example, half 50W,
- if the motor is "low power" 0W at a standstill, while running 8W, the threshold will go to 4W,
- if motor with 10W absorption from standstill and 40W in motion it will be (10 + 40) / 2 = 25W.
How about it? Great idea?
It would be good for everyone, it would be automatic and without complicating the webui or the difficulty of configuration by less experienced people
I add one last gem:
- if the motor when stopped is 0W, and in motion it is still 0W it means that there is no absorption and therefore it is necessary to do the manual "timed" calibration (issue #404 and  #449)
I hope it will be considered for a future implementation!
Thanks for the replies! @andyblac @rojer