mongoose-os-apps/shelly-homekit #175: Calibration fails - no power comsumption  - Shelly 2.5

ocsele, 2020-10-05:
Dear Rojer,

First of all, let me tell what a fantastic job you guys are doing here.

Second, I'd like to shortly describe a weird issue, which I think is not related to your firmware, 'cause it happened to me on stock as well, but maybe you can point me in some direction in troubleshooting. So problem description applies to stock, and your homekit fw as well.

After setting up Shelly 25 in shutter mode, when controlling the shutter natively (Shelly app, or web gui of device in LAN), I see absolutely no power consumption. When I try to calibrate the device, it always fail: in shelly app: maxtime timeout error message, in web gui (both stock and yours) Calibration failed.

I assume for this reason (device not calibrated)  I can absolutely not control the shutter in HomeKit (with stock fw via Shelly plugin in Homebridge, with your fw obviuosly after simply adding the accessory): pressing the homekit roller button absolutely does nothing).

Question: 
- could it be simply a faulty power metering chip problem? If yes, are the two problems related: not being able to calibrate and operate it in homekit?
- could it be any wiring issue? Wiring was not done by me but a professional electrician, so I can't judge it, but can any wiring problem cause any similar issues? Again: controlling the shutter (up,down,stop, scheduled operation) from Shelly software, AND from physical buttons on the wall is working absolutely flawlessly.

Many thanks, and sorry if this is not the appropriate place to pose this question.

Keep up the great work, you guys are doing miracle here.
Best:
John

kzizi, 2020-10-05:
Hi, Rojer! 
I will use this topic for my problem which appeared when calibrating shelly 2.5 in roller mode. I have installed firmware 2.2.1 and started calibrating device. After 1-2sec calibration stopped and shelly starts to randomly click (like button is pressed) every 3-7 seconds. At this point shelly became unresponsive on wifi and webUI. I have reset it via power down and 5 clicks. After reset you can see it on wifi (shellyswitch25-xxxx) but you can’t connect. The physical buttons also doesn’t work or sometimes moves the roller for 1sec and then stop. I am pretty much stuck at this point.

For information, I have successfully completed calibration on two roller blinds in different rooms which are the same models.
I love all of the work you guys done so far!!
Regards, Žiga

ocsele, 2020-10-06:
Dear Rojer, Ziga,

first, I just donated a little sum to you Roger, because what you guys do here is a real miracle and a lifesaver for a lot of us: no more unstable, non responsive plugin usage via unsupported bridges, it just freaking simply works ALL the time, without ANY lag.

Second, about my problem: it seems I had a faulty device which didn't measure power consumption thus the calibration failed I assume. I replaced it with another one, power metering working, calibration working. Well, not at first, and maybe this could help you Ziga: I don't know if you came across this description, troubleshooting calibration failures: https://www.shelly-support.eu/lexikon/index.php?entry/61-calibration-tips-roller-shutter/

Particularly this section:
"Third - some rollers do not cut power measurement at all when a final position is reached."
This was exactly my case, but I think this can be corrected only with the stock firmware, since this URL gives nothing on this homekit enabled fw:  

http://<DeviceIP>/settings/roller  

With the original fw you get an overview of ALL current setting in json format. With this fw you get nothing when checking this link, I don't know if cut off power adjustment can be done or not, in the web gui there's no setting for this.

When I first tried to calibrate the second device (with stock fw) it failed at first attempt. After reading above description I had to adjust the cut off power 4W, cause the stop power flow was 2W, and after that calibration was successfully done (still on stock fw)!

After switching to this homekit fw, I had to recalibrate, and calibration was done flawlessly. I don't know because the modified cut off power setting was inherited from stock fw, or this fw is insensitive to having power flow after cut-off, but it worked.
Maybe this has nothing to do with your problem Ziga, I'm just sharing my experience.

andyblac, 2021-03-03:
can this be closed ?, or still an issue in 2.8.0 alpha fw ?

dominikfrey, 2023-01-11:
I could fix the power measurement by just doing a **power-cycle** as suggested here: https://github.com/arendst/Tasmota/issues/9692#issuecomment-968106851

To make it clear: power-cycling means to cut the power and turn it on after a minute (this is not the same as a reboot from the app)

This might help someone else having the problem.