mongoose-os-apps/shelly-homekit #177: WebUI: Phone-friendly layout

rojer, 2020-10-05:
Right now layout is not very phone-friendly, we get breaks in awkward places between elements.
@schemhad could you help tune the layout so it looks nice on the phone? making buttons smaller should to make them fit, fixing the header, etc.

schemhad, 2020-10-05:
Sure, I'll take a look...

Rappelsack, 2020-10-05:
I just have to take this thread here to say a big (!!!) thank you. Awesome what ure doing here. My whole house is now automated with shelly and your software.

@all other guys out there: give a donate!

rojer, 2020-10-05:
@Rappelsack thank you for your kind words!

andyblac, 2021-01-04:
pull request here https://github.com/mongoose-os-apps/shelly-homekit/pull/380 thanks to @ZeevoX

andyblac, 2021-01-25:
@ZeevoX any reason you closed the pull request ?

ZeevoX, 2021-01-25:
@andyblac because HTML `<table>` should not be used purely as a means to layout document content, and instead for presenting tabular data. I am going to re-do the layout using `<div>`s and CSS.

rojer, 2021-02-12:
fixed by #488

closed 2021-02-12 by rojer