mongoose-os-apps/shelly-homekit #178: Homekit automation doesnt work

Rappelsack, 2020-10-06:
Hi there.

One bis issue I have since 2.2.0 (and also with 2.2.1).

I have 6 shutter in my configuration with shelly 2.5. Also 5 shelly 1 for my floor heating and 4 shelly 1 for some lights.
Also I have 50 Philips Hue bulbs.

My Homekit control center is a HomePod.

Just the automation with Shelly doesnt work any more.
No one.
All other (with the hue lights work finde).

I restarted the homepod several times. Also I set up my iPad Pro 2018 as control center but it still does not work any more.

Any idea?

rojer, 2020-10-06:
very weird. my automations definitely work, i have ipad as a hub.
what kind of automations do you have set up?

andyblac, 2020-10-06:
@Rappelsack are you on iOS14 on iPhone, if there seems to be an issue with the HomePod's as they are still on iOS 13.4, I have to reboot my HomePod's several times a day, it a right pita. I have several shelly's in automations all week find once the HP has been rebooted.

Rappelsack, 2020-10-06:
It's possible that the reason is the HomePod with 13.4.

But also even I have my iPad with 14.0.1 as control center activated, it didn't work this morning.

I have my shutter to be opened 15min after sunrise. They all stayed down.
Also my temp-control of the floor heating was off (there I have a automation: checking weather, if temperature ist below XY turn it on).
This works fine with 2.2.0. And there the HP was on 13.4.8 and the iDevices on 14.0.1.

I will try to reboot again.

andyblac, 2020-10-06:
> It's possible that the reason is the HomePod with 13.4.
>
> But also even I have my iPad with 14.0.1 as control center activated, it didn't work this morning.
>
> I have my shutter to be opened 15min [...]

but was the iPad master control ?, as you can't turn off Control in HomePod, try turning off your HP, and wait at least 5 mins, for HomeKit to see it as Disconnected, check in HomeKit settings that your iPad is master.

then try automation again.

Rappelsack, 2020-10-06:
No. As soon I set my iPad „home on“ it becomes the master control. Every time.

It was while the automations did not work.

andyblac, 2020-10-06:
> No. As soon I set my iPad „home on“ it becomes the master control. Every time.
> 
> It was while the automations did not work.

still worth a try turning off HomePod. anyway my automations work here.

Rappelsack, 2020-10-06:
I apologize that this is in german.

While the iPad was master (over night and this morning) the shutter and the heating didn‘t start this morning.
I am wondering that one shelly is there as „bridge“?
[image]

andyblac, 2020-10-06:
bride is used if you're in detached mode.

Rappelsack, 2020-10-06:
What do you mean?
[image]
[image]

andyblac, 2020-10-06:
Roller shutter must also require bridge in homekit

Rappelsack, 2020-10-06:
I do not understand. 
I have 6 roller shutter in my house but only one is there as a bridge?

andyblac, 2020-10-06:
@rojer will have to confirm why then.

rojer, 2020-10-06:
hub is created when there's more than one accessory. in case of shutters, it's when input is in single or detached mode - then there's a free input to be exposed as a stateless switch, and to enable two unrelated accessories primary accessory is set to bridge.

Rappelsack, 2020-10-06:
So in that case my setting is correct and it is correct that one shelly 1 is a bridge?

Rappelsack, 2020-10-06:
I don‘t know what it is. The iPad is now connected and set as master (HomePod standby).
At 6pm the automation checks the weather and if it is below 13° outside, the floor heating switches on.

But at 6pm. Nothing happens.
I opened the Home app and „test automation“ and than they switched on.

Rappelsack, 2020-10-07:
Now I am confused which thread should be the active one.

Again my answer in Thread #47 😊 

So my next step was:
shut off the Homepod (so that only my iPad is active and the hub). Switching all automations to deactivate and activate again. Restart iPad.
And: the shutter (only this one who is displayed as a bridge) still is down even it should go up at 06:30 am this morning.
And all Shelly 1 (floor heating) stayed off this morning.

So I did the same troubleshooting but now with the Homepod as active hub:
same problems.

Therefore I suggest that it must be the firmware of the shellys? It worked finde all the way to 2.1.8.

Rappelsack, 2020-10-07:
I figured it out!

All automations with geolocation (if I am at home / if I am not at home) are not working! 

All other automations are working well.

So this should not be a bug in your FW. It must be a bug in HomeKit with any device at 14.0.1.

andyblac, 2020-10-07:
> I figured it out!
> 
> All automations with geolocation (if I am at home / if I am not at home) are not working!
> 
> All other automations are working well.
> 
> So this should not be a bug in your FW. It must be a bug in HomeKit with any device at 14.0.1.

ye, as I expected a HomeKit issue, my guess is that HomeKit hubs (any device) are still running iOS13.4 portion of the code. I suspect we will have to wait for HomePod OS 14, before it hopefully fixed. As I have said my HomeKit automations are pants atm, have to reboot my HomePod several times a day, right pita.

rojer, 2020-10-07:
@Rappelsack thank you so much for getting to the bottom of it and reporting! yeah, that sounds like homepod bug.
but i'm curious: you said all the automations work with your lights, does that include geo-location automations?
or do those also fail with the hue lights?

Rappelsack, 2020-10-07:
I never tested it if it works because it was so set that the lights just turn on when I am away from home. And now I guess they never switched on ;-)

andyblac, 2020-10-07:
fingers crossed for 14th Oct, lets hope we get HomeKit Hub updates.

Rappelsack, 2020-10-07:
I think so. The latest leaker say there will come a homepod mini. Hope so! I would buy 2 of them straight away.

andyblac, 2020-10-07:
> I think so. The latest leaker say there will come a homepod mini. Hope so! I would buy 2 of them straight away.

snap, prob more, kitchen, bedrooms etc.

andyblac, 2020-10-08:
found out today, my issues in HomeKit seemed to be an issue with Homebridge 2.2.3, I have reverted back to 1.1.6, and all my automation issues are gone, If any of you are using HB 2.2.x, try reverting to 1.1.6 see if it helps. I do not understand why HB would effect Hue, Shelly's, etc but it did.

Rappelsack, 2020-10-10:
The problem now is completely solved!
As always the problem was behind the screen. 

I got a new second iPhone as my master iPhone. But the „share location“ was still on the old iPhone. This stays at home all the time and therefore it detects me always at home. When I switched it off the HomePod deleted it as „not at home“ and my Shutter didn’t go up in the morning like it should when I am at home. 

Now I switched the share location to my main iPhone and it seems to work!

andyblac, 2020-10-10:
cool, I'll close this ticket then, thanks for letting us know...

closed 2020-10-10 by andyblac

rojer, 2020-10-10:
hah, ok! thanks for letting us know