mongoose-os-apps/shelly-homekit #179: Home Add Accessory doesn't work.

fd007, 2020-10-06:
After entering setup code I get following response.

Unable to Add Accessory -  An unknown error occurred.

Some Ideas?

andyblac, 2020-10-06:
> After entering setup code I get following response.
> 
> Unable to Add Accessory - An unknown error occurred.
> 
> Some Ideas?

entering where webUI or HomeKit ?, also what code was used ? some simple codes like 111-11-111 or not valid in homekit.

fd007, 2020-10-06:
1. Entering in HomeKit.
2. I don't use invalid code.

andyblac, 2020-10-06:
maybe code was used before, try reset button in HomeKit section in webUI, codes are only valid once iirc.

fd007, 2020-10-06:
Silly question, but can you give hint how to open Homekit webUI?

andyblac, 2020-10-06:
> Silly question, but can you give hint how to open Homekit webUI?
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Device-webUI
https://github.com/mongoose-os-apps/shelly-homekit/wiki/HomeKit-settings

andyblac, 2020-10-07:
did it help ?

fd007, 2020-10-10:
No, it did not - unfortunately.

Is it possible that Apple disable adding device to Homekit after Homekit setup code have been changed for several times?

Is there some way how to reset is again?

Thx!

andyblac, 2020-10-10:
You reset it in webUI reset HomeKit

fd007, 2020-10-10:
No, what I mean is that Apple block the device, after setup code has been reseted in webUI several times?

andyblac, 2020-10-10:
> No, what I mean is that Apple block the device, after setup code has been reseted in webUI several times?

no they don't

andyblac, 2020-10-10:
are you using. mesh wifi?, I have herd that can cause issues with homekit.

fd007, 2020-10-10:
Yeah, that could it be. Thanks for this Idea!

rojer, 2020-10-10:
right, i've reset and re-added device many times, they don't get banned or anything.
pressing "reset" in homekit section completely wipes homekit state, including unique device id, after that it should definitely appear in the list of devices to add.

andyblac, 2020-10-10:
> right, i've reset and re-added device many times, they don't get banned or anything.
> pressing "reset" in homekit section completely wipes homekit state, including unique device id, after that it should definitely appear in the list of devices to add.

I think his issue is mesh wifi, and multicast

andyblac, 2020-10-12:
you get his sorted ?

rojer, 2020-10-22:
closing as stale

closed 2020-10-22 by rojer