mongoose-os-apps/shelly-homekit #180: Roller Shutter mode support for Shelly 2 (not 2.5)

plantoschka, 2020-10-06:
Hello,

I've a couple of Shelly 2 which I use to control shutters in several rooms. I was excited to learn that this firmware now also offically supports roller shutter mode and flashed the new firmware.
Just then I noticed this support is only for Shelly 2.5 and not 2.

Is it planned to add this to Shelly 2? 
Is there a reason why this is not implemented for Shelly 2?

Thanks

rojer, 2020-10-06:
that's because we use power meter of shelly 2.5 to tell when movement has stopped.
shelly 2 has no pwer meters, so we can't do it.

plantoschka, 2020-10-06:
Shelly2 has Power Measurement

Just did an screenshot of mine from the web interface while opening the shutter.
[image]

rojer, 2020-10-06:
hm, i was mistaken. yes, then shelly2 can support roller shutter mode.
it has a different power meter chip however, we'll need driver for it.

rojer, 2020-10-06:
i don't have a shelly2 to test with, however

rojer, 2020-10-06:
oh, there's a problem: it measures consumption on both load channels, we need separate power measurements.

plantoschka, 2020-10-07:
Shelly 2 works perfectly fine for me with roller shutter in four rooms. I don't see any reason to upgrade to 2.5
But if it's to hard to implement this I will still use Homebridge with the official firmware.

rojer, 2020-10-07:
i think we can accommodate shelly 2, let me think about it.

rojer, 2020-10-08:
i think this is possible but i need a device to test with.
i checked and nobody sells them anymore, so unless you can send me one i don't think this is going to happen.

rojer, 2020-10-11:
closing as not feasible, please reopen if you can provide hardware.

plantoschka, 2020-10-12:
> closing as not feasible, please reopen if you can provide hardware.

I can provide a device.
I have a spare Shelly 2.5 which I will use to replace one of my Shelly 2.

rojer, 2020-10-12:
good, please reach out to me at rojer@rojer.me to arrange delivery.

plantoschka, 2020-11-27:
@rojer 
You think this can be implementend sometimes in the coming weeks/months?
I'm thinking about just buying some Shelly 2.5 on the Black Friday offer.

Thank you

rojer, 2020-11-27:
unfortunately, this is not easy. i've had a look and the metering device is connected using UART, not I2C (as on shelly 2.5).
this is the same UART that is used for flashing and console, so it further complicates things.
so, weeks - definitely not, months - maybe, but not for sure.

andyblac, 2021-03-03:
@rojer close as not doable or add to enhancement flag ?

closed 2021-03-11 by github-actions[bot]