mongoose-os-apps/shelly-homekit #181: Roller Shutter Input Mode and Apple Home App

socialsubs, 2020-10-07:
Hi,
I would like to know if there is a way to set the roller shutter input mode to cycle between **close-stop-open** when I short press the roller shutter icon in the the Apple Home App.

I tried all available settings but failed to get the desired result. My roller shutters always goes down (or up if i swap outputs). The cycle is always close-stop, close-stop, and so on...

Thanks in advance,

andyblac, 2020-10-07:
this sounds like this ticket here https://github.com/mongoose-os-apps/shelly-homekit/issues/142

andyblac, 2020-10-07:
or maybe you are only using one button not 2 ? 1 for up, 1 for down ?

socialsubs, 2020-10-07:
I have two separate buttons to control my roller shutters. one for open and the other for close, but there is only one icon for the roller shutter in the Apple Home App. The only two actions available are: short press and long press. The short press action, cycles between close-stop, close-stop and so on, while the long press action shows a pop-up to manually adjust the roller shutter opening percentage.

andyblac, 2020-10-07:
ah ok, I'll leave this one for @rojer or others who use roller shutter, sorry but I don't have one to test / help with.

rojer, 2020-10-07:
right, actions when tapping on the tile are not very intuitive, i find the cycle behavior (like you describe and like we have on the external input in single input mode) more intuitive.
right now we just execute orders it sends: tapping the tile sends either fully open or full close, with observed behavior being to "prfer close", i.e. if the curtain is open even a little bit, tapping the tile will send close on every tap until it's fully closed.
i'd say this should be implemented in the home app, so it behaves like you described.
however, as that is probably not happening any time soon, we could in the mean time be more smart on the device side and interpret 0 and 100% positions as tap signals, ignore what they say and perform the same open-stop-close routine we do when handling single external input. i think i'll make it an option, enabled by default.

socialsubs, 2020-10-07:
Great, many thanks rojer!!

rojer, 2020-10-07:
@socialsubs fixed on master, will be in the next release.
it is so intuitive that i decided to not even make it an option.
thanks for reporting!

closed 2020-10-07 by rojer